Arduino-IRremote/Arduino-IRremote #567: ATtiny85 working frequencies

DaveCalaway, 2018-05-01:
**Board:** ATtiny85 ( trinket Adafruit )
**Library Version:** 2.2.3
**Example:** IRsendDemo

According to the example, it send 3 signal and wait for 5 seconds, but if i try to use different frequency this is the results:
**8Mhz**
[image]

**16Mhz**
[image]

The waiting time is always much lower then 5 seconds.... 
Somebody have an idea how can i fix?

AnalysIR, 2018-05-01:
Some possibilities:

1. IRremote uses Timer0 inthe '85. It could be that your arduino core is using the same timer...which gets messed up by IRremote. If so, you may need to re-init the 'arduino' timer after sending???
Atmel, has non timer based delay routines available to use for delays or you could write a loop to simulate a timer.

_A good test would be to run the led blink example ot see if the timing is aaccurate without IRremote!_

2. Make sure that the F_CPU 'define' is properly defined to match your CPU frequency, otherwise it defaults to 16MHz

_A good test would be to print the value for F_CPU or bink a led once for every 1 MHz._

Hopefully, that will get you closer to a resolution.

_Working on small devices ususally requires a higher degree of expertise than standard Arduinos & you should really be comfortable working at the lower AVR register level etc.
These days I would just get a few  ~$2 (incl delivery) Arduino Nanos from Aliexpress instead of the lower spec 'ATTinys'._

AnalysIR, 2018-05-03:
Physical pin 1 of the IC may also be the reset pin. (Based on the timer being used see below)

The Arduino digital pin 1 may actually be physical pin 6 on the ATTiny IC (subject to the core being used, please verify).

Unfortunately, I don't have your setup to hand so it will be hard for me to provide any meaningful assistance, beyond informed guesses.

DaveCalaway, 2018-05-03:
Im sorry, my mistake.
I use the physical pin 6 ( aka pin 1 ) to control all sketch i used.

I runed the Blink sketch with 8Mhz and 16Mhz ( with ATTinyCore ) and in both cases it works.

I have runed the last [sketch](https://goo.gl/8haaYu) i have sent to you ( 16Mhz ) but the physical pin 6 ( aka pin 1 ) did not give me nothing... On my digital analyzer i have 0V ( signal low ) every time.

AnalysIR, 2018-05-03:
OK, unfortunately, I can't offer any more suggestions, other than detailed debugging.

Are your certain the FUSES are set correctly on the AVR for the correct oscillator & speed.

DaveCalaway, 2018-05-03:
This is my setting, 
do you know what is "Timer 1 Clock"?
[image]

AnalysIR, 2018-05-03:
Yes, Just leave it with CPU selected. For more details read the 234 page datasheet :)

FYI: Just because you select CLock speed of 16 or 8Mhz in that menu, doesnt mean it works that way...FUSE setting is also important (details in Data sheet). Sarch fo some turorials on setting up AVRs & FUSES or just revert to the original settings of the manufacturer.

AnalysIR, 2018-05-03:
It looks like you have selected the internal 64MHz PLL@ with effective system clock of 16MHz.

So based on a quick review it should be fine. However, I would personally prefer to go with 8MHz internal OSC, without a crystal.

I suggest you do some searching to find a post or blog about configuring the '85....presumably there is some stuff on the module manufacturers website?.

`if (F_CPU == 16000000) clock_prescale_set(clock_div_1);`
this could be an issue as the PLL is automatically set to div/4 to get from 64 to 16. However, I would need to study the data sheet in detail to be sure.

DaveCalaway, 2018-05-05:
I did some tests and with the ATtinycore it is not necessary to add in the code the following lines to set the clock speed or prescaler:
`if (F_CPU == 16000000) clock_prescale_set(clock_div_1);`
It fixes the fuse itself by burning the bootloader function.

So i tested some basics examples and it works fine.
But every time i try to load ones of your example, the pin 1 ( aka physical  6 ) don't produce nothing.

What type of core did you use to do the test?

AnalysIR, 2018-05-05:
I didn't test it & haven't used ATTiny's  with IRremote for years, because it is so cheap to get a Nano or micro...sorry.

Maybe someone who has used an ATtiny85 with IRremote will jump in?

[ATtiny85 related issues open+closed](https://github.com/z3t0/Arduino-IRremote/issues?utf8=✓&q=++attiny85+)

DaveCalaway, 2018-05-05:
Nice and good solution for ATtiny85: https://goo.gl/cAxs1T
Do you think to update the guide in main repo?

AnalysIR, 2018-05-06:
> Nice and good solution for ATtiny85: https://goo.gl/cAxs1T
> Do you think to update the guide in main repo?

@z3t0  ....FYI

hugovangalen, 2020-01-28:
I also had issues with sending IR codes on ATtiny85.

I can confirm that the linked patched version [in the comment above](https://github.com/z3t0/Arduino-IRremote/issues/567#issuecomment-386838152) works properly on ATtiny85.

ArminJo, 2020-10-12:
IRremote and [IRMP ](https://github.com/ukw100/IRMP) now supports ATtinies and Digispark.

closed 2020-10-12 by ArminJo