Arduino-IRremote/Arduino-IRremote #572: memory overkill for atmega8A

orzel1244, 2018-05-22:
**Board:** atmega8A
**Library Version:** 2.1.0
**Protocol:** ...

Hey, all I want from the lib, is just reading signals, I don't want to recognize if it's Sony, Lg or something else, sadly the lib is overkill for atmega8A, it uses around 98% of memory, any ideas how to deal with this issue? I'm only waiting for recive signal and blinking led.
No, I don't want to change for better micro controller, I want to keep it cheep

AnalysIR, 2018-05-22:
Look at the file IRremote.h and disable (set to 0) any send or receive protocol you do not need. That should save you lots of resources.

> // Supported IR protocols
> // Each protocol you include costs memory and, during decode, costs time
> // Disable (set to 0) all the protocols you do not need/want!
> //......

orzel1244, 2018-05-22:
Thanks, I'll check it out later

AnalysIR, 2018-05-31:
Dont forget to cloase the issue when you are ready.....

closed 2018-06-08 by orzel1244