Arduino-IRremote/Arduino-IRremote #573: I think my IR codes are sending too fast.

matv201, 2018-05-31:
**Board:** ARDUINO PRO MINI (3.3V atmega128 8mhz)

**Code Block:**
`#include <IRremote.h>
IRsend irsend;
void setup() {}

void loop() {
irsend.sendNEC(0xE1A2E817, 32);
irsend.sendNEC(0xC40BF, 32);
//
//69 other lines are in between the above and the delay
//but I didn't paste them here
//because they are exactly like the first two up there
//
delay(1500);
}`

I have 71 lines like those in loop(). They are all "irsend.sendNEC". All of them execute in under 10 seconds. No errors are thrown, but I'm worried that the codes are sent out too fast, because the TV-B-Gone has 250 codes and that takes about 2 minutes to run. The proportion seems wrong here.

AnalysIR, 2018-05-31:
Interesting question :)

I did a quick check...a single NEC signal can be sent in about 70 mSecs.
So 71 takes about 5 seconds.
If you send the 2 lines 71 times then it would work out at about 10 Secs.

So it all looks fine !!

I presume this is not a real world test, because you need to leave circa 100mSecs between each send to avoid confusing the IR receiver!

bengtmartensson, 2018-06-01:
The initial part of the NEC1 signal is ends with a silence, making it exactly 108 ms long. The repeat-packets also ends with a silence, making each one 108 ms. (The IRP notation for NEC1 reads `{38.4k,564}<1,-1|1,-3>(16,-8,D:8,S:8,F:8,~F:8,1,^108m,(16,-4,1,^108m)*) [D:0..255,S:0..255=255-D,F:0..255]`  .)

A few years ago, I performed some experiments with a Yamaha receiver (which is using the NEC1 commands), using its volume up/down commands. It turned out that it was somewhat picky: If the timings were either too short or too long, the signal was not recognized as a repeating signal.

IRremote does not do this very well. It was not designed for it. Instead, you may like to try Infrared4Arduino (available in the Arduino library manager using the name `Infrared`). See its example `Nec1Renderer`, which sends (surprise!) the Yamaha volume down command + 10 repeats.

forgottenecho, 2018-06-01:
@AnalysIR 
Thanks for the speedy and knowledgeable reply!

So I actually timed the length of my LED's "on" (lit up) time, which turned out to be 5-6 seconds. I am only sending 71 signals. Not 2*71! (But your original answer still applies, so I'm good on that issue)

But, I have one more question:
I am making a Projector-B-Gone, which works exactly like the TV-B-Gone but it shuts off projectors instead. Every code is a power code. I do have multiple power codes in there that all correspond to the same brand (but not necessarily the same model) of projector. So, should I put delays in between codes (which I'm guessing the TV-B-Gone does because it sends 250 codes in about 2 minutes) or should I let them send one after the other without delay?

AnalysIR, 2018-06-01:
So it looks like a 400mSec gap approx. for 250 in 120 secs.

FYI: Some IR receivers will effectively turn off if they detect too much IR.

So yes, create a variable(uint16_t val) with a value of 400 and then just issue a delay(val) between each signal send.

Once you get that working, you can reduce it by 50 in steps to see where it stops working reliably.

Another thing to consider is that some devices may have the on/off as a toggle function, so when you send the power signal the state is toggled. So you could end up in all sorts of a mess, if you only want to turn things off or on.

Many systems will also have what is referred to as discrete codes which may be undocumented. If you can find them you then get full command of individual ON or off signals. (you could go thru each command value for every unit if you cant find discrete codes...also check remotecentral.com)

Finally, beware of the toggle bits in RC6 & RC5 style protocols.

I am curious as to the project, is it DIY, school or commercial...what is the broader use case?

forgottenecho, 2018-06-01:
@AnalysIR 
I read up on discrete codes... very interesting!

Thanks, you're a big help!
I'm a "grey-hat" high school student. I have some other REALLY cool projects that I can't talk about on here, but yes. I like to experiment.

For this particular project, I actually brought an IR receiver to school and had my laptop running arduino code in my backback. So, the receiver module was discretely sticking out my bag, and I waited for my teacher to turn on the projector to get power codes. 

The project is coming along great and I'm going to enclose it in a flashlight. Otherwise, people freak out when they see components and a pcb.

closed 2018-06-01 by forgottenecho

AnalysIR, 2018-06-01:
:)
....Pranks are fun...but don't overdo it!