Arduino-IRremote/Arduino-IRremote #576: Library miskate the brand sanyo and samsung

nazuna1, 2018-06-18:
**Board:** ARDUINO UNO
**Library Version:** 2.2.3
**Protocol:** Samsung

**Code Block:**
```c

#include <IRremote.h> 

void  encoding (decode_results *results)
{
  switch (results->decode_type) {
    default:
    case UNKNOWN:      Serial.print("UNKNOWN");       break ;
    case NEC:          Serial.print("NEC");           break ;
    case SONY:         Serial.print("SONY");          break ;
    case RC5:          Serial.print("RC5");           break ;
    case RC6:          Serial.print("RC6");           break ;
    case DISH:         Serial.print("DISH");          break ;
    case SHARP:        Serial.print("SHARP");         break ;
    case JVC:          Serial.print("JVC");           break ;
    case SANYO:        Serial.print("SANYO");         break ;
    case MITSUBISHI:   Serial.print("MITSUBISHI");    break ;
    case SAMSUNG:      Serial.print("SAMSUNG");       break ;
    case LG:           Serial.print("LG");            break ;
    case WHYNTER:      Serial.print("WHYNTER");       break ;
    case AIWA_RC_T501: Serial.print("AIWA_RC_T501");  break ;
    case PANASONIC:    Serial.print("PANASONIC");     break ;
    case DENON:        Serial.print("Denon");         break ;
  }
}

void  loop ( )
{
  decode_results  results;        // Somewhere to store the results

  if (irrecv.decode(&results)) {  // Grab an IR code
    dumpInfo(&results);           // Output the results
    dumpRaw(&results);            // Output the results in RAW format
    dumpCode(&results);           // Output the results as source code
    Serial.println("");           // Blank line between entries
    irrecv.resume();              // Prepare for the next value
  }
}

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [✔] I have **read** the README.md file thoroughly
- [✔] I have searched existing issues to see if there is anything I have missed.
- [✔] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [✔] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [✔] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

This library shows incorrect data.
My remote control  is Samsung but Lib return Sanyo.
So, returm code is 0xFFFFFFFF.

AnalysIR, 2018-06-18:
First of all, there is not neccessarily a direct relationship between the 'arbitrary' protocol name and the manufacturer, as many brands re-sell devices form other OEMs.

Unfortunately, there is not enough info above to help you further. Please post the raw output from the irrecvDumpv2 example. 

Also note, that in NEC protocols the library reports 0xFFFFFFFF for repeat signals. These repeats are sent when the button is kept pressed down.

nazuna1, 2018-06-19:
@AnalysIR The below code is irrecvDumpV2 and same code outputs first issue data

Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
Timing[115]: 
     + 450, - 550     + 500, -1450     + 500, - 500     + 500, - 450
     + 550, - 450     + 550, - 450     + 450, - 550     + 450, - 500
     + 500, - 500     + 500, -1450     + 500, - 500     + 500, - 500
     + 450, -1550     + 450, - 500     + 450, - 550     + 450, -1500
     + 550, -1450     + 500, -1450     + 550, -1450     + 450, -1500
     + 500, - 500     + 450, - 500     + 500, - 500     + 500, - 500
     + 500, - 500     + 450, - 500     + 500, - 550     + 450, - 500
     + 500, - 500     + 450, - 550     + 450, - 500     + 500, - 500
     + 500, - 500     + 500, - 450     + 550, - 450     + 500, - 500
     + 500, - 500     + 450, - 500     + 500, - 500     + 500, - 500
     + 500, - 450     + 550, - 450     + 550, - 450     + 500, - 500
     + 500, - 500     + 450, - 500     + 500, - 500     + 450, - 550
     + 450, - 550     + 500, - 450     + 550, - 450     + 500, - 500
     + 500, -1450     + 500, -1450     + 500, -1500     + 450, -1500
     + 500, -2950     +2950
unsigned int  rawData[115] = {450,550, 500,1450, 500,500, 500,450, 550,450, 550,450, 450,550, 450,500, 500,500, 500,1450, 500,500, 500,500, 450,1550, 450,500, 450,550, 450,1500, 550,1450, 500,1450, 550,1450, 450,1500, 500,500, 450,500, 500,500, 500,500, 500,500, 450,500, 500,550, 450,500, 500,500, 450,550, 450,500, 500,500, 500,500, 500,450, 550,450, 500,500, 500,500, 450,500, 500,500, 500,500, 500,450, 550,450, 550,450, 500,500, 500,500, 450,500, 500,500, 450,550, 450,550, 500,450, 550,450, 500,500, 500,1450, 500,1450, 500,1500, 450,1500, 500,2950, 2950};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;

Master811129, 2018-08-02:
hello @AnalysIR  i have a question:
you said:  in NEC protocols the library reports 0xFFFFFFFF for repeat signals. These repeats are sent when the button is kept pressed down.

but i received 0xFFFFFFFF in the Sony protocol when i pushed and kept the button but is it nature for this? (the protocol was detected with the IRremote lib and i used the sony remote so it's actually a sony protocol)

AnalysIR, 2018-08-02:
Please open anew issue...to avoid needless confusion.

closed 2020-07-03 by ArminJo