Arduino-IRremote/Arduino-IRremote #577: Additional Remotes

a2304, 2018-06-22:
I am using a universal remote to operate an Insignia home audio system.  There are some functions that the universal remote is missing and I would like to use my Arduino to make up for those.  I have decoded a few commands (power, volume up, volume down, and input) and I have managed to send those from the Arduino to the home audio system.  Is there a way to figure out the codes for buttons not on the remote?  Rather than trial and error with 36^4 possible codes?

Remote: RCA Universal Remote
Home Audio System: Insignia NS-R5101HD
Protocol: NEC 32bit

Some codes that I know:

Power:               A1B7A05F
Volume Up:       A1B78679
Input:                A1B7EE11

AnalysIR, 2018-06-22:
You will note that A1B7 is constant for each signal...

the last 2 bytes e.g. A05F are the inverse of each other(A+5=F, 0+F=F etc)....so all you need is to try 256 combinations (less the ones you know already!)

a2304, 2018-06-22:
Awesome!  Thanks!

closed 2018-06-22 by a2304

SirNate0, 2020-09-01:
@a2304 I have the same receiver and was wondering if you could share a bit more about how this worked for you? I've tried it with sendNEC() with the provided codes and none of them have worked, so I was wondering if you could provide some more details.

a2304, 2020-09-01:
[HEX Codes.xlsx](https://github.com/z3t0/Arduino-IRremote/files/5153902/HEX.Codes.xlsx)

Here are the HEX codes that worked for my receiver.

a2304, 2020-09-01:
> @a2304 I have the same receiver and was wondering if you could share a bit more about how this worked for you? I've tried it with sendNEC() with the provided codes and none of them have worked, so I was wondering if you could provide some more details.

Here are the HEX codes that worked for my receiver.

[HEX Codes.xlsx](https://github.com/z3t0/Arduino-IRremote/files/5153904/HEX.Codes.xlsx)

AnalysIR, 2020-09-01:
e.g. to send as NEC...

irsendNEC(0x**A1B7**A05F,32);

_Note: The spread sheet doesn't have the first part 'A187'_