Arduino-IRremote/Arduino-IRremote #579: Power too low when sending IR codes

iamtheyammer, 2018-06-30:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** NEC (if any)

I'm trying to control my air conditioner with my arduino uno, and I'm having an interesting time. I've captured the off code for the air conditioner and the power code for my Samsung TV as a test.

When I send codes, they work. However, a phone camera shows that the IR LED is sending a low power codes. I have to hold the IR LED up to the TV for it to see it, and I think that the air conditioner can't see the IR Emitter because the power is too low.

Is there a way to up the broadcasting power?

**Code Block:**
[https://gist.github.com/iamtheyammer/a9eaa475fa89577647e57b20a01b0532](https://gist.github.com/iamtheyammer/a9eaa475fa89577647e57b20a01b0532)

AnalysIR, 2018-07-01:
Yes.

Search for a transistor based IR emitter circuit or driver

Also a slightly more advanced one is covered here:  [blog post](https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/)

iamtheyammer, 2018-07-01:
Today, I took off the plastic on the air conditioner and I put the IR Emitter right next to the receiver with no barriers and still, the command didn't work.

AnalysIR, 2018-07-02:
post the output of the irecvdump example....with a few signals

iamtheyammer, 2018-07-02:
https://pastebin.com/4kjDMtCZ

iamtheyammer, 2018-07-02:
I should mention that the remote sends two signals for power on/off.

iamtheyammer, 2018-07-02:
What following sketch?

I tried playing around with the gap length.
[Here's](https://pastebin.com/Bwp1pJ6E) a script I used to test the delay, and [here's](https://gist.githubusercontent.com/iamtheyammer/17867ee0c3dd9d3c7a494f811cc57d7e/raw/86836a7cfe155ae931335c80cff9fe62762fab5f/fromSerial.txt) output from that script.

AnalysIR, 2018-07-02:
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

...sorry :)

iamtheyammer, 2018-07-02:
Here's the output.
https://pastebin.com/3efGULEk

AnalysIR, 2018-07-02:
try this....

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : ELECTROLUX67_134AC :  : 2338010E9540200044670021F200000C08 : 134 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {9000,4500,600,1700,600,600,600,600,600,600,600,1700,600,1700,600,600,600,600,600,1700,600,1700,600,1700,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,600,600,600,600,600,600,600,600,1700,600,1700,600,1700,600,600,600,1700,600,600,600,600,600,1700,600,600,600,20028,600,1700,600,600,600,1700,600,600,600,1700,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,600,600,600,600,600,600,40036,9000,4500,600,1700,600,600,600,600,600,600,600,1700,600,1700,600,600,600,600,600,1700,600,1700,600,1700,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,600,600,600,600,600,600,600,600,1700,600,1700,600,1700,600,1700,600,1700,600,600,600,600,600,1700,600,600,600,20004,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,1700,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,600,600,600,600,600,600}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: ELECTROLUX67_134AC, Key:  
```

Note: I have cleaned the signal using AnalysIR.
[image]

iamtheyammer, 2018-07-02:
This code won't run: nothing is in a loop. Shall I put everything in a setup loop?

iamtheyammer, 2018-07-02:
Holy crap!
```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : ELECTROLUX67_134AC :  : 2338010E9540200044670021F200000C08 : 134 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/
#include <IRremote.h>
// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

IRsend irsend;

unsigned int Signal_0_0[] = {9000,4500,600,1700,600,600,600,600,600,600,600,1700,600,1700,600,600,600,600,600,1700,600,1700,600,1700,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,600,600,600,600,600,600,600,600,1700,600,1700,600,1700,600,600,600,1700,600,600,600,600,600,1700,600,600,600,20028,600,1700,600,600,600,1700,600,600,600,1700,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,600,600,600,600,600,600,40036,9000,4500,600,1700,600,600,600,600,600,600,600,1700,600,1700,600,600,600,600,600,1700,600,1700,600,1700,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,600,600,600,600,600,600,600,600,1700,600,1700,600,1700,600,1700,600,1700,600,600,600,600,600,1700,600,600,600,20004,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,1700,600,600,600,600,600,600,600,600,600,600,600,600,600,1700,600,600,600,600,600,600,600}; //AnalysIR Batch Export (IRremote) - RAW

void setup() {}

void loop() {
  irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
  delay(1000);
 // AnalysIR IR Protocol: ELECTROLUX67_134AC, Key: 
}
```
worked!

iamtheyammer, 2018-07-02:
I wrote a python script that does what you just did and it's working every time. It just needed some cleaning.
[The repository is available here with the script](https://github.com/iamtheyammer/ir-cleaner/)

closed 2018-07-02 by iamtheyammer