Arduino-IRremote/Arduino-IRremote #580: Help with multiple buttons for sony TV IR

ovi4, 2018-07-01:
Hi, Board Arduino UNO R3, 6 buttons + IR blaster all on breadboard.
The goal: emulate 6 of my sony TV IR remote via arduino setup.
My desired buttons are : Vol+, Vol-, Prog+, Prog-, On/Off, Mute.
So far: i've managed to extract the hex codes for each of the above buttons and created a code.
The problem: Only Power ON/OFF button works and nothing else. Any help is much appreciated Thank you
My level in "C" language bear minimum (an absolute newbie that tries hard to get by and learn on the go)
The setup is: pin 3 to IR blaster an pins 2, 4, 5, 6, 7, 8 go to my desired buttons.
My code (so far) is the following:
` // Include IR Remote Library by Shirriff
 #include <IRremote.h>

 // Define Vol+ as pin 2
 const int Vplus = 2;
  // Define Vol- as pin 4
 const int Vminus = 4;
  // Define Prog+ as pin 5
 const int Pplus = 5;
  // Define Prog- as pin 6
 const int Pminus = 6;
  // Define OnOff as pin 7
 const int OnOff = 7;
  // Define Mute as pin 8
 const int Mute = 8;

  // Define a variable for the initial buttons state
 int stateVplus = 0;
 int stateVminus = 0;
 int statePplus = 0;
 int statePminus = 0;
 int stateOnOff = 0;
 int stateMute = 0;

  // Create IR switch SEND Object (one object per switch)
 IRsend sendVplus;
 IRsend sendVminus;
 IRsend sendPplus;
 IRsend sendPminus;
 IRsend sendOnOff;
 IRsend sendMute;

void setup() {
 // Set Switch pins as Inputs
  pinMode(Vplus, INPUT);
  pinMode(Vminus, INPUT);  
  pinMode(Pplus, INPUT);
  pinMode(Pminus, INPUT);    
  pinMode(OnOff, INPUT);   
  pinMode(Mute, INPUT); 
   Serial.begin(9600);    
}

void loop() {
 // Set state of each button accordindigly
  stateVplus = digitalRead(Vplus);
  stateVminus = digitalRead(Vminus);
  statePplus = digitalRead(Pplus);
  statePminus = digitalRead(Pminus);
  stateOnOff = digitalRead(OnOff);
  stateMute = digitalRead(Mute);

 //If any button is pressed send code accordingly
  if (stateVplus == HIGH){
    for (int i = 0; i < 3; i++){
   sendVplus.sendSony(0x490, 12); //Vol+ Sony code
    }
  }
  if (stateVminus == HIGH){
    for (int i = 0; i < 3; i++){
   sendVminus.sendSony(0xC90, 12); //Vol- Sony code
    }
  }
  if (statePplus == HIGH){
    for (int i = 0; i < 3; i++){
   sendPplus.sendSony(0x90, 12); //Prog+ Sony code
    }
  }
  if (statePminus == HIGH){
    for (int i = 0; i < 3; i++){
   sendPminus.sendSony(0x890, 12); //Prog- Sony code
    }
  }
  if (stateOnOff == HIGH){
    for (int i = 0; i < 3; i++){
   sendOnOff.sendSony(0xA90, 12); //OnOff Sony code
    }
  }
  if (stateMute == HIGH){
    for (int i = 0; i < 3; i++){
   sendMute.sendSony(0x290, 12); //Mute Sony code
    }
  }
  // Add a small delay before repeating to bebounce
  delay(200);
}`

What I do know is that I have to send every code 3 times in a row for the TV to respond (according to SONY) What I don't know is how do I tell the loop to run 3 times @ every button press???
And where in the code Have I gone wrong that only Power (ON/OFF) button works???
Regarding the wiring of buttons every button has 1 leg to 5V+ and the other leg in series to a 10K ohm resistor to ground then the arduino pins are connected to the junction between res. and push button
And finally, after sorting out the issue in my code how do I write my code in a more compact and more elegant manner? It seems too repetitive, too long winded :)))

AnalysIR, 2018-07-01:
> What I don't know is how do I tell the loop to run 3 times @ every button press???

It looks like your for loops are doing this already!

```
// Creates IR switch SEND Object (one object per switch)
IRsend sendVplus;
IRsend sendVminus;
IRsend sendPplus;
IRsend sendPminus;
IRsend sendOnOff;
IRsend sendMute;
```
You should be able to use just one object to send every signal!

Might be worth using pullups on the input pins for the switches..

> [And finally, after sorting out the issue in my code how do I write my code in a more compact and more elegant manner? It seems too repetitive, too long winded]

Yes, indeed. I suggest you get it working with one button first. The progress to 2 buttons, using a switch statement to store the hex value in a variable. Then after the switch statement, just dend the sony signal as you already do in one of your for loops. You may need to study up on this first, which is the best way to learn & retain :)

Finally, you didnt place your code inside code tags, which are available in the menu above the edit box <>

Plese re-edit your original post to tidy it up.

ovi4, 2018-07-03:
Hi, tnx. for your help I've  now edited my post and changed the code and it works to an extend but still issues and,  came up with PART 2. see above .
Kind regards

AnalysIR, 2018-07-03:
Your attempt at using code tags does not seem to have worked. Try again and remember to use the preview option. It might help to insert the code and select it....then press the <> option for the code tags.
Use preview to veryify that it works.

Also, use the auto-format option in the Arduino IDE to format the code before pasting it in.

Otherwise it is harder to read your code or make sense of it. I didn't try to check your use of an encoder for that reason.

Some issues I noticed:
- The Serial prints inside the for loops will/may mess up the sending of signals.
- the delay(60) for the debounce, will also mess up the signals
- However, you will still need a delay between sending each of the 3 SONY signals in the for loop of approx 20 mSecs

In summary you need to restructure a bit more and consider my earlier suggestion of starting simpler and building it up slowly with more features as you get things working.

ovi4, 2018-07-04:
Hi, well.... I've had another attempt @ trying to correctly insert my codes in  the comment. Hopefully it worked this time.
On the other hand some days back  I've ordered (online) a book that hopefully will shed some light into my AVR programming skills. The book is called : Make: AVR Programming   Learning to Write Software for Hardware by Elliot Williams.  I hope I've made the right choice since this is gonna be the very first book I've ever purchased regarding programming micro-controllers. 
Thank you once again.

AnalysIR, 2018-07-04:
The code tags are correct now.

For learning Arduino coding, there are lots of resources on line & probably the best approach is to work thru the various examples supplied with most Arduino libraries.

AnalysIR, 2018-07-04:
the following may improve things for you (but it is untested)
```
void loop() {
  previousA = currentA;
  previousB = currentB;
  currentA = digitalRead(2);
  currentB = digitalRead(4);

  unsigned long signalValue = 0;
  if (previousA == LOW && previousB == LOW && currentA == HIGH && currentB == LOW) signalValue = 0x490;
  else if (previousA == HIGH && previousB == HIGH && currentA == LOW && currentB == HIGH)  signalValue = 0x490;
  else if (previousA == LOW && previousB == LOW && currentA == LOW && currentB == HIGH)  signalValue = 0xC90;
  else if (previousA == HIGH && previousB == HIGH && currentA == HIGH && currentB == LOW)  signalValue = 0xC90;

  if ( signalValue != 0) {
    for (int i = 0; i < 3; i++) {  // sends code 3 times
      irsend.sendSony(signalValue, 12);
      delay(20);
    }
    delay(100); // between bursts of triple signals
  }
}
```

The count variable didn't seem to have any purpose, so I removed it.
I also didnt look into the rotary encoder operation and noticed that you didn't set pinMode for pins 2 & 4.

ovi4, 2018-07-04:
Hi, thank you for the help with your Code. It helped indeed . the speed and responsiveness has now greatly improved. Is closer to the original IR responsiveness. However the range (the distance) is only about 1M or there about. I think that the IR LED that I'm using might not be the right one (is 850nm) i know that certain IR 's have IR  LED's that transmit over 920nm. Also, the 150 ohm resistor that is in series with the IR LED might need to be reduced down to 100 ohms or do I need some sort of 1 transistor amplification ? I really don't know.   Anyways, I have revised the code ( including some change in naming to try shorten some long winded words ) and added the extra (Mute) functionality to the SW button on the Rot. Enc. I've also changed the delay value to 10 milliseconds as response  seems a little smoother . as for the PinMode of 2 and 4 it was my mistake when copy/paste it the code to the comment window. I must have somehow erased it by mistake.
So, here is the code: 

```
#include <IRremote.h> // IR Library by Shirriff

// Define the rotary encoder pins
#define outA 2 //CLK
#define outB 4 //DT
#define sw 8 //SW for the MUTE function

// initial var. state
int stMute, count, precount, diffcount = 0;
byte currA, currB, preA, preB = 0;

IRsend irsend; // define object ?

void setup() {
  pinMode [(sw, outA, outB), INPUT];

  preA = currA = digitalRead(outA);
  preB = currB = digitalRead(outB);
}

void loop() {
  preA = currA;
  preB = currB;
  currA = digitalRead(outA);
  currB = digitalRead(outB);
  stMute = digitalRead(sw);

  unsigned long sigVal = 0;
  if (preA == LOW && preB == LOW && currA == HIGH && currB == LOW) sigVal = 0x490;
  else if (preA == HIGH && preB == HIGH && currA == LOW && currB == HIGH) sigVal = 0x490;
  else if (preA == LOW && preB == LOW && currA == LOW && currB == HIGH) sigVal = 0xC90;
  else if (preA == HIGH && preB == HIGH && currA == HIGH && currB == LOW) sigVal = 0xC90;
  else if (stMute == LOW) sigVal = 0x290;
  if ( sigVal != 0) {
    for (int i = 0; i < 3; i++) {  // sends code 3 times
      irsend.sendSony(sigVal, 12);
      delay(10);
    }
    delay(100); // between bursts of triple signals
  }
}

```

AnalysIR, 2018-07-05:
I think that the IR LED that I'm using might not be the right one (is 850nm).

Obviously that would not be good. Ususally with 850nm you can see a dim red/pink light. IR for remote control is usually specified @ 940nm or 944nm. TSAL6x00 from Vishay are good quality ones. (where x=1,2 or 4)

100R or even 68R should be OK when directly driven by a standard (5V) Arduino GPIO pin.
_(Putting 2 150R resistors in parallel to get 75R should also be fine)_

> do I need some sort of 1 transistor 

That would help a lot & you can find more information on that by searching.

closed 2020-10-17 by ArminJo