Arduino-IRremote/Arduino-IRremote #582: ESP32 Error - error: 'SEND_PIN' was not declared in this scope

mrmonteith, 2018-07-10:
**Board:** ESP32 NodeMCU
**Library Version:** 2.4.0b
**Protocol:** n/a

I'm using the example IRrecvDemo with pin set to 2;

 I'm getting an error:
 
     IRremote.h:365:23: error: 'SEND_PIN' was not declared in this scope

   const int sendPin = SEND_PIN;

    It doesn't complain about RECV_PIN.  

**checklist:**
- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [X] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [X] The title of the issue is helpful and relevant

AnalysIR, 2018-07-10:
> #define TIMER_DISABLE_PWM Serial.println("IRsend not implemented for ESP32 yet");

on line 58 of boarddefs.h...speaks for itself.

platform support is usually added when contributed by the community.

I believe the Rx should work OK. 

If you search thru the issues, you might(?) find some example of working code for ESP32.

bengtmartensson, 2018-07-10:
The support for ESP32 in the master branch is only partial (You did check that you had read the README.md thoroughly :-)), IIRC, there is more support for ESP32 in the branch called `beta`.

mrmonteith, 2018-07-10:
I noted that I'm using the 2.4.0b.  My only guess is that since support for ESP32 is receive only that it's getting kicked out that in the library send is still somewhat declared.   I've had another library that caused a similar error and I just declared a dummy reference to get rid of it.  But forgot what I did exactly in that instance.

AnalysIR, 2018-07-10:
Did you comment out all the send protocols in IRremote.h?

mrmonteith, 2018-07-10:
Not sure if this is the right way or not but it worked.  I changed the end of remote.h from:
`#else`
`     const int sendPin = SEND_PIN;`
`#endif`
`} ;`

to:
`#elif defined(SEND_PIN)`
`const int sendPin = SEND_PIN;`
`#endif`
`} ;`

 Hope that helps.  At least it compiles now without errors.

bmitov, 2018-08-17:
I did it with this change:
#elif defined(ESP32)
        // No system LED on ESP32, disable blinking by NOT defining BLINKLED

        // avr/interrupt.h is not present
#       undef HAS_AVR_INTERRUPT_H

        // Sending not implemented
#       undef SENDING_SUPPORTED
#		define SEND_PIN -1

        // Supply own enbleIRIn
#       undef USE_DEFAULT_ENABLE_IR_IN

This should be updated one way or another in the library. Please note that there was also a typo in the : 
#       undef SENDING_SUPPORTED

In the library it was : 
#       undef SENDING_SUPPORTED#
This is a typo, so I removed the extra # 
Please add at least this fix in the library so people can use it with the ESP32 ;-)

ArminJo, 2020-07-04:
ESP32 send is now supported.

closed 2020-07-04 by ArminJo

devilsnare007, 2020-07-19:
Thank you @mrmonteith!!! I implemented your fix and its working - doesn't help when the IDE says "Error compiling for board DOIT ESP32 DEVKIT V1." -- I reinstalled ESP32 drivers n all thinking it was an issue with that.... 
@ArminJo - you say ESP32 send is now supported but I was still having this issue on the latest version: 2.4.0 beta... 
I was trying to compile the following code:
```
#include <IRremote.h>
int receiver = 13;
IRrecv irrecv(receiver);
decode_results results;

void setup() {
  Serial.begin(115200);
  irrecv.enableIRIn();
}

void loop() {
  // put your main code here, to run repeatedly:
  if (irrecv.decode(&results)){
    Serial.println(results.value,HEX);
    irrecv.resume();
  }
```
I kept getting the following error:
`Arduino: 1.8.12 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None" In file included from sketch\IRremote.cpp:23:0: sketch\IRremote.h:365:23: error: 'SEND_PIN' was not declared in this scope const int sendPin = SEND_PIN; ^ In file included from sketch\esp32.cpp:5:0: sketch\IRremote.h:365:23: error: 'SEND_PIN' was not declared in this scope const int sendPin = SEND_PIN; ^ In file included from sketch\irSend.cpp:1:0: sketch\IRremote.h:365:23: error: 'SEND_PIN' was not declared in this scope const int sendPin = SEND_PIN; ^ In file included from sketch\irRecv.cpp:1:0: sketch\IRremote.h:365:23: error: 'SEND_PIN' was not declared in this scope const int sendPin = SEND_PIN; ^ In file included from sketch\ir_Aiwa.cpp:1:0: sketch\IRremote.h:365:23: error: 'SEND_PIN' was not declared in this scope const int sendPin = SEND_PIN; ^ exit status 1 Error compiling for board DOIT ESP32 DEVKIT V1. This report would have more information with "Show verbose output during compilation" option enabled in File -> Preferences.`

ArminJo, 2020-07-19:
Please try the new Version 2.5.0. If it still fails we must do a hotfix.

devilsnare007, 2020-07-19:
> Please try the new Version 2.5.0. If it still fails we must do a hotfix.

Tested with Version 2.5.0 and it works fine. Thanks for your quick reply @ArminJo