Arduino-IRremote/Arduino-IRremote #584: Compile time error: Multiple definition of '__vector_13' (using include NewPing.h of Tim Eckel)

jagganath69, 2018-07-20:
**Board:** ATMega2560
**Library Version:** 2.2.3

**Code Block:**
```c

#include <NewPing.h>
#include <IRremote.h> 

//------------------------------
Arduino: 1.8.4 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
libraries\z3t0-Arduino-IRremote-2dc97b2\IRremote.cpp.o (symbol from plugin): In function `MATCH(int, int)':
(.text+0x0): multiple definition of `__vector_13'
libraries\NewPing\NewPing.cpp.o (symbol from plugin):(.text+0x0): first defined here

d:/users/ray licuanan/appdata/local/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

//----------------------------

Tim Eckel's NewPing URL:
https://bitbucket.org/teckel12/arduino-new-ping/wiki/Home

//----------------------------

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [Y] I have **read** the README.md file thoroughly
- [Y] I have searched existing issues to see if there is anything I have missed.
- [Y] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [Y] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [Y] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

bengtmartensson, 2018-07-31:
Both of IRremote and NewPing are using the same timer. Since both are (obviously) defining their own interrupt service routine for that timer, it comes to a loading error, instead of much more difficult to debug run-time error. So you should be happy :-).

To fix, "tell" (at least) one of the libraries to use another timer. (The Mega 2560 has 5 timers IIRC: This is a lot.)

bengtmartensson, 2020-07-11:
Please close. Answered.

closed 2020-07-11 by jagganath69