Arduino-IRremote/Arduino-IRremote #585: Trouble using AnalysIR

Jackkt, 2018-07-27:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** LG

Hello, Im having trouble transferring the IR data I received from Arduino to AnalysIR. I tried using the examples in IRremote but to no avail. Is it supposed to pick up the IR signal automatically or am I missing something obvious?
The remote number is AKB74955603
IR receiver is 815114

Thanks

AnalysIR, 2018-07-27:
Yor query has been answered on our IRforum, as it is not really an issue with the library.

[IRforum response](https://irforum.analysir.com/viewtopic.php?f=16&t=603&p=1343#p1343)

mrmonteith, 2018-07-27:
The forum looks like a good place.  Is it possible to have it added to the Readme on here?  That way they have another avenue.  I didin't know about it myself either.  Thanks