Arduino-IRremote/Arduino-IRremote #587: IRremote sees phantom codes (repeats of last key)

gitterman, 2018-08-18:
**Board:** ARDUINO Nano
**Library Version:** 2.2.3

I want to build a simple IR controlled power plug which should be able to learn a single code from some remote and than toggle a relay if it sees this code.
For this I use the IRremote library.
Everything works so far but sometimes the receiver sees the latest code again after a few seconds.
This toggles my relay in error.
I have already added a HOLDOFF delay of 2000ms after I see the matching IR code but still sometimes the IR receiver sees the matching code again even after these 2s.
The transmitter isn't the problem as I covered the sending IR LED after sending a signal but the problem still occurred.
The code can be found at https://forum.arduino.cc/index.php?action=dlattach;topic=564141.0;attach=270726
Is there a way to clear/initialize the IR receiver between codes or did I do somenthing wrong?

AnalysIR, 2018-08-18:
There is a lot going on in your code....you seem to have taken over Timer0 which is used by the arduino system...so maybe your millis() etc are not working as expected.??? 

I suggest you start off with a simple receive example provided with the library to establish your IR stuff is working correctly as expected and only then start adding in anything else you need one thing at a time. That way you will know what if anything is causing an issue.

sineir, 2018-08-18:
try this.
after reading the code received, insert the line below;
"irrecv.resume(); // Receive the next value"

this clears the buffer.

gitterman, 2018-08-18:
@sineir 
 there is an irrecv.resume in my code in the readIRinput routine and well before the 2s holdoff.

@AnalysIR 
 thanks for pointing me in that direction.
According to https://learn.adafruit.com/multi-tasking-the-arduino-part-2/timers it is fine to double use Time0. 
Nevertheless I have made a stripped down version of the code which has no interrupts or pushbuttons which can be found here:
https://forum.arduino.cc/index.php?action=dlattach;topic=564141.0;attach=270822
I still see the spurious double toggling every other time.

gitterman, 2018-08-18:
Here is an output of the serial monitor where I pressed the match key exactly 2 times and always covered the sending IR LED after sending the signal to assure there is no double triggering.
On the 1st press there was no double toggling.
After the 2nd press the output showed the phantom toggle.
starting program
 RC5  irCode=0x281           => output ON
281
0
ir HOLDOFF ended
 RC5  irCode=0x281             => 2nd press, output OFF
A81
0
ir HOLDOFF ended
 RC5  irCode=0x281        => spurious event without any keypress, output ON
A81
0
ir HOLDOFF ended
switch off output            => output ON timeout after 20s

AnalysIR, 2018-08-18:
> According to https://learn.adafruit.com/multi-tasking-the-arduino-part-2/timers it is fine to double use Time0.

In some circumstances this may be true, but unless you have good AVR expertise...I would advise not using timer0. 

> I suggest you start off with a simple receive example provided with the library to establish your IR stuff is working correctly as expected

It sounds like you havent done this yet! If you follow this approach, you will be more successful with your own code.

gitterman, 2018-08-19:
Of course I had started off with the examples from the library before I built my complex code.
I had even looked into other libraries like irlib and irlib2 but as my final goal is to build it on an ATtiny85 I chose IRremote.
As I said I had taken out any timer, button and eeprom stuff and still saw the problem.
Today I tested with a NEC remote and did not see the phantom effect during > 5minutes of testing.
This makes me believe that there is something in the RC5 part which causes the problem.

NEC is not an option for the final solution though as almost all of my remotes are RC5 and I want to build a more universal power plug which could be used with almost any remote.

gitterman, 2018-08-19:
Meanwhile I checked the issue with IRlib2 and found the same behaviour on RC5 codes but not on NEC codes.
I observed from the examples that codes are transmitted from the remotes quite fast.
Are there possibly internal buffers in the RCx ibrary which already store a second received code and then decode it if irrecv.resume is called ?

AnalysIR, 2018-08-19:
I am assuming, you got it working as expected using the examples in the library.

RC5 includes a toggle bit which 'toggles' everytime you press a button on the remote. So the returned value may have a different bit set/unset on every other button press. (The value may be different for every other press of the same button)

Alternatively, If you are getting multiple signals decoded for every button press, then simply add some 'de-bouncing' into your code. Some remotes will send multiple signals for every button press.

So if you put a delay before the resume, it may eliminate this impact.Or simply keep a timestamp of when you got the last signal and ignore any more within 1 second of the last received.

gitterman, 2018-08-19:
Thanks, doing it the resume **only** after the HOLDOFF did the trick.

closed 2018-08-19 by gitterman