Arduino-IRremote/Arduino-IRremote #590: IR reception on ESP32 is incompatible to EEPROM library

johannesmaibaum, 2018-09-10:
Hi, thanks for this library which I've been using for a long time on Arduino!

Reading that this library claims to support IR reception on the ESP32, I wanted to try this. All worked fine until I added the EEPROM library into the mix. The minimal example sketch posted below crashes as soon as you start using the EEPROM after IR reception has been enabled (which in my code happens as soon as an IR message comes in).

The crash output on the Serial Monitor is the following for this example sketch:

```
ets Jun  8 2016 00:22:57
16:46:10.288 -> 
16:46:10.288 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:46:10.288 -> configsip: 0, SPIWP:0xee
16:46:10.288 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:46:10.321 -> mode:DIO, clock div:1
16:46:10.321 -> load:0x3fff0018,len:4
16:46:10.321 -> load:0x3fff001c,len:952
16:46:10.321 -> load:0x40078000,len:6084
16:46:10.321 -> load:0x40080000,len:7936
16:46:10.321 -> entry 0x40080310
Enabling IR reception
Received value: f12dfa05
16:46:14.559 -> Save to EEPROM... Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
16:46:14.559 -> Core 1 register dump:
16:46:14.559 -> PC      : 0x400d0d30  PS      : 0x00060034  A0      : 0x400815dc  A1      : 0x3ffc0be0  
16:46:14.559 -> A2      : 0x00000001  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x40086c8c  
16:46:14.592 -> A6      : 0x3ffae270  A7      : 0x00000005  A8      : 0x80081100  A9      : 0x3ff5f024  
16:46:14.592 -> A10     : 0x3ffc1024  A11     : 0x20000000  A12     : 0x00000400  A13     : 0x00000002  
16:46:14.592 -> A14     : 0x3ff42000  A15     : 0x700000eb  SAR     : 0x00000011  EXCCAUSE: 0x00000007  
16:46:14.592 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  
16:46:14.625 -> Core 1 was running in ISR context:
16:46:14.625 -> EPC1    : 0x40089b2a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d0d30
16:46:14.625 -> 
16:46:14.625 -> Backtrace: 0x400d0d30:0x3ffc0be0 0x400815d9:0x3ffc0c00 0x40089b27:0x00000000
16:46:14.625 -> 
16:46:14.625 -> Rebooting...
```

Skimming through the ESP-IDF docs about this panic, I found [this resource](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/fatal-errors.html?highlight=cache%20disabled), which basically says that all interrupt handlers (and all non-local data accessed from within interrupt handlers) on the ESP32 have to be put into IRAM/DRAM in order to not crash the system when data is been written to EEPROM.

I already tried to put `IRAM_ATTR` and `DRAM_ATTR` in some places (`IRAM_ATTR` added to the definition of https://github.com/z3t0/Arduino-IRremote/blob/2dc97b29f2da74e342a308f6e99f153e132ce06a/IRremote.cpp#L127 , and `DRAM_ATTR` to https://github.com/z3t0/Arduino-IRremote/blob/2dc97b29f2da74e342a308f6e99f153e132ce06a/IRremoteInt.h#L69), but without success so far.

Perhaps someone knowing the internals of this library better than me knows how to handle this properly?

**Board:** ESP32 Dev Module
**Library Version:** master
**Protocol:** NEC

**Code Block:**
```c++
#include <EEPROM.h>
#include <IRremote.h>

IRrecv irrecv(32);
decode_results irdecr;

void setup() {
  Serial.begin(115200);
  if (!EEPROM.begin(64)) {
    Serial.println("Error initializing EEPROM library, rebooting");
    delay(1000);
    ESP.restart();
  }

  Serial.println("Enabling IR reception");
  irrecv.enableIRIn();
}

void loop() {
  if (irrecv.decode(&irdecr)) {
    int32_t last_value = irdecr.value;
    Serial.printf("Received value: %x\n", last_value);
    Serial.print("Save to EEPROM... ");
    EEPROM.put(0, last_value);
    EEPROM.commit();  // BOOM!
    Serial.println("done.");
    irrecv.resume();
  }
}
```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [x] I have **read** the README.md file thoroughly.
- [x] I have searched existing issues to see if there is anything I have missed.
- [ ] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used: _I'm using the master branch because of_ #582.
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

Aircoookie, 2018-11-21:
I experience the same issue. `EEPROM.commit()` crashes and corrupts the contents of the flash/EEPROM (but not in all cases, strangely) and even disabling all interrupts doesn't help. If I have time later, I'll try to write a PR which adds `irrecv.disableIRIn()`. You'd just need to call that before EEPROM.commit() to disable the interrupt. Not sure if it'd work, nevertheless this would be a great addition to this library. For some reason, the ESP8266 fork of this library plays along nicely with EEPROM, but doesn't work for ESP32, of course.

Aircoookie, 2018-11-21:
This is probably related to https://github.com/espressif/arduino-esp32/issues/1141

1uuc4asb, 2018-11-21:
Didn't worked...
It stills congestioning all the time after receive any data...
I think I'll have to live with this lol

bengtmartensson, 2018-11-21:
I am not sure if this helps you, but IIRC there is considerably more support for ESP32 in the (Git) branch called `beta` (not to be confused with a release with beta in its name), as well as a number of pull requests and issues.

skrodahl, 2018-12-07:
There's a solution here:
https://github.com/espressif/arduino-esp32/issues/928

The instructions weren't correct, so I posted a working solution here:
https://hackaday.io/project/46280-muffsy-stereo-relay-input-selector/log/156747-full-functionality-rotational-encoder-and-remote-control

Aircoookie, 2018-12-07:
Thank you very much, @skrodahl ! I will check later if your solution also works for my project!

commit e8161d10 (2020-07-04): added the IRAM_ATTR and disableIRIn() for ESP32, closes #590
closed 2020-07-04 by ArminJo