Arduino-IRremote/Arduino-IRremote #592: JVC irsend arduino ide 1.6.13 *works* vs 1.8.1 *does not work*

Pimmetje, 2018-10-09:
**Board:** ARDUINO UNO
**Library Version:** 2.3.3
**Protocol:** JVC)

More or less like (not tested this particular piece of code, so it's reference only)
**Code Block:**
```c
#include <IRremote.h> 
IRsend irSend;
irSend->sendJVC(0xC5B8, 16,0); // hex value, 16 bits, no repeat
delayMicroseconds(50); // see http://www.sbprojects.com/knowledge/ir/jvc.php for information
irSend->sendJVC(0xC5B8, 16,1); // hex value, 16 bits, repeat
```
I could not find if this is a known issue. Else i would like to bring it on to attention that some arduino versions could break IR codes (i switched back to 1.6.13 for this project). Ill add the latest 1.8 and report back. 
I still have to test the latest 1.8 but that one is not yet available on my CI server. I hope i can add it somewhere this week or in the weekend.

AnalysIR, 2018-10-09:
I may be missing something, but what is the actual issue?

Maybe you are referring to Library versions and/or Arduino IDE versions - which is not clear from your post.

It looks like it is implemented as above on the latest 2.2.3 library version. (There will always be features not supported in older versions of the library??)

The code you posted above compiles OK on Arduino IDE 1.8.5. with the latest version of library installed. If you are using an older version of the IDE, make sure you have installed the latest version of the library for that IDE. (or is that your issue??)

Pimmetje, 2018-10-09:
Sorry if i was unclear. 
The code compiles on both IDE's but when compiled with 1.8.1 it does not work. If compiled with 1.6.13 it works as expected (same code).

AnalysIR, 2018-10-09:
OK.

That sounds weird. I don't have that test setup to try it out here at present.

Maybe someone else can try it out on IDE 1.8.1

Pimmetje, 2018-10-09:
Is there anything i can supply to help?

closed 2020-07-04 by ArminJo