Arduino-IRremote/Arduino-IRremote #595: Problem with signal size received - Raw Signals

1uuc4asb, 2018-10-31:
**Board:** ARDUINO UNO, MEGA, NANO and ESP-32
**Library Version:** 2.1.0

I went into a project that used the Arduino Uno. I'll not go into details, but we used the TSOP 4838 to receive **raw signals** and send them to air conditioning.
Worked well for a time but our project has grown and we had to move to ESP-32, but we didn't have him yet, so we used the Arduino Mega.
In Uno we were receiving a signal with 199 elements of size. 
In Mega we were receiving signals with 99 elements ( that didn't work  ) and sometimes with 199 elements ( that work ).
Our tsop broke so we bought a similar, that was receiving only signals with 99 elements.
We decided to get the TS4838 instead of a similar and the behavior was the same.
Switched back to Uno and the behavior was the same of the Mega: receiving 199 sometimes and 99 almost ever.
(Tested on nano too, received only 99)
In ESP we received all types of sizes from 30 to 160, but never 199.
None of these signals worked.
Someone have any ideia of what can be wrong?

**Code Block:**
```c

#include <IRremote.h> 

USING EXAMPLE SKETCH SendRawDemo AND irrecvDUMPv2

.....

```

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant

AnalysIR, 2018-10-31:
First, I would suggest TSOP34438.

When I see 99 elements I immediately think of the RAWBUF limit...so make sure to check how that is configured in your IDE for IRremote.
However, some ACs, send some signals at one length and other signals at another length.. maybe something like OFF might be shorter!.

Are you using this version of the librayr or another fork? (IRremote-ESP8266???)

Make sure you use 5V on the UNO & Mega to power teh receiver & 3V3 on the ESP!

You could also try this sketch [Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)
You may need to adjust it for ESP and Mega platforms.

AnalysIR, 2018-10-31:
Forgot to mention...make sure to verify the receiver pinout against the data sheet...as it can vary between models.

1uuc4asb, 2018-11-01:
@AnalysIR 

- [x] I had changed already the RAWBUF limit to 256.
- [x] All ACs signals that i was receiving had size 199 (those that worked at least ). 
PS: Once i got all signals from 17 to 30, turn on and turn off. All worked. All had 199 elements.
- [x] Using version 2.1.0. No fork.

Okay, 3V3 on ESP. This i didn't know.
I'll try the other sketch too.
I'll report any changes.

1uuc4asb, 2018-11-01:
Ok. I did what you say.
Used 3v3 on ESP and i got a relatively interesting result.
Now the ESP behaves like the others boards.
99-elements signals only.
(and they don't work)

But it is an advance,
Should i get a TSOP34438 ?

AnalysIR, 2018-11-01:
> Should i get a TSOP34438 ? I can't see the difference between him and the TSOP4838...

Try the sketch I linked first &  post the output. Some receivers will not accept longer AC signals, as they are designed for shorter TV style signals. I suggest you also stick with only one platform(UNO) until you get that working reliably.

> Any chance that RAWBUF's value is getting reassigned to it's default (99) or is truncating somehow?

Unlikely. _(unless its not actually set in the first place , plus  the OP is also getting signals 199 long.)_

1uuc4asb, 2018-11-19:
@AnalysIR The sketch worked like a charm. Solved my problem, thanks!

closed 2018-11-19 by 1uuc4asb