Arduino-IRremote/Arduino-IRremote #596: Multi-tasking on Arduino Uno: 'RECV_PIN' is not a Type

Xbox180, 2018-11-08:
**Board:** ARDUINO UNO
**Library Version:** 2.2.3
**Protocol:** Sony (if any)

**Code Block:**
https://gist.github.com/Xbox180/fc8db2c94ed74197d3009b316b0a4b8a

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant 

So before anyone asks, yes I did change TIMER2 to TIMER1 so it doesn't interfere with pitches.h
I'm expanding off of the Adafruit guide: [Multi-tasking the Arduino](https://learn.adafruit.com/multi-tasking-the-arduino-part-1/overview) for a class and one of the things that I wanted to do is make parts of it controlled by IR, like changing how the lights flash and stuff like that. The only problem is that it keeps erroring out at the line I marked on the Gist.

Xbox180, 2018-11-09:
I forgot to add that the demo code does work for me.

AnalysIR, 2018-11-09:
_(I originally made this response on the gist, by mistake)_

look at line 95 ...but not your main issue.

Sorry, I don't have that pitches library installed, but it is likely the way you are defining the IRremote IRrecv instance within your own class and setting the pin number.

closed 2020-07-04 by ArminJo