Arduino-IRremote/Arduino-IRremote #601: Can I make a remote?

williamhmic, 2019-01-14:
Is it possible to make an actual remote with this? I want to be able to sort of make my own remote for a philips TV because.. Why not? :P Kind of just for practice. Can I use the code for this? Thank you and sorry if I am doing something wrong.

AnalysIR, 2019-01-14:
Yes :)

Work thru the examples first and you should get a better idea.

z3t0, 2019-04-04:
@williamhmic that's an excellent idea, and as @AnalysIR has mentioned it is entirely possibly.

I hope you have fun trying different encodings! Feel free to share the results of your project :smile:

z3t0, 2019-04-04:
Closing for now, if you feel that this was done in error please do not hesitate to re-open the issue.

closed 2019-04-04 by z3t0