Arduino-IRremote/Arduino-IRremote #604: Issue with multiple definition of `__vector_13'

Oku1243, 2019-01-26:
**Board:** ARDUINO MEGA2560
**Library Version:** 2.2.3

it juste a little issue but i have  a error message taht say : 

 Arduino : 1.8.8 (Windows 10), Carte : "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':

(.text+0x0): multiple definition of `__vector_13'

libraries\Arduino-IRremote-master\IRremote.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

exit status 1
Erreur de compilation pour la carte Arduino/Genuino Mega or Mega 2560

AnalysIR, 2019-01-27:
Difficult to comment without seeing your code!

Are you using any interrupts in your code that could be conflicting with IRremote?

Try to compile some of the examples provided.

Oku1243, 2019-01-28:
this is my command

Oku1243, 2019-01-28:
#include <Wire.h> // bibliothéque pour le I2C
#include <LiquidCrystal_I2C.h> // Pour l'ecrand Lcd avec I2C
#include <IRremote.h>
#include <dht.h>
#define DHT11_PIN 30
#define Recept_infra 9
#define ledRouge 3 // Constante pour la broche 3
#define ledVert 5 // Constante pour la broche 5
#define ledBleu 6 // Constante pour la broche 6
#define Son 31
dht DHT;

int pir = 33;               // choose the input pin (for PIR sensor)
int pirState = LOW;             // we start, assuming no motion detected
int val = 0;
int pwmRouge=0;
int pwmBleu=0;
int pwmVert=0;
int Led=0;
int Time=0;
int Lcd=1;
int Charg=2;

IRrecv Toto(Recept_infra);
decode_results message;
LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4 ,5, 6, 7, 3, POSITIVE); // déclare l'affiltation des broche
const int LCD_NB_ROWS = 2;
const int LCD_NB_COLUMNS = 16;

/*----------------------------Caractères personnalisés-------------------------------------------*/
byte START_DIV_0_OF_1[8] = {
  B01111, 
  B11000,
  B10000,
  B10000,
  B10000,
  B10000,
  B11000,
  B01111
}; // Char début 0 / 1

byte START_DIV_1_OF_1[8] = {
  B01111, 
  B11000,
  B10011,
  B10111,
  B10111,
  B10011,
  B11000,
  B01111
}; // Char début 1 / 1

byte DIV_0_OF_2[8] = {
  B11111, 
  B00000,
  B00000,
  B00000,
  B00000,
  B00000,
  B00000,
  B11111
}; // Char milieu 0 / 2

byte DIV_1_OF_2[8] = {
  B11111, 
  B00000,
  B11000,
  B11000,
  B11000,
  B11000,
[... 511 lines cut ...]
      analogWrite(ledRouge, pwmRouge);
      analogWrite(ledVert, pwmVert);
      analogWrite(ledBleu, pwmBleu);
}

void Task_DHT11 () {
  if (Lcd==1){
      lcd.scrollDisplayLeft();
      delay(500);
      }
      if (Lcd==0){
        int chk = DHT.read11(DHT11_PIN);
        lcd.setCursor(0,0); 
           lcd.print("Temp: ");
           lcd.print(DHT.temperature);
           lcd.print((char)223);
           lcd.print("C");
           lcd.setCursor(0,1);
           lcd.print("Humidity: ");
           lcd.print(DHT.humidity);
           lcd.print("%");
           Time=0;
        delay(1200);
      }
}

Oku1243, 2019-01-28:
it work if i dont use the command tone

AnalysIR, 2019-01-28:
Sounds like a timer conflict...but not obvious..

Try changing the pin you are using to receive IR. try 10 different spare pins around the Arduino

AnalysIR, 2019-01-28:
..and..please put the code you posted inside code tags "<>2 in the edit menu

ArminJo, 2020-07-02:
Try other IR timers. You can change it [here](https://github.com/z3t0/Arduino-IRremote/blob/8f665d60f065c139bf641a152518b078bec84437/src/private/IRremoteBoardDefs.h#L186)

closed 2020-07-02 by ArminJo