Arduino-IRremote/Arduino-IRremote #605: FIOS Remote

jn3va, 2019-01-29:
I'm trying to decode a FIOS remote (VZ P265v5).  I'm getting similar problems as other "different code for same button" type issues I see in other threads.  I'm getting a different/better IR receiver as suggested and looking into other tips.  In the meantime, does anyone know of any references about these remote e.g. info about what protocol these remotes use?  or has already decoded this?
Thanks!

AnalysIR, 2019-01-30:
OK there is an issue there...

possible culprits:
- bad battery in remote
- bad IR receiver (what model(link?) are you using
- Try holding the remote 1m(1 yard) away from the receiver and point it directly at the receiver and see if the captures are any better.
- Suspect settings change in IRremote

Another thing to try is to correct the captures above as follows:
- remove all the 12 values.
- change the remaining values to the nearest of 9266; 790; 1938 and 4176 and try it out!

You could also try recording using the following sketch from our blog:
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

If all else fails==>Did you make any changes to thedefault library settings(_I suspect so_). Either way try removing & reinstalling the IRremote library  and record the signals again after restarting the IDE.

jn3va, 2019-01-31:
Batteries are good, brand new, but I replaced them just to check.  Same result.  

No changes to the code.  I recently pulled a new version but I can try to pull a fresh one down this weekend to be 100% sure.   

Same results when trying from 1m away.  

The IR Receiver is a Velleman 1838
https://www.microcenter.com/product/476345/velleman-1838-ir-infrared-379khz-receiver---2-pack

I have 2 of these and tried swapping, but no change.  

I replaced the values as you described below (except 1) and sent this but it did not work.

uint16_t rawData[39] = {9266, 4176, 790,  4176, 790,  4176, 790,  4176, 790,  4176, 790,  4176, 790,  1938,  790,  1938, 790,  1938, 790,  1938, 790,  1938, 790,  1938, 790,  1938, 790,  1938, 790,  1938, 790,  1938,  790, 1938, 790, 30478,  9266, 1939,  790};  // UNKNOWN 16B7557B

There was one pulse that was 5 digits: 30478  In each of the new reads captures the numbers were close 30478, 30456, 30476 and in the same position so I left this one in thinking there must be a reason with this remote for the longer pulse.    Do you think I should replace this with a different value also? e.g. 9266?  

I'll also give the AC Long recording link you sent a try this weekend.

Really appreciate the help!
Thanks!
Joe

AnalysIR, 2019-02-02:
Great News :)

It might be useful if you uploaded a set of working signals, given how difficult it was to find some.

Once you do that, please close the issue.

jn3va, 2019-02-03:
Excellent idea.  

I am not decoding the whole remote just specific keys.  For this project I just need some basic functions so we can control the playback when cooking, eating and need to pause, mute etc, or are not close to don't want to touch a remote.  We'd still use the remote for more complex tasks like changing the channel.  e.g Alexa pause on, Alexa pause off, Alexa fast forward on, alexa skip back on etc.

The buttons I have decoded are: Play, Pause, Fast Forward, Skip Forward, Skip Backward

The values you provided (494, 2210, 4446, 8866, 41418) seem to work great, all of these worked first try.  

For anyone else learning this, below is the process I used:

1) Use the IrecvDumpV2.ino sketch to capture code.  Capture at least 3 sets.   

e.g.
Captured:
uint16_t rawData[47] = {9276, 12, 4138, 816, 12, 4138, 790, 1932, 794, 4176, 776, 12, 4174, 788, 4176, 786, 1958, 762, 12, 1932, 782, 12, 1930, 792, 1958, 768, 1936, 790, 1936, 786, 12, 1930, 790, 1934, 792, 4180, 788, 1934, 790, 1958, 768, 30454, 9256, 12, 1912, 790, 12}; // UNKNOWN EAB04EBC

uint16_t rawData[51] = {9260, 4164, 790, 12, 4164, 788, 12, 1920, 784, 12, 4174, 790, 12, 4166, 784, 4178, 780, 12, 1954, 768, 1938, 790, 1936, 774, 14, 1958, 756, 14, 1936, 792, 12, 1926, 776, 12, 1940, 788, 12, 1924, 782, 12, 4174, 784, 1940, 788, 1958, 768, 30450, 9258, 1928, 790, 12}; // UNKNOWN 6F74CED0

uint16_t rawData[42] = {9262, 4164, 818, 4148, 790, 14, 1920, 784, 12, 4172, 788, 4178, 786, 4198, 766, 1958, 768, 1938, 788, 1958, 768, 1940, 788, 1938, 786, 1962, 768, 1938, 788, 1960, 756, 12, 4178, 786, 1958, 768, 1958, 770, 30470, 9234, 1930, 788 }; // UNKNOWN 3BA20A45

Pull these into a text editor, set the font to courier and no wrap.  It will be easier to check and understand.

2) Clean up the values by deleting the 12s and 14s and removing extra spaces.  You might have different random values to clean up.  e.g. other "small" values - I think I had a few 51s  54s etc. in the sequences.  This is one reason to capture multiple sequences and compare.

Cleaned up:
uint16_t rawData[47] = {9276, 4138, **816**, 4138, 790, 1932, 794, 4176, 776, 4174, 788, 4176, 786, 1958, 762, 1932, 782, 1930, 792, 1958, 768, 1936, 790, 1936, 786, 1930, 790, 1934, 792, 4180, 788, 1934, 790, 1958, 768, 30454, 9256, 1979, 790}; // UNKNOWN EAB04EBC

uint16_t rawData[51] = {9260, 4164, **790**, 4164, 788, 1920, 784, 4174, 790, 4166, 784, 4178, 780, 1954, 768, 1938, 790, 1936, 774, 1958, 756, 1936, 792, 1926, 776, 1940, 788, 1924, 782, 4174, 784, 1940, 788, 1958, 768, 30450, 9258, 1928, 790}; // UNKNOWN 6F74CED0

uint16_t rawData[42] = {9262, 4164, **818**, 4148, 790, 1920, 784, 4172, 788, 4178, 786, 4198, 766, 1958, 768, 1938, 788, 1958, 768, 1940, 788, 1938, 786, 1962, 768, 1938, 788, 1960, 756, 4178, 786, 1958, 768, 1958, 770, 30470, 9234, 1930, 788}; // UNKNOWN 3BA20A45

3) Use the 3 values in each column of each set to compare to make sure you have some kind of correlation in the numbers so when you are replacing vales in step 4 you know you are matching correctly.  
e.g. In the 3rd value of the sets above, the numbers are 816, 790 and 818.  If you had 816, 790 and 4148 you'd know that something may have been missed and you might ignore the last set.
Also, a couple of times I had numbers like 1901 and if I had not had 3 captures to compare I might have replaced 1901 with 2210 instead of 494 for the 3rd pulse (see step 4)

4) Replace the values with the closest to these values: 494, 2210, 4446, 8866, 41418

Result after replacement: 
uint16_t rawData[39] = {8866, 4446, 494, 4446, 494, 2210, 494, 4446, 494, 4446, 494, 4446, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 4446, 494, 2210, 494, 2210, 494, 41418, 8866, 2210, 494}; // Fast Forward

5) If you are using the array in code, count the number of sets and replace the array index value.  In these captures, after cleanup there are 39 numbers in each set, so change uint16_t rawData[42] to uint16_t rawData[39]

**The codes**
Pause
uint16_t rawData[39] = {8866, 4446, 494, 4446, 494, 4446, 494, 4446, 494, 4446, 494, 4446, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 41418, 8866, 2210, 494}; // Pause

Play
uint16_t rawData[39] = {8866, 4446, 494, 4446, 494, 4446, 494, 2210, 494, 4446, 494, 4446, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 4446, 494, 2210, 494, 41418, 8866, 2210, 494}; // Play

Fast Forward
uint16_t rawData[39] = {8866, 4446, 494, 4446, 494, 2210, 494, 4446, 494, 4446, 494, 4446, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 4446, 494, 2210, 494, 2210, 494, 41418, 8866, 2210, 494}; // Fast Forward

Skip Backward
uint16_t rawData[39] = {8866, 4446, 494, 2210, 494, 2210, 494, 4446, 494, 4446, 494, 4446, 494, 4446, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 4446, 494, 2210, 494, 2210, 494, 2210, 494, 41418, 8866, 2210, 494}; // Skip Backward

Skip Forward
uint16_t rawData[39] = {8866, 4446, 494, 4446, 494, 4446, 494, 4446, 494, 4446, 494, 4446, 494, 4446, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 2210, 494, 4446, 494, 4446, 494, 4446, 494, 41418, 8866, 2210, 494}; // Skip Forward

closed 2019-02-03 by jn3va