Arduino-IRremote/Arduino-IRremote #606: Issue with receiving and sending raw to LG air conditioner

iamohcinefrikh, 2019-02-14:
**Board:** Arduino UNO
**Library Version:** 2.1.0
**Protocol:** SANYO

**Issue:** 
I used **_IRrecvDumpV2.uno_** to get the power button signal from the remote, and this is what i've got. 
The issue is when i used the **_IRsendRawDemo.uno_** to send this raw signal, i didn't get any response from the A/C.

**Additional informations:**
**A/C:**  LG KS-H1264NA5
**IR Receiver:** VS1838B

**IRrecvDumpV2:**
```c
13:42:17.520 -> Encoding  : SANYO
13:42:17.520 -> Code      : FFFFFFFF (0 bits)
13:42:17.555 -> Timing[57]: 
13:42:17.555 ->      + 450, -1550     + 500, - 550     + 500, - 500     + 500, - 600
13:42:17.659 ->      + 450, -1550     + 500, - 500     + 500, - 600     + 450, - 550
13:42:17.695 ->      + 500, -1600     + 450, -1600     + 450, - 550     + 450, - 550
13:42:17.800 ->      + 500, - 600     + 450, - 500     + 550, - 550     + 450, - 550
13:42:17.872 ->      + 500, - 550     + 450, - 550     + 450, - 600     + 450, - 550
13:42:17.939 ->      + 450, - 550     + 500, -1600     + 450, - 550     + 500, -1550
13:42:18.010 ->      + 500, - 500     + 500, - 600     + 450, - 550     + 450, -1600
13:42:18.083 ->      + 450
13:42:18.083 -> unsigned int  rawData[57] = {450,1550, 500,550, 500,500, 500,600, 450,1550, 500,500, 500,600, 450,550, 500,1600, 450,1600, 450,550, 450,550, 500,600, 450,500, 550,550, 450,550, 500,550, 450,550, 450,600, 450,550, 450,550, 500,1600, 450,550, 500,1550, 500,500, 500,600, 450,550, 450,1600, 450};  // SANYO FFFFFFFF
13:42:18.408 -> unsigned int  data = 0xFFFFFFFF;
13:42:18.443 -> 
13:42:19.850 -> Encoding  : SANYO
13:42:19.884 -> Code      : FFFFFFFF (0 bits)
13:42:19.918 -> Timing[57]: 
13:42:19.918 ->      + 450, -1600     + 500, - 550     + 500, - 550     + 500, - 500
13:42:19.990 ->      + 500, -1550     + 500, - 500     + 500, - 550     + 500, - 600
13:42:20.060 ->      + 450, - 500     + 500, - 600     + 450, - 550     + 450, - 550
13:42:20.134 ->      + 500, - 550     + 450, - 600     + 450, - 550     + 500, - 550
13:42:20.207 ->      + 450, - 600     + 450, -1600     + 450, - 550     + 450, -1600
13:42:20.279 ->      + 450, - 600     + 450, -1600     + 450, - 550     + 450, - 600
13:42:20.350 ->      + 450, -1550     + 500, - 600     + 450, - 550     + 450, -1600
13:42:20.424 ->      + 450
13:42:20.461 -> unsigned int  rawData[57] = {450,1600, 500,550, 500,550, 500,500, 500,1550, 500,500, 500,550, 500,600, 450,500, 500,600, 450,550, 450,550, 500,550, 450,600, 450,550, 500,550, 450,600, 450,1600, 450,550, 450,1600, 450,600, 450,1600, 450,550, 450,600, 450,1550, 500,600, 450,550, 450,1600, 450};  // SANYO FFFFFFFF
13:42:20.779 -> unsigned int  data = 0xFFFFFFFF;
```

**IRsendRawDemo:**

```c
#include <IRremote.h>

IRsend irsend;

void setup()
{

}

void loop() {
  int khz = 38; // 38kHz carrier frequency for the NEC protocol
  unsigned int  irSignal[57] = {550,1600, 450,500, 500,600, 450,550, 450,1600, 450,600, 450,500, 500,550, 500,500, 500,600, 450,550, 450,600, 450,550, 450,550, 500,550, 450,600, 450,550, 500,1600, 450,500, 500,1600, 500,550, 450,1550, 500,550, 500,550, 500,1550, 450,600, 450,600, 450,1550, 500};
  
  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.

  delay(5000); //In this example, the signal will be repeated every 5 seconds, approximately.
}
```

**checklist:**
- [✔] I have **read** the README.md file thoroughly.
- [✔] I have searched existing issues to see if there is anything I have missed.
- [✔] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used.
- [✔] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [✔] The title of the issue is helpful and relevant .

AnalysIR, 2019-02-14:
It looks like you are only getting a partial capture for some reason? It is possible you are just missing the headers for some reason? Other than the missing headers, the timings are in the ball park for some LG AC units.

Make sure you point the remote directly at the receiver about 1m(1yd) away.

FYI: That particular IR receiver is very poor (the worst in my opinion) and causes lots of support issues for users. See the post on our blog about it. A better choice would be from Vishay eg TSOP34438.

Try the following sketch to capture the signals:
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

closed 2020-07-02 by ArminJo

ArminJo, 2021-01-17:
With the missing Header prepended it is the LG code for the power button with the intended parity error, see #755