Arduino-IRremote/Arduino-IRremote #607: ATMEGA328PB support

garageeks, 2019-02-14:
The recently introduced ATMEGA328PB chip added, among other things, two new 16-bit timers called TC3 and TC4 as well as 4 more PWM pins for a total of 10.

These are the output pins available:
PB1 pin - OC1A Timer/Counter1 Output Compare Match A Output
PB2 pin - OC1B Timer/Counter1 Output Compare Match B Output
PB3 pin - OC2A Timer/Counter2 Output Compare Match A Output
PD0 pin - OC3A Timer/Counter3 Output Compare Match A Output
PD1 pin - OC4A Timer/Counter4 Output Compare Match A Output
PD2 pin - OC3B Timer/Counter3 Output Compare Match B Output
               OC4B Timer/Counter4 Output Compare Match B Output
PD3 pin - OC2B Timer/Counter2 Output Compare Match B Output
PD5 pin - OC0B Timer/Counter0 Output Compare Match B Output
PD6 pin - OC0A Timer/Counter0 Output Compare Match A Output

In boarddefs.h I see there is already provisioning for Timer 3 and 4 (IR_USE_TIMER3 and IR_USE_TIMER4) however it's unlikely a theoretical Arduino Uno with 328PB matches with the MEGA pins specified.

On Arduino Uno you are using Timer 2 on Pin 3, which is PD3 and OC2B. Alternatively, you can use Timer 1 on Pin 9, which corresponds to PB1 and OC1A.

I'd like to help, but I don't know much about these timers. Can we use either Match A Output or Match B Output or are there reasons on why you settle specifically on Pin 3 or 9?

On 328PB we should be able to use at least Pin 2, and, if we use the secondary UART in place of the primary one, we can set up also Pin 0 and 1.

The chip costs the same as 328P and is more flexible - I discovered too late, when I produced the PCB, that you can't use any PWM pin with IRremote :-)

AnalysIR, 2019-02-14:
The library tries to stick with standard Arduino platforms, where possible and relies on contributors to add support for newer platforms. (not everyone will have access to a standard 328PB Arduino. platform for testing) The limitation on pins is more related to AVRs & timers, which is reflected then in the IRremote library. If the avr supported pwm on every pin, I suspect support would be added. There is a map of output oins supported in the README, which is always worth checking first.

There was some activity on  a softPWM implementation a while ago, but I couldnt find it easily and am not sure if  it made it into the library proper.

If you cant find it, we have a few examples of softPWM & uPWM on our blog (assumes 16MHz oscillator).
softPWM or simple PWM can be output on any pin for IR, uPWM uses the UART's XCK pin, but can also be made to work with the UARTs Tx pin. _(see examples for other platforms)_

[search here using PWM & uPWM for more info --->>>](https://www.analysir.com/blog/sitemap_analysir_posts/)

Shaheers, 2019-09-30:
hi i also have problem with IR library
i just added the following line in known_16bit_timers.h on line no 104
#elif defined (__AVR_ATmega168__) || defined (__AVR_ATmega328P__) || defined (__AVR_ATmega328__) ||  defined (__AVR_ATmega8__) 
i just added the Atmega328pb value at end so he can recognise it, just replace the below line with above line.

#elif defined (__AVR_ATmega168__) || defined (__AVR_ATmega328P__) || defined (__AVR_ATmega328__) ||  defined (__AVR_ATmega8__) || defined (__AVR_ATmega328PB__)

AnalysIR, 2019-10-01:
@Shaheers 
_Note: I think the words 'below' & 'above' should be switched in your post.???_

FYI: I have been using the Atmega328PB for a while now with the IRremote library. (I don't recall having to make any changes to get it working?...or maybe I have just forgotten...)

When I checked, I couldn't find a file named "**known_16bit_timers.h**" in the library???

closed 2020-07-04 by ArminJo