Arduino-IRremote/Arduino-IRremote #608: AT&T U-verse Remote Support

gudenau, 2019-02-16:
**Board:** Teensy 3.2
**Library Version:** 2.2.3
**Protocol:** AT&T U-verse

[gist](https://gist.github.com/gudenau/3dab23ba656df23cea330410a4ba8bb4)

When using the AT&T U-verse remote commands it spits out "UNKNOWN", for example:
```
Unknown
Address: -1
Values:  1376725544
Bits:    32
Raw:
	1796 9 5 4 5 4 11 4 5 4 15 4 8 4 4 5 5 4 5 4 11 4 11 4 8 5 11 4 5 4 14 5 5 3 5 4 
```

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2019-02-16:
Its not clear what the question is?

Howeer, thise timings seem off. Also try IRrecvdumpv2 example.

It might help to  post your receiver model # (& link)

I also found this via google.....

> The U-Verse boxes use a Philips RCMM code clocked at 36.023 kHz.

RCMM can be a challenge even for the best receivers! ...and IRremote would not work very well decoding it.

Your best option may be to us use RAW captures only as in the example mentioned above and try to get a good 36kHz receiver.

gudenau, 2019-02-17:
```
Encoding  : UNKNOWN
Code      : 6B6AC24F (32 bits)
Timing[11]: 
     + 400, -2900     + 200, - 400     + 200, -3700     + 200, - 550
     + 200, -1650     + 200
unsigned int  rawData[11] = {400,2900, 200,400, 200,3700, 200,550, 200,1650, 200};  // UNKNOWN 6B6AC24F

Encoding  : UNKNOWN
Code      : 2BACFC33 (32 bits)
Timing[9]: 
     + 400, -3450     + 200, -3750     + 200, - 550     + 200, -1600
     + 200
unsigned int  rawData[9] = {400,3450, 200,3750, 200,550, 200,1600, 200};  // UNKNOWN 2BACFC33

Encoding  : UNKNOWN
Code      : 8917D545 (32 bits)
Timing[35]: 
     + 450, - 250     + 200, - 200     + 250, - 550     + 250, - 200
     + 250, - 650     + 250, - 400     + 250, - 200     + 250, - 150
     + 250, - 200     + 250, - 550     + 250, - 500     + 200, - 450
     + 200, - 550     + 200, - 550     + 250, - 200     + 250, - 200
     + 250, - 650     + 250
unsigned int  rawData[35] = {450,250, 200,200, 250,550, 250,200, 250,650, 250,400, 250,200, 250,150, 250,200, 250,550, 250,500, 200,450, 200,550, 200,550, 250,200, 250,200, 250,650, 250};  // UNKNOWN 8917D545

Encoding  : UNKNOWN
Code      : A090BF58 (32 bits)
Timing[35]: 
     + 500, - 200     + 250, - 200     + 250, - 500     + 300, - 150
     + 250, - 700     + 250, - 350     + 250, - 200     + 250, - 200
     + 250, - 200     + 250, - 500     + 250, - 550     + 250, - 350
     + 200, - 550     + 250, - 350     + 250, - 350     + 250, - 400
     + 250, - 350     + 250
unsigned int  rawData[35] = {500,200, 250,200, 250,500, 300,150, 250,700, 250,350, 250,200, 250,200, 250,200, 250,500, 250,550, 250,350, 200,550, 250,350, 250,350, 250,400, 250,350, 250};  // UNKNOWN A090BF58

Encoding  : UNKNOWN
Code      : 69DBBC3D (32 bits)
Timing[7]: 
     + 450, -2850     + 200, -4300     + 200, -2550     + 200
unsigned int  rawData[7] = {450,2850, 200,4300, 200,2550, 200};  // UNKNOWN 69DBBC3D

Encoding  : UNKNOWN
Code      : D6A38EBD (32 bits)
Timing[21]: 
     + 400, -2900     + 200, - 400     + 200, -1900     + 200, - 550
     + 200, - 400     + 200, - 600     + 200, - 400     + 200, - 400
     + 200, - 400     + 200, - 400     + 200
unsigned int  rawData[21] = {400,2900, 200,400, 200,1900, 200,550, 200,400, 200,600, 200,400, 200,400, 200,400, 200,400, 200};  // UNKNOWN D6A38EBD

Encoding  : UNKNOWN
Code      : 9388E43E (32 bits)
Timing[17]: 
     + 400, -1500     + 200, -1750     + 200, -1900     + 200, - 600
     + 200, - 400     + 200, - 550     + 200, -1650     + 200, - 400
     + 200
unsigned int  rawData[17] = {400,1500, 200,1750, 200,1900, 200,600, 200,400, 200,550, 200,1650, 200,400, 200};  // UNKNOWN 9388E43E
```
Sorry, had some issues uploading from a different computer.

AnalysIR, 2019-02-18:
You didn't post the model/link of you IR receiver?

The timings are better, but still not good enough to decode.

Try the following sketch (you may have to adjust for teensy) which should provide more granular timings.
[sketch =>](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

PS: You can also adjust the granularity of the library from the fdefault of 50uSecs, but it may not help much with this particular protocol.

gudenau, 2019-02-23:
Again, sorry about the delays and forgetting the model thing.

It is a `ARRIS VIP2500 / 405`, not sure how I can modify `AnalysIR` for  Teensy pin 15.

AnalysIR, 2019-02-24:
Sorry, I meant the model number of your IR Receiver, not the STB...

However, try the following for the POWER button (note RCMM makes use of a toggle, so a single signal per button is not sufficient). The carrier frequency is indeed 36kHz.

/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : RCMM :  : 2340260C : 32 : 36
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=36; //NB Change this default value as neccessary to the correct carrier frequency

uint16_t Signal_0_0[] = {417,278,167,278,167,611,167,278,167,778,167,444,167,278,167,278,167,278,167,278,167,611,167,444,167,611,167,278,167,278,167,778,167,278,167,90468}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: RCMM, Key:

AnalysIR, 2019-02-24:
I believe the teensy is supported, so the example for sendRAW should work. You should also get a good 36kHz IR receiver and use the sketch I linked above.

gudenau, 2019-02-24:
Ahh, it's a [TSOP38238 from Adafruit](https://www.adafruit.com/product/157).

And I don't have the L.E.D. driver thing setup yet. I will try and get that done soonish.

AnalysIR, 2019-02-24:
> And I don't have the L.E.D. driver thing setup yet. I will try and get that done soonish.

It is not clear what this refers to

A good IR receiver would be TSOPxxx36  (as long as it ends in 36, it should;d be better than what you have.

closed 2020-08-17 by ArminJo