Arduino-IRremote/Arduino-IRremote #610: Timer(?) issues with ATTiny84

tgherzog, 2019-02-21:
**Board:** ATTiny84
**Library Version:** 2.1.0
**Protocol:** n/a

**Code Block:**
```c
#include <IRremote.h> // Include the IRremote library

const int RECV_PIN = 3, NOD_PIN = 0;

IRrecv irrecv(RECV_PIN);
decode_results results; // This will store our IR received codes

void nod(int, int);  // user feedback

void setup()
{
  irrecv.enableIRIn(); // Start the receiver

  pinMode(NOD_PIN, OUTPUT);
  digitalWrite(NOD_PIN, LOW);
  nod(3, 1);
}

void loop() {
}

void nod(int n, int len) {

  int i;
  for(i=0;i<n;i++) {
      digitalWrite(NOD_PIN, HIGH);
      delay(len * 250);
      digitalWrite(NOD_PIN, LOW);
      delay(250);
  }
}

```

I'm having some very basic issues just getting the IR library to configure as expected. I'm using an AVR programmer to program the ATTiny84, so physical pins 4, 7, 8, and 9 are connected to RESET, MOSI, MISO and SCK respectively. The ATTiny84 is driven by a 16mHz external crystal, and I'm using [these board definitions](https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json) (which BTW are different than ones I used several years ago, and the logical pin mappings are also different than what I remember - possible clue).

As as you might guess there is an LED attached to pin 0 (physical pin 13).

Anyway, as you can see, this sketch doesn't do much, doesn't even try to send or receive IR data. I would expect the LED to blink 3 times and do nothing else.

Instead, the LED turns on solid and stays that way.  UNLESS I comment out this line:

```
//  irrecv.enableIRIn(); // Start the receiver
```

in which case it behaves as expected.

I've tried lots of different combinations of pin assignments for both the LED and the IRrecv object but get the same results.

I'm really at a loss. I suspect there's an incompatibility with either the programmer or the board definition. Something that would cause the chip to flip out when `enableIRin()` turns on the timer. But then again, I get the same results if I disconnect the programmer.

I feel like I'm missing something obvious. Appreciate any suggestions, and sorry if this is a massive pilot error.

AnalysIR, 2019-02-21:
Obvious candidates are timer conflicts (IRremote uses Timer1), Arduino core issues and error in FUSES. However, as this is not a common platform relative ot standard Arduinos, you are likely going to have to troubleshoot it yourself.

Have you verified the Arduino definitions match the physical pins on the IC?

try this to eliminate or otherwise the delay() function. ..avoids using delay() function
```
#define F_CPU 16000000 //16MHz
#include <util/delay.h>

#include <IRremote.h> // Include the IRremote library
const int RECV_PIN = 3, NOD_PIN = 0;
IRrecv irrecv(RECV_PIN);
decode_results results; // This will store our IR received codes
void nod(int, int);  // user feedback
void setup()
{
  irrecv.enableIRIn(); // Start the receiver
  pinMode(NOD_PIN, OUTPUT);
  digitalWrite(NOD_PIN, LOW);
  nod(3, 1);
}
void loop() {
}

void nod(int n, int len) {
  int i;
  for(i=0;i<n;i++) {
      digitalWrite(NOD_PIN, HIGH);
      delay_ms(250);
      digitalWrite(NOD_PIN, LOW);
      delay_ms(250);
  }
```
 irrecv.enableIRIn(); jsut enable the timer..& its interrupt...nothing else.

Other things to try is to use different pins for receiver and LED. It is usually OK to reuse MOSI, MISO and SCK for other purposes _(Obviously they shouldnt be be shorted HIGH/LOW externally)_, as the programmer will capture them during reset for programming purposes.  The Reset pin should be used only for Reset.

tgherzog, 2019-02-22:
It was indeed a core compatibility issue. I switched to [Spence Konde's](https://github.com/SpenceKonde/ATTinyCore) version (which it turns out I prefer anyway) and problem solved.

closed 2019-02-22 by tgherzog