Arduino-IRremote/Arduino-IRremote #611: any plan to support microbit?

kenken64, 2019-02-23:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Sony (if any)

**Code Block:**
```c

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2019-02-23:
I am not aware of any mention of support. But I just checked and see it is possible to run microbit with the Arduino IDE.

Usually, additional/new platforms are added by community contributors who have access to a platform like the microbit.

So I guess the conclusion is: You would most likely have to add in support for this platform, yourself....sorry.

Yours is the first time I recall any mention of microbit with IRremote!

However, it is entirely possible that there is already a fork of IRremote that does support it or the NRF MCU at least.

kenken64, 2019-03-12:
https://learn.adafruit.com/use-micro-bit-with-arduino/install-board-and-blink

kenken64, 2019-03-12:
Is there any tutorial or guideline for me to help you to port it to microbit?

AnalysIR, 2019-03-12:
I am not aware of any tutorial per-se.

However, if you mimic the changes made/#define(s) for Teensy 3.x in boarddefs.h...it should be pretty straightforward to do. The adafruit link above should also help.

Obviously the cahnges ou make will have to match the characteristics of the new platform.

pmalasp, 2019-10-29:
I've just written a few lines for it .... first simple tests are successful.

It uses Timer2 so you cannot use the Adafruit_Microbit display driver

[IRRemote_NRF5.zip](https://github.com/z3t0/Arduino-IRremote/files/3785598/IRRemote_NRF5.zip)

ArminJo, 2020-07-04:
@pmalasp Can you fork the repository, add your code and open a pull request?

nevercopter, 2020-08-19:
I'm getting errors trying to compile my sketch for micro:bit when using the solution provided by @pmalasp.

If the file provided used as is (i.e. replacing the original IRremoteBoardDefs.h file), Arduino says 'IR_SEND_PIN' was not declared in IRremote.h.
Changing either BoardDefs or IRemote in order to have it declared makes multiple other errors appear. What am I missing?

ArminJo, 2020-08-19:
@nevercopter Please try the latest repo version. It should support micro:bit now.

nevercopter, 2020-08-20:
> @nevercopter Please try the latest repo version. It should support micro:bit now.

Armin, thank you. Indeed, the latest master version seems to support NRF5 microbit. However BoardDefs.h, line 330 has a comment that sending is not implemeted. Unfortunately I was short-sighted enough to choose microbit boards for my home IR control system including transmitters.

I'd also want to report that I can't get Arduino IDE compile even receive scenarios from the example folder if any NRF5 board is selected (works fine for Arduino boards though). I lack experience to tell as to why that's happening, but I feel it's worth mentioning in case any of the contributors are interested in further polishing of the library. I can provide details if anyone's interested.

I don't think I can contribute in terms of coding since I'm a total rookie, but I have access to several microbit boards and can help testing future versions if it can be of any use of course.

ArminJo, 2020-08-20:
What board do you have selected in Arduino IDE and what are the last few error messages.

nevercopter, 2020-08-20:
The selected board is "BBC micro:bit" under "Nordic Semiconductor nRF5 Boards".
Errors differ depending on the script I'm trying to compile.

- For IRrecvDemo it would be:

> D:\Documents\Arduino\libraries\Arduino-IRremote-master\src\irSend.cpp: In member function 'void IRsend::enableIROut(int)':
> D:\Documents\Arduino\libraries\Arduino-IRremote-master\src\irSend.cpp:176:5: error: 'TIMER_DISABLE_RECEIVE_INTR' was not declared in this scope
>      TIMER_DISABLE_RECEIVE_INTR; //Timer2 Overflow Interrupt
>      ^
> D:\Documents\Arduino\libraries\Arduino-IRremote-master\src\irSend.cpp:182:27: error: 'timerConfigForSend' was not declared in this scope
>      timerConfigForSend(khz);
> 

- This is what I get if I try using any send commands (sendRaw for instance, but the result is somewhat similar with NEC and Sony):

> sketch\IRsendRawDemo.ino.cpp.o: In function `loop':
> D:\Documents\Arduino\libraries\Arduino-IRremote-master\examples\IRsendRawDemo/IRsendRawDemo.ino:64: undefined reference to `IRsend::sendRaw_P(unsigned int const*, unsigned int, unsigned int)'
> collect2.exe: error: ld returned 1 exit status
>

ArminJo, 2020-08-20:
You must delete all versions of the IRRemote library and then install the repo download.zip manually.
Then it will compile without an error (just tested).

nevercopter, 2020-08-20:
Thanks, the receive scripts are now working. I've already done clean installs before, but for some reason it only worked now. The only difference is that I wiped the whole library folder this time.

As to the sending capability, I understand it is not possible to use IRRemote with microbit for this purpose yet.

ArminJo, 2020-08-20:
Maybe you find a C example code I can use to implement sending. Maybe i some fork or elsewhere. 
I do not have any BBC:micro at hand and have no deep knowlege of it.

tomaculum, 2020-08-27:
Would it be possible to use this for other nrf5 boards as the Arduino Nano 33 BLE (sense)?

When I try to compile an empty file with just the include header, I get the following error.

``` ino
#include <IRremote.h>

void setup() {
}

void loop() {
}
```

```
In file included from /Users/user/Documents/Arduino/libraries/Arduino-IRremote-master/src/private/IRremoteInt.h:27:0,
                 from /Users/user/Documents/Arduino/libraries/Arduino-IRremote-master/src/IRremote.h:28,
                 from /Users/user/Documents/Arduino/test/test.ino:1:
/Users/tom/Documents/Arduino/libraries/Arduino-IRremote-master/src/private/IRremoteBoardDefs.h:367:2: error: #error SYSCLOCK cannot be determined. Define it for your board in IRremoteBoardDefs.h.
 #error SYSCLOCK cannot be determined. Define it for your board in IRremoteBoardDefs.h.
  ^~~~~
exit status 1
```

I tried to find the equivalent of `defined(NRF5)` for my Nano 33 BLE and found `nano33ble.build.variant=ARDUINO_NANO33BLE` inside a boards.txt file. 

Replacing every occurrence in `IRremoteBoardDefs.h` with  `defined(NRF5) || defined(ARDUINO_NANO33BLE)` did not work either.

```
In file included from /Users/user/Documents/Arduino/libraries/Arduino-IRremote-master/src/private/IRremoteInt.h:27:0,
                 from /Users/user/Documents/Arduino/libraries/Arduino-IRremote-master/src/IRremote.h:28,
                 from /Users/user/Documents/Arduino/test/test.ino:1:
/Users/tom/Documents/Arduino/libraries/Arduino-IRremote-master/src/private/IRremoteBoardDefs.h:328:23: error: token "�" is not valid in preprocessor expressions
 #elif defined(NRF5) || defined(ARDUINO_NANO33BLE) // nRF5 BBC MicroBit
                       ^
/Users/user/Documents/Arduino/libraries/Arduino-IRremote-master/src/private/IRremoteBoardDefs.h:367:2: error: #error SYSCLOCK cannot be determined. Define it for your board in IRremoteBoardDefs.h.
 #error SYSCLOCK cannot be determined. Define it for your board in IRremoteBoardDefs.h.
  ^~~~~
exit status 1
```

ArminJo, 2020-08-29:
@tomaculum Try the [IRMP library](https://github.com/ukw100/IRMP), it should work for Nano 33 BLE.

closed 2020-09-07 by ArminJo