Arduino-IRremote/Arduino-IRremote #612: Decode Velux WLR 100 50 02

lSh4dowl, 2019-02-26:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Unknown

Hey, I am trying to replicate the protocol of a Velux WLR 100 50 02, is there any preexisting stuff that I could use ? 
Or how could I go about this ?
It is supposed to be sent by an esp 8266 in the end.

I didn't find anything helpful on the internet, therefore I am asking here.
Thanks in advance.

AnalysIR, 2019-02-26:
Yes try out the examples provided with IRremote. 

After that you will be better placed 

You may also want to check it the esp8266 fork of the library

lSh4dowl, 2019-02-27:
Oh yeah I forgot to mention, I tested before and It isn't any known protocol.
it is kind of a rolling code, if that makes sense. 
Does it even make sense trying to decode this or am I better off trying to install a 8266 inside the control box, bypassing their infrared system ?
I don't know IR very well, but this is what results in me pressing the "up" Button 10 Times:

Received unknown code, saving as raw
 m350 s1350 m300 s1350 m1200 s500 m1150 s500 m1200 s500 m1150 s500 m350 s1350 m350 s1350 m300 s1350 m300 s1400 m1150 s500 m350 s1350 m300 s1350 m350 s1350 m300 s1400 m300 s1400 m300 s1350 m350 s1350 m300 s1400 m300 s1350 m300 s1400 m300 s1350 m1150 s500 m350
Received unknown code, saving as raw
 m350 s1300 m350 s1350 m1150 s500 m1200 s500 m1200 s450 m1200 s500 m350 s1350 m300 s1400 m300 s1350 m350 s1350 m1150 s500 m350 s1300 m350 s1350 m350 s1350 m300 s1350 m350 s1350 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m1200 s500 m350
Received unknown code, saving as raw
 m350 s1350 m350 s1300 m1200 s500 m1150 s500 m1200 s500 m1200 s450 m350 s1350 m350 s1300 m350 s1350 m350 s1350 m1150 s500 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m350 s1400 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m1150 s500 m350
Received unknown code, saving as raw
 m350 s1300 m350 s1350 m1150 s500 m1200 s500 m1200 s450 m1200 s500 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m1200 s500 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m350 s1350 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m1200 s500 m350
Received unknown code, saving as raw
 m350 s1350 m350 s1300 m1200 s500 m1200 s450 m1200 s500 m1200 s500 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m1200 s450 m350 s1350 m350 s1300 m350 s1400 m300 s1350 m300 s1400 m350 s1300 m350 s1350 m350 s1350 m300 s1350 m350 s1350 m350 s1300 m1200 s500 m350
Received unknown code, saving as raw
 m350 s1300 m350 s1350 m1200 s450 m1200 s500 m1200 s450 m1200 s500 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m1200 s450 m350 s1350 m350 s1350 m350 s1300 m350 s1350 m350 s1350 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m1200 s500 m350
Received unknown code, saving as raw
 m350 s1350 m350 s1300 m1200 s500 m1200 s450 m1250 s450 m1200 s500 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m1200 s450 m350 s1350 m350 s1350 m350 s1300 m350 s1350 m350 s1350 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m1200 s500 m350
Received unknown code, saving as raw
 m350 s1350 m350 s1350 m1150 s500 m1200 s500 m1200 s450 m1200 s500 m350 s1350 m300 s1350 m350 s1350 m350 s1300 m1200 s500 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m350 s1350 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m1200 s500 m350
Received unknown code, saving as raw
 m350 s1350 m300 s1350 m1200 s450 m1200 s500 m1200 s500 m1150 s500 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m1200 s500 m350 s1300 m350 s1350 m350 s1350 m300 s1350 m350 s1350 m350 s1350 m350 s1300 m350 s1350 m350 s1350 m350 s1300 m350 s1350 m1150 s500 m350
Received unknown code, saving as raw
 m350 s1300 m350 s1350 m1150 s500 m1200 s500 m1200 s450 m1200 s500 m350 s1300 m350 s1350 m350 s1350 m300 s1350 m1200 s500 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m350 s1350 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m350 s1350 m350 s1300 m1200 s500 m350

AnalysIR, 2019-02-27:
Please edit your last post and place the signal inside code tags "<>"

I checked a few of the signals above and they al look the same (timing will always vary slightly)

So try out the sendRawDemo example to send using the following:

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History: TSOP34438 RECEIVER
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : RAW :  :  : 0 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

uint16_t Signal_0_0[] = {350,1300,350,1350,1150,500,1200,500,1200,450,1200,500,350,1300,350,1350,350,1350,300,1350,1200,500,350,1300,350,1350,350,1300,350,1350,350,1350,350,1350,350,1300,350,1350,350,1300,350,1350,350,1300,1200,500,350}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: RAW, Key:  

```

lSh4dowl, 2019-02-27:
Ok so I tired what you gave me with a slight modification, because it seemed odd to me:
```
#include <IRrecv.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <IRtimer.h>
#include <IRutils.h>
#include <ir_Argo.h>
#include <ir_Coolix.h>
#include <ir_Daikin.h>
#include <ir_Fujitsu.h>
#include <ir_Gree.h>
#include <ir_Haier.h>
#include <ir_Hitachi.h>
#include <ir_Kelvinator.h>
#include <ir_LG.h>
#include <ir_Magiquest.h>
#include <ir_Midea.h>
#include <ir_Mitsubishi.h>
#include <ir_NEC.h>
#include <ir_Panasonic.h>
#include <ir_Samsung.h>
#include <ir_Toshiba.h>
#include <ir_Trotec.h>
#include <ir_Vestel.h>
#include <ir_Whirlpool.h>

/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History: TSOP34438 RECEIVER
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : RAW :  :  : 0 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

IRsend irsend(4);

uint16_t Signal_0_0[] = {350,1300,350,1350,1150,500,1200,500,1200,450,1200,500,350,1300,350,1350,350,1350,300,1350,1200,500,350,1300,350,1350,350,1300,350,1350,350,1350,350,1350,350,1300,350,1350,350,1300,350,1350,350,1300,1200,500,350}; //AnalysIR Batch Export (IRremote) - RAW

void setup() {
  irsend.begin();
  irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/ sizeof(Signal_0_0[0]), khz); //AnalysIR Batch Export (IRremote) - RAW
  // AnalysIR IR Protocol: RAW, Key:
}

void loop() {

}

where I modified --- sizeof(Signal_0_0)/sizeof(Signal_0_0[0])
```

This didnt work, but upon just spamming the code a whole lot, I eventually performed 2 unwanted commands on other shutters. I am genreally a total noob to IR, why does is say s and m in this example readout sketch for each value ? And why do I seem to get the same Hex Code everytime ? 😄 
Thanks for helping me btw ^^. rly appreciate it.

AnalysIR, 2019-02-27:
You should only need one include statement in your sketch...so make sure you have installed the library correctly & follow the example provided for sendraw.

To get the RAW data use the example irrecvDumpV2

s= Space
m=Mark

The V2 example will give better details to use.

lSh4dowl, 2019-02-27:
loading this sketch and pressing the button 3 times gives me this:

```
IRrecvDumpV2 is now running and waiting for IR input on Pin 14
Timestamp : 000029.543
Encoding  : UNKNOWN
Code      : 6ABB2F08 (48 bits)
Library   : v2.5.5

Raw Timing[95]:
   +   454, -  1242,    +   448, -  1242,    +  1302, -   394,    +   454, -  1240, 
   +   456, -  1238,    +  1300, -   394,    +   454, -  1240,    +   454, -  1242, 
   +   450, -  1242,    +   454, -  1238,    +  1276, -   450,    +   400, -  1264, 
   +   450, -  1244,    +   454, -  1240,    +   452, -  1242,    +   452, -  1286, 
   +   454, -  1240,    +   456, -  1240,    +   426, -  1268,    +   452, -  1238, 
   +  1304, -   422,    +   428, -  1236,    +   458, -  1236,    +   458, - 26522, 
   +   456, -  1240,    +   456, -  1234,    +  1304, -   422,    +   400, -  1266, 
   +   456, -  1236,    +  1304, -   420,    +   402, -  1264,    +   430, -  1266, 
   +   456, -  1238,    +   458, -  1234,    +  1304, -   420,    +   424, -  1240, 
   +   456, -  1238,    +   456, -  1240,    +   456, -  1240,    +   452, -  1284, 
   +   432, -  1262,    +   456, -  1238,    +   458, -  1238,    +   456, -  1234, 
   +  1300, -   426,    +   426, -  1238,    +   458, -  1236,    +   428

uint16_t rawData[95] = {454, 1242,  448, 1242,  1302, 394,  454, 1240,  456, 1238,  1300, 394,  454, 1240,  454, 1242,  450, 1242,  454, 1238,  1276, 450,  400, 1264,  450, 1244,  454, 1240,  452, 1242,  452, 1286,  454, 1240,  456, 1240,  426, 1268,  452, 1238,  1304, 422,  428, 1236,  458, 1236,  458, 26522,  456, 1240,  456, 1234,  1304, 422,  400, 1266,  456, 1236,  1304, 420,  402, 1264,  430, 1266,  456, 1238,  458, 1234,  1304, 420,  424, 1240,  456, 1238,  456, 1240,  456, 1240,  452, 1284,  432, 1262,  456, 1238,  458, 1238,  456, 1234,  1300, 426,  426, 1238,  458, 1236,  428};  // UNKNOWN 6ABB2F08

Timestamp : 000034.543
Encoding  : UNKNOWN
Code      : 6ABB2F08 (48 bits)
Library   : v2.5.5

Raw Timing[95]:
   +   432, -  1262,    +   456, -  1234,    +  1312, -   414,    +   434, -  1228, 
   +   436, -  1256,    +  1308, -   418,    +   432, -  1232,    +   460, -  1236, 
   +   464, -  1228,    +   464, -  1228,    +  1314, -   410,    +   436, -  1228, 
   +   490, -  1206,    +   472, -  1222,    +   466, -  1230,    +   490, -  1248, 
   +   464, -  1230,    +   464, -  1230,    +   464, -  1230,    +   488, -  1204, 
   +  1310, -   416,    +   432, -  1234,    +   460, -  1232,    +   494, - 26490, 
   +   456, -  1240,    +   452, -  1238,    +  1300, -   396,    +   452, -  1240, 
   +   454, -  1240,    +  1300, -   396,    +   450, -  1242,    +   452, -  1242, 
   +   456, -  1240,    +   454, -  1240,    +  1302, -   392,    +   456, -  1238, 
   +   454, -  1242,    +   454, -  1242,    +   454, -  1238,    +   456, -  1284, 
   +   452, -  1240,    +   454, -  1240,    +   454, -  1240,    +   456, -  1236, 
   +  1300, -   396,    +   452, -  1242,    +   454, -  1240,    +   458

uint16_t rawData[95] = {432, 1262,  456, 1234,  1312, 414,  434, 1228,  436, 1256,  1308, 418,  432, 1232,  460, 1236,  464, 1228,  464, 1228,  1314, 410,  436, 1228,  490, 1206,  472, 1222,  466, 1230,  490, 1248,  464, 1230,  464, 1230,  464, 1230,  488, 1204,  1310, 416,  432, 1234,  460, 1232,  494, 26490,  456, 1240,  452, 1238,  1300, 396,  452, 1240,  454, 1240,  1300, 396,  450, 1242,  452, 1242,  456, 1240,  454, 1240,  1302, 392,  456, 1238,  454, 1242,  454, 1242,  454, 1238,  456, 1284,  452, 1240,  454, 1240,  454, 1240,  456, 1236,  1300, 396,  452, 1242,  454, 1240,  458};  // UNKNOWN 6ABB2F08

Timestamp : 000035.671
Encoding  : UNKNOWN
Code      : 6ABB2F08 (48 bits)
Library   : v2.5.5

Raw Timing[95]:
   +   464, -  1228,    +   468, -  1226,    +  1316, -   408,    +   462, -  1202, 
   +   464, -  1230,    +  1338, -   386,    +   430, -  1232,    +   492, -  1206, 
   +   468, -  1226,    +   492, -  1200,    +  1344, -   380,    +   464, -  1200, 
   +   462, -  1232,    +   466, -  1228,    +   494, -  1200,    +   466, -  1272, 
   +   494, -  1200,    +   464, -  1230,    +   466, -  1226,    +   468, -  1226, 
   +  1314, -   410,    +   458, -  1204,    +   460, -  1238,    +   462, - 26516, 
   +   456, -  1236,    +   462, -  1232,    +  1306, -   416,    +   432, -  1234, 
   +   456, -  1234,    +  1308, -   418,    +   428, -  1234,    +   460, -  1236, 
   +   460, -  1234,    +   434, -  1260,    +  1282, -   442,    +   430, -  1234, 
   +   460, -  1232,    +   458, -  1238,    +   458, -  1238,    +   458, -  1280, 
   +   456, -  1240,    +   456, -  1238,    +   460, -  1234,    +   458, -  1234, 
   +  1300, -   422,    +   430, -  1236,    +   460, -  1234,    +   456

uint16_t rawData[95] = {464, 1228,  468, 1226,  1316, 408,  462, 1202,  464, 1230,  1338, 386,  430, 1232,  492, 1206,  468, 1226,  492, 1200,  1344, 380,  464, 1200,  462, 1232,  466, 1228,  494, 1200,  466, 1272,  494, 1200,  464, 1230,  466, 1226,  468, 1226,  1314, 410,  458, 1204,  460, 1238,  462, 26516,  456, 1236,  462, 1232,  1306, 416,  432, 1234,  456, 1234,  1308, 418,  428, 1234,  460, 1236,  460, 1234,  434, 1260,  1282, 442,  430, 1234,  460, 1232,  458, 1238,  458, 1238,  458, 1280,  456, 1240,  456, 1238,  460, 1234,  458, 1234,  1300, 422,  430, 1236,  460, 1234,  456};  // UNKNOWN 6ABB2F08

Timestamp : 000036.799
Encoding  : UNKNOWN
Code      : 6ABB2F08 (48 bits)
Library   : v2.5.5

Raw Timing[95]:
[... cut ...]
uint16_t rawData[95] = {494, 1198,  488, 1206,  1304, 418,  438, 1228,  496, 1196,  1336, 386,  462, 1202,  498, 1198,  486, 1208,  488, 1204,  1336, 388,  466, 1200,  490, 1202,  492, 1204,  496, 1198,  490, 1248,  494, 1200,  494, 1200,  492, 1202,  492, 1198,  1316, 410,  466, 1200,  488, 1206,  490, 26482,  462, 1232,  462, 1230,  1312, 414,  434, 1232,  466, 1228,  1306, 416,  434, 1232,  462, 1230,  464, 1230,  466, 1226,  1308, 416,  430, 1236,  464, 1232,  458, 1234,  460, 1234,  466, 1274,  464, 1230,  464, 1232,  490, 1204,  488, 1204,  1310, 414,  432, 1236,  458, 1234,  466};  // UNKNOWN 6ABB2F08
```

How could I read out the needed Stuff out of this ? 😄

AnalysIR, 2019-02-27:
use the raw data buffer to send the signal back using sendRAW.

PS: due to travel, I will not be able to assist any further for a while.

lSh4dowl, 2019-04-13:
Thanks, after a break I found the Control Box, and I will hook my system up to that. Thanks for your amazing help anyway.

closed 2019-04-13 by lSh4dowl

Norde, 2020-07-16:
Hello,

Sorry to resuscitate this topic.

Here is some Pronto code working for the WSL 100 50 02 remote (with Velux Skylight).
Device 1 and "all" are listed (8 to 9 lacking).

https://github.com/Norde/Velux_Skylights/blob/master/Velux.csf

It send ProtonHex infrared code.
There's also an arduino sketch to convert these code to burst signal :

https://github.com/rossdanderson/Triggr

If it can help...

stonehage, 2020-07-18:
You need following:

tasmota_ir.bin (https://github.com/arendst/Tasmota/releases/tag/v8.3.1)

nodeMCU
IR Receiver -> D5 -> GPIO14
IR Transmitter -> D2 -> GPIO4

Tasmota -> Console -> Enter Command

Backlog SetOption58 1; Module 0; Template {"NAME":"nodeMCU+IR","GPIO":
[255,255,29,255,255,8,255,255,255,255,51,255,255],"FLAG":15,"BASE":18}; WebButton1 LED;

[image]

Now you can control with following Command the Velux Rollerschutter

**All Up**
[http://192.168.6.242/cm?cmnd=IrSend 0,474,1158,474,1160,1288,348,1284,346,1288,344,1286,346,470,1160,474,1160,472,1160,474,1160,1288,348,470,1162,472,1160,472,1160,474,1158,474,1160,472,1160,474,1160,474,1158,474,1160,472,1160,474,1160,1288,346,472,20044,474,1160,472,1160,1290,346,1288,378,1254,348,1286,376,440,1160,472,1160,474,1160,474,1160,1288,378,440,1160,474,1160,472,1160,474,1160,474,1160,474,1160,472,1160,474,1160,472,1162,472,1160,474,1160,1290,344,470](url)

**All Stop**
[http://192.168.6.242/cm?cmnd=IrSend 0,1288,344,476,1158,1288,346,1290,344,1286,346,1286,346,472,1160,474,1158,474,1160,474,1160,1290,346,470,1160,474,1160,472,1160,474,1160,474,1160,472,1160,474,1160,472,1158,474,1158,1290,348,470,1160,474,1158,474,20046,1288,346,472,1160,1288,348,1286,344,1288,346,1286,378,440,1158,474,1160,472,1160,474,1160,1290,346,470,1160,474,1160,474,1158,474,1160,472,1160,474,1160,474,1160,472,1160,474,1158,1290,346,472,1158,474,1160,474](url)

**All Down**
[http://192.168.6.242/cm?cmnd=IrSend 0,472,1160,1288,344,1288,346,1286,348,1286,346,1288,346,472,1158,474,1160,472,1162,470,1160,1288,346,472,1160,472,1160,474,1160,474,1158,472,1162,472,1160,474,1160,474,1160,474,1160,474,1160,1288,348,1286,346,1286,19228,474,1160,1288,346,1288,346,1288,344,1288,376,1256,348,470,1158,474,1158,474,1158,474,1160,1290,346,470,1160,474,1160,472,1160,472,1160,472,1162,472,1160,472,1158,474,1162,472,1160,472,1160,1290,346,1288,346,1286](url)

**M1 Up**
[http://192.168.6.242/cm?cmnd=IrSend 0,500,1132,500,1132,1288,378,470,1130,502,1132,1312,352,470,1130,502,1132,502,1130,1316,350,1284,348,470,1132,500,1132,502,1130,502,1132,502,1132,502,1130,502,1132,500,1132,500,1132,1314,350,470,1130,504,1130,1314,19204,500,1132,500,1132,1316,350,470,1132,500,1132,1314,350,470,1130,502,1130,502,1134,1314,350,1282,350,470,1130,502,1130,502,1132,502,1132,502,1130,502,1132,502,1132,502,1132,502,1132,1314,350,470,1132,500,1132,1316](url)

**M1 Stop**
[http://192.168.6.242/cm?cmnd=IrSend 0,1312,350,472,1130,1314,352,470,1130,502,1130,1316,350,470,1132,500,1132,502,1132,1316,350,1282,350,470,1130,500,1134,500,1132,500,1132,502,1130,500,1132,502,1130,502,1130,502,1132,502,1132,502,1132,1316,348,1284,19202,1316,350,470,1130,1314,352,468,1132,500,1134,1314,350,470,1130,502,1132,500,1132,1316,350,1282,350,470,1130,502,1132,502,1130,500,1132,502,1130,504,1130,502,1132,500,1132,500,1132,502,1132,500,1130,1316,350,1284](url)

**M1 Down**
[http://192.168.6.242/cm?cmnd=IrSend 0,500,1132,1314,352,1254,378,470,1136,496,1132,1316,350,470,1132,500,1132,502,1132,1314,350,1284,350,468,1134,500,1134,500,1132,500,1132,502,1132,500,1132,500,1134,500,1132,502,1132,1316,348,1284,350,470,1134,498,20016,502,1132,1314,350,1282,350,470,1130,500,1132,1316,350,470,1130,500,1132,502,1130,1316,350,1284,350,470,1132,500,1130,502,1134,498,1132,502,1132,502,1130,502,1130,504,1130,502,1132,1316,350,1282,350,470,1132,500](url)

**M2 Up**
[http://192.168.6.242/cm?cmnd=IrSend 0,500,1132,502,1132,1312,352,470,1130,1316,350,470,1132,502,1132,500,1132,504,1130,1316,350,1284,348,470,1130,502,1132,502,1130,502,1132,500,1132,502,1132,500,1132,502,1132,500,1132,1316,352,1282,350,470,1130,502,20046,502,1130,502,1132,1314,350,470,1130,1314,350,470,1130,502,1132,500,1134,500,1132,1316,350,1282,350,470,1130,502,1130,502,1132,502,1132,500,1132,502,1130,504,1130,502,1130,504,1132,1314,350,1284,350,468,1130,502](url)

**M2 Stop**
[http://192.168.6.242/cm?cmnd=IrSend 0,1314,350,468,1130,1290,376,470,1130,1316,350,470,1132,500,1130,502,1132,500,1132,1314,350,1282,352,468,1132,502,1132,502,1134,498,1132,502,1132,500,1132,500,1132,502,1134,500,1132,502,1132,1314,352,1282,350,470,20016,1314,350,470,1130,1316,350,470,1130,1316,350,470,1132,502,1130,502,1130,500,1132,1316,348,1284,350,470,1130,502,1132,502,1130,502,1132,500,1132,500,1132,502,1132,502,1130,502,1130,502,1132,1316,350,1284,350,472](url)

**M2 Down**
[http://192.168.6.242/cm?cmnd=IrSend 0,502,1132,1290,376,1258,376,468,1132,1316,350,470,1132,500,1134,502,1130,500,1132,1314,352,1280,352,470,1130,502,1130,502,1130,502,1132,500,1132,502,1130,502,1132,500,1132,502,1132,1314,352,468,1130,500,1134,1314,19172,502,1132,1316,318,1316,318,498,1132,1316,318,500,1132,502,1132,498,1134,502,1132,1314,320,1314,320,498,1134,500,1132,500,1132,500,1134,500,1132,500,1134,498,1134,474,1158,500,1132,1316,320,498,1132,500,1132,1316](url)

**M3 Up**
[http://192.168.6.242/cm?cmnd=IrSend 0,476,1158,498,1134,1314,320,1314,320,472,1160,474,1158,498,1134,474,1158,474,1160,1314,320,1312,322,472,1158,498,1134,474,1160,474,1158,474,1160,474,1158,474,1160,474,1158,498,1134,474,1158,1314,320,1288,346,472,20042,498,1134,474,1160,1292,342,1288,344,474,1158,474,1160,474,1158,474,1158,474,1158,1290,346,1312,322,472,1158,474,1160,476,1156,490,1142,500,1134,474,1158,474,1158,474,1158,474,1158,476,1158,1290,346,1286,346,472](url)

**M3 Stop**
[... cut ...] 32,500](url)

**M3 Down**
[http://192.168.6.242/cm?cmnd=IrSend 0,500,1130,1318,318,1314,318,1314,318,502,1130,504,1128,502,1130,502,1130,502,1134,1314,324,1310,318,500,1130,502,1132,502,1132,502,1132,502,1132,498,1134,500,1132,500,1132,502,1132,500,1132,502,1132,1316,320,1314,19202,500,1132,1316,318,1316,320,1312,320,500,1132,502,1132,500,1132,500,1132,502,1132,1314,320,1314,318,500,1132,502,1132,502,1130,500,1134,500,1132,500,1132,500,1134,500,1132,500,1132,500,1134,500,1134,1316,318,1314](url)

Norde, 2020-07-18:
Thanks for this complete guideline (and for the discover of Tasmota) :smile_cat:

FabriceReynolds, 2023-08-09:
A few years later...

Wanted to thank you for publishing these codes and add that using the following [library ](https://github.com/haimkastner/broadlink-ir-converter) I was able to convert these IR codes into [Broadlink base64 encoded strings](https://www.home-assistant.io/integrations/broadlink/#learned-codes-storage-location) for use with Broadlink IR hardware and Home Assistant. Note the photo above that shows the settings for the Velux remote. This important of these codes won't work.

```
"skylights": {
      "All Up": "JgBlAA8mDyYqCyoLKgsqCw8mDyYPJg8mKgsPJg8mDyYPJg8mDyYPJg8mDyYPJg8mKgsPAAKSDyYPJioLKgwpCyoMDiYPJg8mDyYqDA4mDyYPJg8mDyYPJg8mDyYPJg8mDyYqCw8ADQUAAAAAAAA=",
      "All Stop": "JgBlACoLDyYqCyoLKgsqCw8mDyYPJg8mKgsPJg8mDyYPJg8mDyYPJg8mDyYqCw8mDyYPAAKSKgsPJioLKgsqCyoMDiYPJg8mDyYqCw8mDyYPJg8mDyYPJg8mDyYPJioLDyYPJg8ADQUAAAAAAAA=",
      "All Down": "JgBlAA8mKgsqCyoLKgsqCw8mDyYPJg8mKgsPJg8mDyYPJg8mDyYPJg8mDyYPJioLKgsqAAJ3DyYqCyoLKgsqDCkLDyYPJg8mDyYqCw8mDyYPJg8mDyYPJg8mDyYPJg8mKgsqCyoADQUAAAAAAAA=",
      "M1 Up": "JgBlABAlECUqDA8lECUrCw8lECUQJSsLKgsPJRAlECUQJRAlECUQJRAlECUrCw8lECUrAAJ2ECUQJSsLDyUQJSsLDyUQJRAlKwsqCw8lECUQJRAlECUQJRAlECUQJSsLDyUQJSsADQUAAAAAAAA=",
      "M1 Stop": "JgBlACsLDyUrCw8lECUrCw8lECUQJSsLKgsPJRAlECUQJRAlECUQJRAlECUQJRAlKwsqAAJ2KwsPJSsLDyUQJSsLDyUQJRAlKwsqCw8lECUQJRAlECUQJRAlECUQJRAlECUrCyoADQUAAAAAAAA=",
      "M1 Down": "JgBlABAlKwspDA8lECUrCw8lECUQJSsLKgsPJRAlECUQJRAlECUQJRAlECUrCyoLDyUQAAKRECUrCyoLDyUQJSsLDyUQJRAlKwsqCw8lECUQJRAlECUQJRAlECUQJSsLKgsPJRAADQUAAAAAAAA=",
      "M2 Up": "JgBlABAlECUrCw8lKwsPJRAlECUQJSsLKgsPJRAlECUQJRAlECUQJRAlECUrCyoLDyUQAAKSECUQJSsLDyUrCw8lECUQJRAlKwsqCw8lECUQJRAlECUQJRAlECUQJSsLKgsPJRAADQUAAAAAAAA=",
      "M2 Stop": "JgBlACsLDyUqDA8lKwsPJRAlECUQJSsLKgsPJRAlECUQJRAlECUQJRAlECUQJSsLKgsPAAKRKwsPJSsLDyUrCw8lECUQJRAlKwsqCw8lECUQJRAlECUQJRAlECUQJRAlKwsqCw8ADQUAAAAAAAA=",
      "M2 Down": "JgBlABAlKgwpDA8lKwsPJRAlECUQJSsLKgsPJRAlECUQJRAlECUQJRAlECUrCw8lECUrAAJ1ECUrCisKECUrChAlECUQJRAlKworChAlECUQJRAlECUQJRAlDyYQJSsKECUQJSsADQUAAAAAAAA=",
      "M3 Up": "JgBlAA8mECUrCisKDyYPJhAlDyYPJisKKwoPJhAlDyYPJg8mDyYPJg8mECUPJisKKgsPAAKSECUPJioLKgsPJg8mDyYPJg8mKgsrCg8mDyYPJRAlECUPJg8mDyYPJg8mKgsqCw8ADQUAAAAAAAA=",
      "M3 Stop": "JgBlACsKECUrCisKECUQJRAlECUQJSsKKwoQJRAlECUQJRAlECUQJRAlECUrCisKECUQAAKRKwoQJSsKKwoQJRAlECUQJRAlKworChAlECUQJRAlECUQJRAlECUQJSsKKwoQJRAADQUAAAAAAAA=",
      "M3 Down": "JgBlABAlKworCisKECUQJRAlECUQJSsKKwoQJRAlECUQJRAlECUQJRAlECUQJRAlKworAAJ2ECUrCisKKwoQJRAlECUQJRAlKworChAlECUQJRAlECUQJRAlECUQJRAlECUrCisADQUAAAAAAAA=",
      "M4 Up": "JgBlABAlECUrCw8lECUrCw8lECUrCw8lKwsPJRAlECUQJRAlECUQJRAlECUrCw8lKwsPAAKRECUQJSsLDyUQJSsLDyUQJSsLDyUrCw8lECUQJRAlECUQJRAlECUQJSsLDyUrCw8ADQUAAAAAAAA=",
      "M4 Stop": "JgBlACsLDyUrCw8lECUrCw8lECUrCw8lKwsPJRAlECUQJRAlECUQJRAlECUQJRAlECUQAAKRKwsPJSsLDyUQJSsLDyUQJSsLDyUrCw8lECUQJRAlECUQJRAlECUQJRAlECUQJRAADQUAAAAAAAA=",
      "M4 Down": "JgBlABAlKwsqCw8lECUrCw8lECUrCw8lKwsPJRAlECUQJRAlECUQJRAlECUrCyoLKgsqAAJ2ECUrCyoLDyUQJSsLDyUQJSsLDyUrCw8lECUQJRAlECUQJRAlECUQJSsLKgsqCyoADQUAAAAAAAA=",
      "M5 Up": "JgBlABAlECUrCw8lKwsPJRAlECUrCw8lKwsPJRAlECUQJRAlECUQJRAlECUrCyoLKgsqAAJ2ECUQJSsLDyUrCw8lECUQJSsLDyUrCw8lECUQJRAlECUQJRAlECUQJSsLKgsqCyoADQUAAAAAAAA=",
      "M5 Stop": "JgBlACsLDyUrCw8lKwsPJRAlECUrCw8lKwsPJRAlECUQJRAlECUQJRAlECUQJSsLDyUrAAJ2KwsPJSsLDyUrCw8lECUQJSsLDyUrCw8lECUQJRAlECUQJRAlECUQJRAlKwsPJSsADQUAAAAAAAA=",
      "M5 Down": "JgBlABAlKwsqCw8lKwsPJRAlECUrCw8lKwsPJRAlECUQJRAlECUQJRAlECUrChAlKwsPAAKRECUrCyoLDyUrCw8lECUQJSsLDyUrCw8lECUQJRAlECUQJRAlECUQJSsLDyUrCw8ADQUAAAAAAAA=",
      "M6 Up": "JgBlABAlECUqDCoLDyUQJRAlECUqDA8lKgwPJRAlECUQJRAlECUQJRAlECUQJSoMDyUrAAJ3ECUQJSsLKgsPJRAlECUQJSsLDyUrCw8lECUQJRAlECUQJRAlECUQJRAlKwsPJSsADQUAAAAAAAA=",
      "M6 Stop": "JgBlACoMDyUqDCkMDyUQJRAlECUrCw8lKwsPJRAlECUQJRAlECUQJRAlECUrCykMKgsqAAJ1KwsPJSoMKgsPJRAlECUQJSoMDyUrCw8lECUQJRAlECUQJRAlECUQJSsLKgsqCyoADQUAAAAAAAA=",
      "M6 Down": "JgBlABAlKgwpDCoLDyUQJRAlECUqDA8lKwsPJRAlECUQJRAlECUQJRAlECUQJRAlECUQAAKRECUqDCkMKQwPJRAlECUQJSoMDyUqDA8lECUQJRAlECUQJRAlECUQJRAlECUQJRAADQUAAAAAAAA=",
      "M7 Up": "JgBlABAlECUqDA8lECUqDA8lECUqDCkMKQwPJRAlECUQJRAlECUQJRAlECUqDA8lKgwpAAJ2ECUQJSoMDyUQJSsLDyUQJSsLKgsqCw8lECUQJRAlECUQJRAlECUQJSoMDyUrCyoADQUAAAAAAAA=",
      "M7 Stop": "JgBlACoMDiYqDA4mDyYqDA8lDyYqDCkMKQwOJQ8mDyYPJhAlECUPJRAlDyYQJRAlECUqAAJ3KgwPJSoMDiYPJioMDyUPJioMKQwpDA8lDyYQJRAlECUPJQ8mDyYQJRAlDyYQJSoADQUAAAAAAAA=",
      "M7 Down": "JgBlAA8mKgwpDA4mDyYqDA4mDyYqDCkMKQwOJg8mDyYPJg8mDyYPJg8mDyYqDCkMKQwOAAKSDyUqDCkMDiYPJioMDiYPJioMKQwpDA4mDyYPJg8mDyYPJg8mDyYPJioMKQwpDA4ADQUAAAAAAAA=",
      "M8 Up": "JgBlAA8mDyYqDA4mKgwOJg8mDyYqDCkMKQwOJg8mDyYPJg8mDyYPJg8mDyYqDCkMKQwOAAKSDyYPJioMDiYqDA4mDyYPJioMKQwpDA4mDyYPJg8mDyYPJg8mDyYPJioMKQwpDA4ADQUAAAAAAAA=",
      "M8 Stop": "JgBlACoMDiYqDA4mKgwOJg8mDyYqDCkMKQwOJg8mDyYPJg8mDyYPJg8mDyYPJioMDiYPAAKSKgwOJioMDiYqDA4mDyYPJioMKQwpDA4mDyYPJg8mDyYPJg8mDyYPJg8mKgwOJg8ADQUAAAAAAAA=",
      "M8 Down": "JgBlAA8mKgwpDA4mKgwOJg8mDyYqDCkMKQwOJg8mDyYPJg8mDyYPJg8mDyYqDA4mKgwpAAJ2DyYqDCkMDiYqDA4mDyYPJioMKQwpDA4mDyYPJg8mDyYPJg8mDyYPJioMDiYqDCkADQUAAAAAAAA=",
      "M9 Up": "JgBlAA8mDyYqDCkMDiYPJg8mDyYqDCkMKQwOJg8mDyYPJg8mDyYPJg8mDyYPJioMDiYPAAKSDyYPJioMKQwOJg8mDyYPJioMKQwpDA4mDyYPJg8mDyYPJg8mDyYPJg8mKgwOJg8ADQUAAAAAAAA=",
      "M9 Stop": "JgBlACoMDiYqDCkMDiYPJg8mDyYqDCkMKQwOJg8mDyYPJg8mDyYPJg8mDyYqDCkMKQwOAAKSKgwOJioMKQwOJg8mDyYPJioMKQwpDA4mDyYPJg8mDyYPJg8mDyYPJioMKQwpDA4ADQUAAAAAAAA=",
[... cut ...]
      "M15 Up": "JgBlAA8lDyYqCyoLDyUPJQ8lKgsPJSoLKgsPJg8mDyUPJQ8mDyUPJg8mDyUPJQ8lKgsPAAKRDyUPJioLKgsPJQ8lDyUqCw8lKgsqCw8lDyUPJQ8lDyUPJQ8mDyYPJg8lDyUqCw8ADQUAAAAAAAA=",
      "M15 Stop": "JgBlACoLDyYqCyoLDyUPJg8lKgsPJSoLKgsPJQ8lDyUPJQ8mDyUPJQ8lDyUqCw8lDyYPAAKRKgsPJSoLKgsPJQ8lDyYqCw8lKgsqCw8lDyUPJQ8lDyUPJQ8lDyUPJSoLDyUPJQ8ADQUAAAAAAAA=",
      "M15 Down": "JgBlAA8lKgsqCyoLDyUPJQ8lKgsPJSoLKgsPJQ8lDyUPJQ8lDyUPJQ8lDyUPJSoLKgsqAAJ3DyUqCyoLKgsPJQ8lDyUqCw8lKgsqCw8lDyUPJg8lDyUPJQ8lDyUPJQ8lKgsqCyoADQUAAAAAAAA="
    }

```

damianam, 2025-08-05:
Thanks a lot @FabriceReynolds and @stonehage! I can control now my old shutters with Home Assistant thanks to you. There is still one small problem though. I have 2 Velux/Windowmaster keyboards next to each other. If I leave the DIP switches on both as in the pictures, then the IR command activates 2 separate shutters (one per keyboard). That brings me to two questions:

- Is it possible to guess how the code would look like when changing one of the DIP switches, so I can kind of brute force finding the other codes I need? I have no clue about IR, so maybe that is nonsense.
- @stonehage any chance I can persuade you to learn and share the codes that the remote emits when the DIP switches are in any of the other positions?

FabriceReynolds, 2025-08-06:
Unfortunately without the remote I'm not able to help further as I don't think you can easily brute force the values so you would need to capture the remote IR codes and compare them to find the codes that represent the dip switch security setting. 

Ironically their user manual suggests not to use the dip settings above and that we should cycle the codes regularly for better security.

damianam, 2025-08-07:
That's super helpful, thanks! I'll give it a shot (somewhen) to generate the equivalent b64 codes with the different dip switch positions. If I succeed I'll share them here.

stonehage, 2025-08-10:
> Setting on Velux Keyboard [image]
> 
> Setting on my Velux Remote [image]
> 
> You need following: tasmota_ir.bin https://github.com/arendst/Tasmota/releases/tag/v8.3.1
> 
> **Option 1** nodeMCU IR Receiver -> D5 -> GPIO14 IR Transmitter -> D2 -> GPIO4
> 
> Tasmota -> Console -> Enter Command
> 
> Backlog SetOption58 1; Module 0; Template {"NAME":"nodeMCU+IR","GPIO": [255,255,29,255,255,8,255,255,255,255,51,255,255],"FLAG":15,"BASE":18}; WebButton1 LED;
> 
> [image]
> 
> **Option 2** Pearl auvisio NX-4519-675 http://www.pearl.de/a-NX4519-3103.shtml?vid=914 Tuya-Convert https://github.com/ct-Open-Source/tuya-convert
> 
> Tasmota -> Console -> Enter Command
> 
> Backlog SetOption58 1; Module 0; Template {"NAME":"NX-4519-675","GPIO":[255,255,255,255,52,51,0,0,255,255,8,255,255],"FLAG":1,"BASE":18}; [image]
> 
> I read the Codes from my Velux Remote WLR1005001 (45AM11B) / WLR 100 50 01 (M1-M15)
> 
> Now you can control with following Command the Velux Rollerschutter
> 
> **All Up** [http://192.168.178.242/cm?cmnd=IrSend 0,474,1158,474,1160,1288,348,1284,346,1288,344,1286,346,470,1160,474,1160,472,1160,474,1160,1288,348,470,1162,472,1160,472,1160,474,1158,474,1160,472,1160,474,1160,474,1158,474,1160,472,1160,474,1160,1288,346,472,20044,474,1160,472,1160,1290,346,1288,378,1254,348,1286,376,440,1160,472,1160,474,1160,474,1160,1288,378,440,1160,474,1160,472,1160,474,1160,474,1160,474,1160,472,1160,474,1160,472,1162,472,1160,474,1160,1290,344,470](url)
> 
> **All Stop** [http://192.168.178.242/cm?cmnd=IrSend 0,1288,344,476,1158,1288,346,1290,344,1286,346,1286,346,472,1160,474,1158,474,1160,474,1160,1290,346,470,1160,474,1160,472,1160,474,1160,474,1160,472,1160,474,1160,472,1158,474,1158,1290,348,470,1160,474,1158,474,20046,1288,346,472,1160,1288,348,1286,344,1288,346,1286,378,440,1158,474,1160,472,1160,474,1160,1290,346,470,1160,474,1160,474,1158,474,1160,472,1160,474,1160,474,1160,472,1160,474,1158,1290,346,472,1158,474,1160,474](url)
> 
> **All Down** [http://192.168.178.242/cm?cmnd=IrSend 0,472,1160,1288,344,1288,346,1286,348,1286,346,1288,346,472,1158,474,1160,472,1162,470,1160,1288,346,472,1160,472,1160,474,1160,474,1158,472,1162,472,1160,474,1160,474,1160,474,1160,474,1160,1288,348,1286,346,1286,19228,474,1160,1288,346,1288,346,1288,344,1288,376,1256,348,470,1158,474,1158,474,1158,474,1160,1290,346,470,1160,474,1160,472,1160,472,1160,472,1162,472,1160,472,1158,474,1162,472,1160,472,1160,1290,346,1288,346,1286](url)
> 
> **M1 Up** [http://192.168.178.242/cm?cmnd=IrSend 0,500,1132,500,1132,1288,378,470,1130,502,1132,1312,352,470,1130,502,1132,502,1130,1316,350,1284,348,470,1132,500,1132,502,1130,502,1132,502,1132,502,1130,502,1132,500,1132,500,1132,1314,350,470,1130,504,1130,1314,19204,500,1132,500,1132,1316,350,470,1132,500,1132,1314,350,470,1130,502,1130,502,1134,1314,350,1282,350,470,1130,502,1130,502,1132,502,1132,502,1130,502,1132,502,1132,502,1132,502,1132,1314,350,470,1132,500,1132,1316](url)
> 
> **M1 Stop** [http://192.168.178.242/cm?cmnd=IrSend 0,1312,350,472,1130,1314,352,470,1130,502,1130,1316,350,470,1132,500,1132,502,1132,1316,350,1282,350,470,1130,500,1134,500,1132,500,1132,502,1130,500,1132,502,1130,502,1130,502,1132,502,1132,502,1132,1316,348,1284,19202,1316,350,470,1130,1314,352,468,1132,500,1134,1314,350,470,1130,502,1132,500,1132,1316,350,1282,350,470,1130,502,1132,502,1130,500,1132,502,1130,504,1130,502,1132,500,1132,500,1132,502,1132,500,1130,1316,350,1284](url)
> 
> **M1 Down** [http://192.168.178.242/cm?cmnd=IrSend 0,500,1132,1314,352,1254,378,470,1136,496,1132,1316,350,470,1132,500,1132,502,1132,1314,350,1284,350,468,1134,500,1134,500,1132,500,1132,502,1132,500,1132,500,1134,500,1132,502,1132,1316,348,1284,350,470,1134,498,20016,502,1132,1314,350,1282,350,470,1130,500,1132,1316,350,470,1130,500,1132,502,1130,1316,350,1284,350,470,1132,500,1130,502,1134,498,1132,502,1132,502,1130,502,1130,504,1130,502,1132,1316,350,1282,350,470,1132,500](url)
> 
> **M2 Up** [http://192.168.178.242/cm?cmnd=IrSend 0,500,1132,502,1132,1312,352,470,1130,1316,350,470,1132,502,1132,500,1132,504,1130,1316,350,1284,348,470,1130,502,1132,502,1130,502,1132,500,1132,502,1132,500,1132,502,1132,500,1132,1316,352,1282,350,470,1130,502,20046,502,1130,502,1132,1314,350,470,1130,1314,350,470,1130,502,1132,500,1134,500,1132,1316,350,1282,350,470,1130,502,1130,502,1132,502,1132,500,1132,502,1130,504,1130,502,1130,504,1132,1314,350,1284,350,468,1130,502](url)
> 
> **M2 Stop** [http://192.168.178.242/cm?cmnd=IrSend 0,1314,350,468,1130,1290,376,470,1130,1316,350,470,1132,500,1130,502,1132,500,1132,1314,350,1282,352,468,1132,502,1132,502,1134,498,1132,502,1132,500,1132,500,1132,502,1134,500,1132,502,1132,1314,352,1282,350,470,20016,1314,350,470,1130,1316,350,470,1130,1316,350,470,1132,502,1130,502,1130,500,1132,1316,348,1284,350,470,1130,502,1132,502,1130,502,1132,500,1132,500,1132,502,1132,502,1130,502,1130,502,1132,1316,350,1284,350,472](url)
> 
> **M2 Down** [http://192.168.178.242/cm?cmnd=IrSend 0,502,1132,1290,376,1258,376,468,1132,1316,350,470,1132,500,1134,502,1130,500,1132,1314,352,1280,352,470,1130,502,1130,502,1130,502,1132,500,1132,502,1130,502,1132,500,1132,502,1132,1314,352,468,1130,500,1134,1314,19172,502,1132,1316,318,1316,318,498,1132,1316,318,500,1132,502,1132,498,1134,502,1132,1314,320,1314,320,498,1134,500,1132,500,1132,500,1134,500,1132,500,1134,498,1134,474,1158,500,1132,1316,320,498,1132,500,1132,1316](url)
> 
[... cut ...]
[009_dip_switch_0000000010_DVES_91B494.txt](https://github.com/user-attachments/files/21706837/009_dip_switch_0000000010_DVES_91B494.txt)
[009_dip_switch_0000000010_DVES_9122F0.txt](https://github.com/user-attachments/files/21706826/009_dip_switch_0000000010_DVES_9122F0.txt)
[010_dip_switch_0000000001_DVES_91B494.txt](https://github.com/user-attachments/files/21706841/010_dip_switch_0000000001_DVES_91B494.txt)
[010_dip_switch_0000000001_DVES_9122F0.txt](https://github.com/user-attachments/files/21706834/010_dip_switch_0000000001_DVES_9122F0.txt)

damianam, 2025-08-12:
That's awesome, thank you so much! That spared me the effort of generating the raw data, it would have taken me a while since I have 0 experience with Arduino or the IRRemote library. The corresponding broadlink codes are below. I haven't test them yet though. When I get the chance to test them I'll report back. Sorry for the flood/long message.

```
"001_dip_switch_1000000000_DVES_91B494.txt": {
    "All    ▲": "JgBlABAmDyUqCykKKgoqChAmDyYPJRAlKwkPJg8mDyUPJg8mDyYPJg8lECYPJQ8mKQoQAAKRECYPJikLKQspCioKDyYPJBAlECUrCg8mDyYPJg8mDiYPJQ8lECYPJQ8mDyYqCg8ADQUAAAAAAAA=",
    "All    ■": "JgBlACsJECUpCisJKgorChAmDyUQJRAmKgoPJg8lECYPJg8mDyYPJBAmDyUqCRAlECYPAAKRKgoPJikLKQoqCioKECUPJg8lECYpCREmDyYPJg8mDyUPJQ8mDyUQJioKDyYPJRAADQUAAAAAAAA=",
    "All    ▼": "JgBlABAmKQspCykLKQoqChAmDyYOJg8mKQoQJg8mDyYPJg8mDyYPJQ8mDyUQJCoKKgoqAAJ2ECUqCioLKQspCioKECUPJBAlECUpCg8lECUPJg8mDyYPJRAlECQQJg8mKQoqCykADQUAAAAAAAA=",
    "M01    ▲": "JgBlABAmDyYpChAmDyYpCw8mDyYPJSkKKwoPJg8mDyYPJg8mDyYPJQ8lECYpCw8mDyQqAAJ2ECYPJikLDyYPJSoJECUPJg8mKQspChAmDyUPJg8mDyYPJRAlDyYPJSoLDyUPJikADQUAAAAAAAA=",
    "M01    ■": "JgBlACoKECYpChAlECUrCRAlDyYPJikKKgoPJg4mDyYPJg8lDyYPJQ8mDyUQJg8mKQoqAAJ2KgoPJikLDyYPJikKECYPJg8lKQoqChAlDyUPJQ8mDyUQJg8lDyYPJg8mDyUqCisADQUAAAAAAAA=",
    "M01    ▼": "JgBlABAmKQoqCg8lECYpChAmDyUPJSoKKgsPJBAmDyUPJg8lDyUQJQ8mDyYpCykKDyUQAAKRECYpCioKDyYPJSoKECUPJg8mKQoqCg8lECUPJg8lDyYPJg8mDyYPJikKKgoQJg8ADQUAAAAAAAA=",
    "M02    ▲": "JgBlABAmDyYpCw8lKQoQJg8mDyUPJikKKgoQJQ8mDyYPJA8mDyYPJg8lDyYpCioKDyUQAAKRECYPJikKECUpChAmDyUQJg8mKQoqCw8lDyYPJRAmDyYPJhAkDyUPJioJKgoQJRAADQUAAAAAAAA=",
    "M02    ■": "JgBlACsLDiUqChAlKgoQJA8mDyUQJikKKgoQJg8mDyUPJRAmDyYPJg8mDyYOJikKKwoPAAKRKgsPJikKDyYqCg8lDyYPJg8kKgoqCg8mDiYPJg8lECUQJRAmDyUPJg8mKQoqChAADQUAAAAAAAA=",
    "M02    ▼": "JgBlABAmKQoqChAlKQoQJA8mDyYPJikKKgoQJg8mDyYPJg8mDyYPJg8mDyYpChAmDyUqAAJ1ECYpCioKECYpCw8lDyYPJg8mKgopCg8mDyYPJQ8mDyUQJg8mDyYPJikKDyYPJSkADQUAAAAAAAA=",
    "M03    ▲": "JgBlABAmDyUqCioKDyYPJg8lECYPJikKKgoQJQ8lDyUPJRAmDyYPJQ8lDyYPJikLKgkQAAKRECYPJCoKKgoQJg8mDyYPJg8lKgorCRAlECUQJg8mDyYPJRAmDyYPJg8lKgoqCg8ADQUAAAAAAAA=",
    "M03    ■": "JgBlACoLDiYpCioLDyUQJQ8mDyYPJSoKKgoQJg8lECYPJg8mDyYPJhAlDyYpCioKDyYPAAKRLQgQJSoKKwkQJQ8lECUPJg8mKQspCg8mDyYPJg8mDyYPJg8mDyYPJikKKgoPJg8ADQUAAAAAAAA=",
    "M03    ▼": "JgBlABAmKQoqCioKECYQJQ8mDyYPJikKKgoPJg8mDiUPJg8mDyYPJg8mDyYPJg8mKQspAAJ2ECYpCisJKgsOJg8lDyYPJRAmKQoqChAmDyUPJg8mDyYPJRAlECUQJQ8mDyUpCioADQUAAAAAAAA=",
    "M04    ▲": "JgBlABAmDyUpChAmDyUpCg8mDyYpCw8mKQoQJg8mDyYPJQ8mDyYPJg8mDyUpChAmKQoQAAKRECYPJikLDyYOJikKECYPJikKECYpCw4lECQQJg8mDyYPJg8mDyYPJikKDyYqCg8ADQUAAAAAAAA=",
    "M04    ■": "JgBlACoKECYpCg8mDyYqCRAmDyYpChAmKQoQJg8mDyYPJRAmDyYPJQ8mDyYPJg8mDyYPAAKRKwkPJSoKECUPJSoKECYPJSoKECYpChAlECYPJg8mDyYPJQ8mDyYPJg8mDyUPJg8ADQUAAAAAAAA=",
    "M04    ▼": "JgBlABAmKQorCw8mDyYpChAmDyUqCw8mKgsPJQ8mDyUQJg8lECYPJg8mDyYpCioKKgspAAJ2ECYpCioLDiYPJikKDyYPJikKECYpChAmDyYOJg8mDyYPJRAmDyUQJikKKgoqCioADQUAAAAAAAA=",
    "M05    ▲": "JgBlABAmDyUqChAmKQoQJQ8mDyUqCw8lKQoQJg8mDyYPJRAlDyYPJg8mDyYqCSoKKgoqAAJ2ECUQJikKDyYqCRAmDiUQJikKDyYpCw8lDyYPJQ8mDyYPJg8mDyYPJikKKgspCioADQUAAAAAAAA=",
    "M05    ■": "JgBlACoLDiUqChAmKQoQJg8lDyUqChAlKgoPJRAmDyYPJg8lECYPJg8mDyYPJikKDyYpAAJ2KwoQJikLDiYpChAkECYPJikKECUpChAmDyYPJQ8mDyYPJQ8mDyUPJg8mKQoQJikADQUAAAAAAAA=",
    "M05    ▼": "JgBlABAmKQspCw8mKQoQJQ8mDyYpChAmKQsPJg8lECYPJQ8mDyYPJg8lECUqChAmKQsOAAKSECUqCioKECUpChAlDyUQJSoLDyYpCg8mDyUQJQ8mDyYPJg8mDyYPJioKDyYpChAADQUAAAAAAAA=",
    "M06    ▲": "JgBlABAmDyYpCioKECYPJg8mDyYqCRAmKgoPJg8lECYPJg8mDyYPJg8lDyYPJSoJECYpAAJ1ECYPJikKKwkQJg8lECYPJikKDyYpCw8lDyYPJg8mDyYPJg8mDyYPJg8lKgoQJSoADQUAAAAAAAA=",
    "M06    ■": "JgBlACoKDyYpCykKDyYPJg4mDyYqCRAmKQsPJg8mDyYPJg8mDyYPJg8mDyUqCykKKgoqAAJ2KgoQJikLKgkQJg8mDyYPJSoKDyYpChAlDyYPJg8mDyYPJg8mDyYPJioKKgorCioADQUAAAAAAAA=",
    "M06    ▼": "JgBlABAmKQspCyoJECYPJg8mDyYpChAmKgkQJQ8kDyYPJQ8mDyUPJRAlECUPJg8kECUQAAKQECYpCioKKgoQJg4lECYPJioKDyYqCRAmDyYPJQ8mDyYPJg8lDyUQJg8mDyUQJBAADQUAAAAAAAA=",
    "M07    ▲": "JgBlABAlDyUqChAmDyYpChAmDyYpCioKKgoQJQ8mDyYPJg8mDyYPJg8mDyYpCw8mKQoqAAJ2ECUQJioKDyYPJCoKECYPJSkKKgspChAmDyYPJg8mDyYPJg8mDyQQJSkKECYpCioADQUAAAAAAAA=",
    "M07    ■": "JgBlACoKECYqCRAkDyYpCw8mDiYpCioKKwkPJg8mDyYPJQ8mDyYPJQ8mDyYPJg8mDyYpAAJ1KgoPJikKECYPJSkKECUQJioJKgspCg8mDyYPJg8mDyUQJg8mDyYPJRAmDyYPJikADQUAAAAAAAA=",
    "M07    ▼": "JgBlABAmKQoqCg8mDyUpCg8lECYpCykKKgsPJQ8lDyYPJg8mDyYPJg8mDyYpCioLKgkQAAKQECYpCioLDyYPJikLDyYPJSoKKgoqChAmDyYPJg4lECYPJg8lECYPJikLKQoqCw8ADQUAAAAAAAA=",
    "M08    ▲": "JgBlABAlDyYpChElKgkQJg8lECUqCisJKgoQJBAmDyYPJg8lDyUPJg8mDyYpCioLKQoQAAKRECUQJSoLDyUpCw4mDyYPJSsJKwkrCRAmDyYPJQ8mDyYPJQ8mDyYPJSoKKgspChAADQUAAAAAAAA=",
    "M08    ■": "JgBlACoKECYpCw8lKgkQJg8mDyYpCioKKgoPJg8mDyYPJg8mDyUPJg8lECYPJioJECUPAAKRKgoQJikKECYpCg8mDyUQJioJKgorCRAlDyUQJQ8mDyYPJQ8lECUQJg8lKgoQJQ4ADQUAAAAAAAA=",
    "M08    ▼": "JgBlABAmKQoqCg8mKQsOJg4lECYpCioKKwoPJg8lECYPJg8mDyYPJg8mDyYqChAlKgspAAJ2ECUqCioKECYpChAmDyYPJSkKKgoqCw4mDyYPJRAlECYPJg8lECYPJikKDyYpCioADQUAAAAAAAA=",
    "M09    ▲": "JgBnABElDyUpCioKECUQJg8lDyYpCisJKwoQJg8lECYPJg8mDyYPJRAmDyUPJikKDyUQAAKPECUQJSkLKQoPJg8lECUPJikKKwopCw8mDyUQJg8lECYPJRAmDyYPJg8mKQoPJg8ABF0FAA0FAAAAAAAA",
    "M09    ■": "JgBlACsLDyYqCSoKECUQJg8mDyYpCioKKgoQJg8lECYPJQ8mDyUPJg8mDyYpCisKKQoQAAKQKgoPJikKKgoQJg8mDyUPJikKKgorCRAlECYPJg8lECYPJg8lDyUPJioJKgoqChAADQUAAAAAAAA=",
    "M09    ▼": "JgBlABAmKQoqCisJECQQJRAlDyUqCioKKgoQJg8mDyUPJQ8kECUPJg8lDyYPJQ8mDyYpAAJ2ECYpCioKKwkPJg8mDyYPJSoLKgkqChAmDyYPJg8mDyYPJg8mDyUQJQ8mDyYPJikADQUAAAAAAAA=",
[... cut ...]
    "M30    ▲": "JgBlABAmDyYpCioLDiUQJikLDyUqCw4mDyYPJg8mDyYPJg8mDyYPJRAmKgopCioLDyYpAAJ1ECYPJikKKgoQJg8lKgsPJikLDyYPJg8lECYPJg8mDyYPJRAmDyYpCykLKQsPJSkADQUAAAAAAAA=",
    "M30    ■": "JgBlACsKDyUqCykLDiUQJikLDyYpCw4mDyYPJRAmDyYPJg8lECUQJg8lKgsPJSoLKQspAAJ2KwsPJikLKgoPJg8mKQsOJikKECYPJg8mDyYPJg8mDiUQJg8mDyYpCw8lKgspCioADQUAAAAAAAA=",
    "M30    ▼": "JgBlABAmKQspCykLDyUQJSoKDyYpCw8lECYPJRAmDyYOJg8lECUQJg8mKQspCw8mDyYOAAKRECYpCisJKgoQJg8lKwoPJikLDyYPJBAlECYPJRAmDyYPJg8mDyUqCioKDyUQJQ8ADQUAAAAAAAA="
}
```

ArminJo, 2025-08-14:
OK Now the checksum is computed for all commands with the automatic bit set.