Arduino-IRremote/Arduino-IRremote #614: Help storing and sending raw signal(unknown protocol)

915479996, 2019-03-10:
**Board:** ARDUINO MEGA 2560
**Library Version:** 2.1.0
**Protocol:** ALL

**Code Block:**
```c

#include <IRremote.h> 
#include <EEPROM.h>
#include "EEPROMAnything.h"
.....

```

**Checklist:**
- [✔] I have **read** the README.md file thoroughly
- [✔] I have searched existing issues to see if there is anything I have missed.
- [✔] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [✔] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [✔] The title of the issue is helpful and relevant 

**Issue:**
In my project, I need to record any remote(AC/tv/lamp) including all protocols and store the signal in EEPROM. I used the example of ‘IRrecord.ino’ as a reference to complete part of the known protocols. I stored the signal as a hex value in EEPROM. Also, the program can read the signal from EEPROM and send it out according to its protocol. 
However, I only have the raw signal data when the signal is from an unknown protocol. And the memory of Arduino board does not allow me to store the data as raw(too big). _**Can I convert the raw data to hex or bit when I store it at EEPROM. (In the 'IRrecord.ino', Is the value of 'codeValue' I need for storing raw signal?)**_ Then, I need to convert it back and use IRsend.raw() function to send the signal out. **_How can I turn it back to raw signal data._**

AnalysIR, 2019-03-10:
It is unlikely, you will find a solution available as you described above.

As you are using a Mega2560, you could actually store the RAW signals in FLAsh. (There are some examples of storing/sending signals from flash on our [blog](http://www.analysir.com/blog/))

To send signals from HEX values, you can check out the how the IRremote library works(usually for up to 32 its). We also have some examples of sending signals from HEX for any number of bits - on our [blog](http://www.analysir.com/blog/)).

915479996, 2019-03-10:
> It is unlikely, you will find a solution available as you described above.
>
> As you are using a Mega2560, you could actually store the RAW signals in FLAsh. (There are some examples of [...]

Can I compress the raw signal to bits or hex value?

ArminJo, 2020-07-04:
It is not supported by the library 😩

closed 2020-07-04 by ArminJo