Arduino-IRremote/Arduino-IRremote #615: IR Remote not recieving another code/signals

NikdoNic, 2019-03-14:
Hello, im controling WS2812 LED strip with Arduino Uno and IR Remote and reciever. Im controling few LED patterns with controler and i ran into the problem. Remote can turn only one of the patterns, than it stops to work.

when i start this i can push any button i want, until i´ll push button which is connected to some of the patterns. After i push the button which is connected to some patern the patern is lightning up, but if i want to change the led strip to different patern it will do nothing, because of not recieving another signal/code from remote. After i push button with connected pattern to it and than i want to change the pattern, the reciever will stop working. (just not recieving another codes/signals).

Can you help me with my problem?
//Im beginner...

Board: ARDUINO UNO Rev 3
Library Version: 2.2.0
````
#include <FastLED.h>
#include <IRremote.h>

#define LED_PIN     6
#define NUM_LEDS    68
#define LED_TYPE    WS2812B
#define COLOR_ORDER GRB
#define BRIGHTNESS  64
#define RECV_PIN  3

IRrecv irrecv(RECV_PIN);
decode_results results;
CRGB leds[NUM_LEDS];

void setup() {
    Serial.begin(9600);
    delay(3000); 
    irrecv.enableIRIn();
    FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
    FastLED.setBrightness(BRIGHTNESS);
}

void loop(){
if (irrecv.decode(&results)){
if (results.value == 0xFF12ED){
   fill_rainbow( leds, 17, millis()/200 ); 
    for (uint8_t i=0; i<17; i++) {
     leds[i+17] = leds[i];  //pixels 0-16 with reversed order
     leds[33-i] = leds[i];  //pixels 17-33 with reversed order
     leds[i+34] = leds[i];  //pixels 34-50
     leds[67-i] = leds[i];  //pixels 51-67 with reversed order
     FastLED.show();}
if (results.value == 0xFF1AE5){
  for(int i=0;i<17;i++){
    leds[i].setRGB(255,0,0);
    leds[33-i].setRGB(255,0,0);
    leds[i+34].setRGB(255,0,0);
    leds[67-i].setRGB(255,0,0);
    delay(40);
    FastLED.show();
    
}
if (results.value == 0xFF1A65){ 
   for(int i=0;i<17;i++){
    leds[i].setRGB(0,255,0);
    leds[33-i].setRGB(0,255,0);
    leds[i+34].setRGB(0,255,0);
    leds[67-i].setRGB(0,255,0);
    delay(40);
    FastLED.show();
    
}
if (results.value == 0xFF02FD){ 
  for(int i=0;i<17;i++){
    leds[i].setRGB(0,0,0);
    leds[33-i].setRGB(0,0,0);
    leds[i+34].setRGB(0,0,0);
    leds[67-i].setRGB(0,0,0);
    delay(40);
    FastLED.show();
   
} 
irrecv.resume();
}}}}}}
````

AnalysIR, 2019-03-14:
The end bracket for the IF statement of each signal may be in the wrong position. Try putting it after the end bracket of the related FOR loop.

It may help to format the code within the IDE (CTRL+T)

NikdoNic, 2019-03-14:
@AnalysIR Thanks a lot, its working now!! :) but it brought one problem, if i press the button with pattern, the pattern will not do anything, it just stay still. I ve got there rainbow or i think its dot running from one part to another but its not doing anything. The dot just runned once the strip and the rainbow is just staying still.

AnalysIR, 2019-03-14:
I suggest you start with just one signal/pattern & get that working. Then add in a second...and so on.

It will be easier for you to debug your own code with an approach like this.

closed 2020-07-04 by ArminJo