Arduino-IRremote/Arduino-IRremote #616: IR control mitsubishi without arduino

wixaw, 2019-03-19:
**Board:** Nothind

Hello everyone

I am looking to automate my Mitshibishi air conditioning.
I have an ESP8266 (with presence sensor, IR "sku 436122 38k", temperature BMP180) under espeasy and a raspberry pi (jeedom + mqtt server)

Can we use your tools without arduino?
Otherwise, do you have any ideas?

thank you in advance
William

AnalysIR, 2019-03-19:
IRremote is writen for Arduinos and similar MCUs in C. It is possible to port to other platforms.

There is a fork that runs on ESP8266 on GitHub.

Otherwise, you could use **LIRC** directly on RPi or Linux.

wixaw, 2019-03-25:
thank you,

I can not find forks documentation and people do not respond.
It is possible to pair LIRC with jeedom?

AnalysIR, 2019-03-25:
Unfortunately, I cannot assist any further as the issue does not really ralte to IRremote/Arduino.

Please refer technocal quries to the actual suppliers ofyour hardware.

IRremote is open source & everything is available on Github. The same apllies for all forks of IRremote.

wixaw, 2019-03-27:
Ok 
Thanks

closed 2019-03-27 by wixaw