Arduino-IRremote/Arduino-IRremote #617: Finding remote code for broken Philips remote

Arduinoslfan, 2019-03-26:
**Board:** ARDUINO UNO
**Library Version:** 2.2.3
**Protocol:** RC5 or RC6

Good day,

I'm a newbie and still learning.I'm looking for some help locating remote control codes for a broken remote.Remote.control model - PRC500-56 by Philips and it has a TM58PR11 micro controller inside. I have checked in various websites without success. I'm planning to make an Arduino based remote after finding codes.Is there any program that can help me to find remote codes? Full functionality can only be obtained through remote as the DVD micro system (model mcd183/98) has few buttons. I bought this in 2011.Thank you

AnalysIR, 2019-03-26:
try this....RC5 @ 36 kHZ  (may be VOL+)

Raw (21): 896,-896,1792,-1792,1792,-896,896,-896,896,-896,896,-896,896,-1792,1792,-896,896,-896,896,-896,896

use the sendRAW feature of IRremote....see examples.

Note: RC5 uses a toggle bit

Arduinoslfan, 2019-03-27:
Appreciate your help for a newbie.
I entered details into IRsendRawDemo example and it wont compile.

```#include <IRremote.h>

IRsend irsend;

void setup()
{

}

void loop() {
  int khz = 36; // 38kHz carrier frequency for the NEC protocol
  unsigned int irSignal[] = {896,-896,1792,-1792,1792,-896,896,-896,896,-896,896,-896,896,-1792,1792,-896,896,-896,896,-896,896}; //AnalysIR Batch Export (IRremote) - RAW

  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.

  delay(5000); //In this example, the signal will be repeated every 5 seconds, approximately.
}`

This program wont compile.Is it ok to change unsigned int to int?

AnalysIR, 2019-03-27:
make sure you have installed the library into the IDE & restart the IDE.

You seem to miss the include statement from the example

Arduinoslfan, 2019-04-06:
Good day,
Followed your instructions and still it wont compile.
However, I managed to get some features work through slightly different path.
I used irscrutinizer  to create a code and it worked.
It seems to use 0x040XX hex code group.
ex: power on is 0x040CU
Only limited sets of functionality is available through this code and some important features still does not work. Any method to find those missing codes?
Thank you Very much!

bengtmartensson, 2019-04-06:
The model-# is in general a pretty useless piece of information, even more so the uP. If you instead tell us/search for the model of the receiving device, you will probably be more successful. You can then likely use IrScrutinizer to generate a rudimentary sketch for you.

What was the code you found?

Arduinoslfan, 2019-04-07:
Model of the receiving device - Philips mcd183/98.
Listed below is the code.
Many Thanks.

``
#include` <IRremote.h>

// Constants
static const long BAUD = 115200UL; // Change if desired

// Global variables
IRsend irsend;
static unsigned int rc6Toggle = 0;

// Command #1: 0
// Protocol: rc6, Parameters: D=4U

// Command #2: 1
// Protocol: rc6, Parameters: D=4U F=1U

// Command #3: 2
// Protocol: rc6, Parameters: D=4U F=2U

// Command #4: 3
// Protocol: rc6, Parameters: D=4U F=3U

// Command #5: 4
// Protocol: rc6, Parameters: D=4U F=4U

// Command #6: 5
// Protocol: rc6, Parameters: D=4U F=5U

// Command #7: 6
// Protocol: rc6, Parameters: D=4U F=6U

// Command #8: 7
// Protocol: rc6, Parameters: D=4U F=7U

// Command #9: 8
// Protocol: rc6, Parameters: D=4U F=8U

// Command #10: 9
// Protocol: rc6, Parameters: D=4U F=9U

// Command #11: POWER
// Protocol: rc6, Parameters: D=4U F=12U

// Command #12: OSD MENU
// Protocol: rc6, Parameters: D=4U F=15U

// Command #13: SHUFFLE
// Protocol: rc6, Parameters: D=4U F=28U

// Command #14: REPEAT
// Protocol: rc6, Parameters: D=4U F=29U

// Command #15: SKIP >>
// Protocol: rc6, Parameters: D=4U F=32U

// Command #16: SKIP <<
// Protocol: rc6, Parameters: D=4U F=33U

// Command #17: PLAY
// Protocol: rc6, Parameters: D=4U F=44U

// Command #18: PAUSE
// Protocol: rc6, Parameters: D=4U F=48U

// Command #19: STOP
// Protocol: rc6, Parameters: D=4U F=49U
[... 158 lines cut ...]
        irsend.sendRC6(0x045BU + 0x10000U*rc6Toggle, 20U);
        rc6Toggle = 1U - rc6Toggle;
        break;
    case 28L:
        irsend.sendRC6(0x045CU + 0x10000U*rc6Toggle, 20U);
        rc6Toggle = 1U - rc6Toggle;
        break;
    case 29L:
        irsend.sendRC6(0x0483U + 0x10000U*rc6Toggle, 20U);
        rc6Toggle = 1U - rc6Toggle;
        break;
    case 30L:
        irsend.sendRC6(0x04C8U + 0x10000U*rc6Toggle, 20U);
        rc6Toggle = 1U - rc6Toggle;
        break;
    case 31L:
        irsend.sendRC6(0x04F7U + 0x10000U*rc6Toggle, 20U);
        rc6Toggle = 1U - rc6Toggle;
        break;
    default:
        Serial.println(F("Invalid number entered, try again"));
        break;
    }
}
`

bengtmartensson, 2019-04-07:
The sketch looks fine (except for two spurious backticks). (Code includes of this size should preferably be gists...)

Everything working now? Then you can close the issue.

Arduinoslfan, 2019-04-07:
Unfortunately, 12 keys still not working. 
Is there any range of values I can check for  (so that I can manually look for missing codes ).
Many Thanks.

AnalysIR, 2019-04-07:
The MCD series also uses RC5 0x4**nn** (along with RC6)

plus a toggle. So you should/may be able to find them by varying the command byte (nn)

bengtmartensson, 2019-04-07:
If the original remote is still available, try to read it out. Otherwise you can search the internet for codes, [this](http://www.harctoolbox.org/IR-resources.html) is my list of places to try. As above, do not forget to look for RC5 codes. As a last resort, you can blindly try "all possible" codes with the same device ("system") address. Note that there is a in IrScrutinizer' remote editor, you can select right mouse button -> advanced -> Add missing Fs to generate all such signal, and then export the table to an IRremote sketch.

Arduinoslfan, 2019-04-08:
Highly Appreciate the continuous support by both of you.
this is a list of missing button codes

1.Disc 2. Tuner 3. USB 4.MP3Link(Aux) 5/Mode 6.Display(RDS) 7.DSC 8.LOUD/DBB 9.Clock. 10.Sleep(Timer ) 11.Angle (Prog) 12.Goto

Today I checked out https://irdb.globalcache.com and it had philips MCD series codes. seems everything is there. But irscrutinizer wont allow me to create arduino code based on that. The list of codes looks different to me and I'm having hard time understanding it. This is a small part of code for one button
its for Number 1 button on the remote 

function, code1, hexcode1, code2, hexcode2

"DIGIT 1","sendir,1:1,1,36000,1,1,96,32,16,32,16,16,16,16,16,32,32,16,16,16,16,16,16,16,16,16,32,32,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,32,3037","0000 0073 0000 0014 0060 0020 0010 0020 0010 0010 0010 0010 0010 0020 0020 0010 0010 0010 0010 0010 0010 0010 0010 0010 0020 0020 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0020 0BDD","sendir,1:1,1,36000,1,1,96,32,16,32,16,16,16,16,48,48,16,16,16,16,16,16,16,16,32,32,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,32,3037","0000 0073 0000 0013 0060 0020 0010 0020 0010 0010 0010 0010 0030 0030 0010 0010 0010 0010 0010 0010 0010 0010 0020 0020 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0020 0BDD"

Tried to use their iconvert software & it was not useful either.
I thought, I'm suppose to see a hex code like  0x410 etc..
But here I'm getting this hex code from converter. 
0000 0073 0000 0014 0060 0020 0010 0020 0010 0010 0010 0010 0010 0020 0020 0010 0010 0010 0010 0010 0010 0010 0010 0010 0020 0020 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0020 0BDD
Sorry I'm confused.
Could you guys please clarify this?
Thank you.

bengtmartensson, 2019-04-08:
Importing these files in IrScrutinizer is straightforward, and descibed in [this thread](http://www.hifi-remote.com/forums/viewtopic.php?t=100804). (The file is in CSV format, columns name, GlobalCache's sendir format, Pronto Hex.) After importing the codes, export as "Arduino/IRremote" (but you appear to have discovered that already.)

The "hex code like 0x410" is not a standard by any means, rather a private language for IRremote. IrScrutinizer knows how to translate many IR signals to that private language. The exact logic is found in the file exportformats.d/IRremote.xml.

closed 2020-07-04 by ArminJo