Arduino-IRremote/Arduino-IRremote #618: Compiler Warnings

ArminJo, 2019-04-01:
**Board:** ARDUINO UNO
**Library Version:** 2.2.3

Hi,
line 55ff in IRremote.cpp reads
```
  if (passed)
    DBG_PRINTLN(F("?; passed"));
  else
    DBG_PRINTLN(F("?; FAILED")); 
        return passed;
```
The compiler gets me a warning:
../src/lib/IRremote/IRremote.cpp:57:3: warning: this 'else' clause does not guard... [-Wmisleading-indentation]

It would be very kind if you could fix this and the warnings on line 83 and 109 too.

Many thanks
Armin

z3t0, 2019-04-04:
Hi Armin,

Thank you so much for reporting the issue.

Could you try replacing the above code with 

```c
  if (passed){ 
   DBG_PRINTLN(F("?; passed"));
}
  else {
    DBG_PRINTLN(F("?; FAILED")); 
        return passed;
}
```

ArminJo, 2019-04-04:
Hi z3t0
do you mean
```
    if (passed) {
        DBG_PRINTLN(F("?; passed"));
    } else {
        DBG_PRINTLN(F("?; FAILED"));
    }
    return passed;
```
Your version lacks the return statement in the if case and gives a warning
`warning: control reaches end of non-void function [-Wreturn-type]`.
I changed it for me like above and it worked fine.

ArminJo, 2019-04-04:
By the way, 
I created a quick comparison table for IR libraries.
https://github.com/ukw100/IRMP#quick-comparison-of-4-arduino-ir-receiving-libraries
Please check, if the data regarding your library is correct.
Thanks
Armin

closed 2020-07-02 by ArminJo