Arduino-IRremote/Arduino-IRremote #619: ESP32 crashes when trying to commit to EEPROM

malebuffy, 2019-04-05:
**Board:** ESP32 DEV KIT
**Library Version:** latest
**Protocol:** 

**Code Block:**
EEPROM.commit();

When using the ESP32 there is a sort of conflict with the EEPROM.h library. Read works ok (it doesnt crash) but as soon as EEPROM.commit(); is used to save data to EEPROM it crashes.

malebuffy, 2019-04-05:
double issue

closed 2019-04-05 by malebuffy

imranics, 2020-06-18:
hi i am  imranics from india ,whenever i save data in eeprom using  EEPROM.commit();  my  esp32 crashes suddenly   ,, i am getting error my code restarts  , i am writing code for ir_remote with glcd interface for my product , it has display library #include <U8g2lib.h> , #include <IRremote.h>, #include "EEPROM.h" ,#include <Arduino.h>  , i will save eeprom  by triggering a remote button from ir remote and saving a integer value which is from (0- 100)  , esp32 send a error report i have attached plz check ,

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400d13b4  PS      : 0x00060034  A0      : 0x40084690  A1      : 0x3ffbe7b0  
A2      : 0x00000001  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x4008a658  
A6      : 0x3ff42000  A7      : 0x700000eb  A8      : 0x80081290  A9      : 0x3ff5f024  
A10     : 0x3ffbec1c  A11     : 0x20000000  A12     : 0x00000400  A13     : 0x00000000  
A14     : 0x3ff42000  A15     : 0x700000eb  SAR     : 0x00000019  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40062230  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d13b4

Backtrace: 0x400d13b4:0x3ffbe7b0 0x4008468d:0x3ffbe7d0 0x4006222d:0x3ffb1c60 0x4008d74f:0x3ffb1c80 0x4008d786:0x3ffb1cb0 0x4008dbb3:0x3ffb1ce0 0x400869e0:0x3ffb1d10 0x40086b7d:0x3ffb1d30 0x400dbef5:0x3ffb1d80 0x400da356:0x3ffb1da0 0x400da6d1:0x3ffb1dc0 0x400d96d7:0x3ffb1e40 0x400d9c76:0x3ffb1ec0 0x400d909d:0x3ffb1f30 0x400d3455:0x3ffb1f70 0x400d11df:0x3ffb1f90 0x400d55b1:0x3ffb1fb0 0x400882b5:0x3ffb1fd0

PC: 0x400d13b4: IRTimer() at C:\Users\imranics 2019\Documents\Arduino\libraries\Arduino-IRremote-master\IRremote.cpp line 131
EXCVADDR: 0x00000000

Decoding stack results
0x400d13b4: IRTimer() at C:\Users\imranics 2019\Documents\Arduino\libraries\Arduino-IRremote-master\IRremote.cpp line 131
0x4008d74f: esp_rom_spiflash_read_status at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 261
0x4008d786: esp_rom_spiflash_wait_idle at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 35
0x4008dbb3: esp_rom_spiflash_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 73
0x400869e0: spi_flash_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c line 192
0x40086b7d: spi_flash_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c line 362
0x400dbef5: nvs::nvs_flash_write(unsigned int, void const*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_ops.cpp line 66
0x400da356: nvs::Page::writeEntry(nvs::Item const&) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp line 98
0x400da6d1: nvs::Page::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int, unsigned char) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp line 230
0x400d96d7: nvs::Storage::writeMultiPageBlob(unsigned char, char const*, void const*, unsigned int, nvs::VerOffset) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp line 200
0x400d9c76: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp line 285
0x400d909d: nvs_set_blob(nvs_handle, char const*, void const*, size_t) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_api.cpp line 431
0x400d3455: EEPROMClass::commit() at C:\Users\imranics 2019\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\EEPROM\src\EEPROM.cpp line 198
0x400d11df: loop() at C:\Users\imranics 2019\Documents\Arduino\ESP___remote__INTEGER_NUMBER__/ESP___remote__INTEGER_NUMBER__.ino line 180
0x400d55b1: loopTask(void*) at C:\Users\imranics 2019\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x400882b5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

i need kind help ,,,,

AnalysIR, 2020-06-18:
FYI: This issue is closed.