Arduino-IRremote/Arduino-IRremote #620: RC5 read twice times

gavron04, 2019-04-10:
Hello. I have simply code with RC5 on arduino uno:

```
//variable for input number (1..6)
int sel;

#define OPT1led 9
#define OPT2led 10

//IR
#include <IRremote.h>
#define irPin 11
IRrecv irrecv(irPin);
decode_results results;

void setup()
{
  pinMode(OPT1led, OUTPUT);
  pinMode(OPT2led, OUTPUT);

  irrecv.enableIRIn();

}

void loop()
{

if (irrecv.decode(&results))
  {
    
    switch (results.value)
    {
      case 0x803:  // "3"
      sel++;
        break;
      case 0x03:  // "3"
      sel++;
        break;

   
      case 0x801:  // "1"
      sel--;
      break;    

      case 0x01:  // "1"
      sel--;
      break;  
    }

    if(sel < 1)
    sel=1;

    if(sel > 6)
    sel=6;
    

    
    irrecv.resume();
  }

}

  //OPT1//
  if (sel == 1)
  {
    digitalWrite(OPT1led, HIGH);
  }

  //OPT2//
  if (sel == 2)
  {
    digitalWrite(OPT2led, HIGH);
  }
  

}
```

And sometimes its work properly - when I click "1" = value of "sel" is decrease by "1" and when I click "3" value of "sel" is increase by "1".

But .... from time to time, when I press "1" the value is decrease by "2" (from value 4 i get "2", from value 6 i get 4). Also same with pressing "3", from time to time is increase by "2", when value is 3 i get 5

Maybe some ideas?

Have a good day!

gavron04, 2019-04-10:
At same time I have opened irDump on arduino uno (tsop4838) + im testing my controller with tsop4838. And when on controller source is changed by "2" on irDump is one code. So i dont know what is wrong, library?

bengtmartensson, 2019-04-11:
What happens when you press and hold a key on an RC5 remote is that it keeps sending the same signal every 114 ms. So, if your program sees n copies of the 1-key, it will decrement n times. Because that is the way the program was written.

If that is not what you want, you can use the toggle bit creatively. It flips between 0 and 1 on different invocations, so that you can tell a repeated hold from multiple presses. Something like this:

````
static unsigned int old_toggle = 2; // Different from 0 and 1 to always accept first press

void loop() {
    if (irrecv.decode(&results)) {
        unsigned int toggle = (results.value & 0x800U) >> 11;
        unsigned int payload = results.value & (~0x800U);
        if (toggle == oldtoggle) // key held down
            return; // Discard!

        old_toggle = toggle;
        switch (payload) {
               case 0x03U:
                       sel--;
                       break;
               case 0x01U:
                        sel++
                        break;
              // No cases for 0x8**!
                    .....
          }
         ...
}````

gavron04, 2019-04-11:
Thanks for reply. I will test it today.

What happens when I press and hold a key on an RC5 remote:
pressing multiple "1" I get 801;01;801;01;801;01
for holding "1" I get 01;01;01;01;01;01;01;01;01 or 801;801;801;801;801

But You are writing "So, if your program sees n copies of the 1-key, it will decrement n times. Because that is the way the program was written.". Ok, but... I have bug only sometimes, when I have pressed one time "1"  and it changes value by "2". Only by "1" or "2" . Not 3,4,5,6..n times.

gavron04, 2019-04-11:
**bengtmartensson** after test with small changes like:

toggle == oldtoggle > toggle == old_togle

sel++ > sel++;

but code works for first click on button on remote control (changes by "1") and then all freezes. I need to restart power, again, click on remote control, sel is changed by "1" and again freezes.

gavron04, 2019-04-11:
But of course same situation when I have in code:

```
case 0x803:  // "3"
      sel++;
        break;

      case 0x801:  // "1"
      sel--;
      break;    

```
(ony 801;803 without 01;03). its enough to have problem with increase/decrease by "1". Sometimes I have decrease/increase by "2". So i think there is a bug in library.

As i wrote, i have parralel connected arduino uno with tsop4838 with IRdump and my driver with tsop4838. When on my driver "sel" was changed by "2" with one click on remote control on IRdump I have only one code. Not two same.

gavron04, 2019-04-11:
SO my question. Its possible to make code for RC5 which will always decrease/increase by "1" when I press one time on remote control? How to "secure" increase/decrease only by "1" ?

gavron04, 2019-04-11:
How about adding parse for 100ms beetween next code? like this https://github.com/iryont/PGA2320-preamplifier/blob/master/src/remote.c

bengtmartensson, 2019-04-12:
You ask for advice, you get advice, you select to ignore advice. (There would not be any `case 0x803` then). Instead something only you know what it is is claimed not to produce the result you expect.

This is a waste of my time.

gavron04, 2019-04-12:
You dint understand me.

I have tested Your code:

static unsigned int old_toggle = 2U; // Different from 0 and 1 to always accept first press

void loop() {
    if (irrecv.decode(&results)) {
        unsigned int toggle = (results.value & 0x800U) >> 11U; // 0 or 1
        unsigned int payload = results.value & (~0x800U);
        if (toggle == old_toggle) // key held down
            return; // Discard!

        old_toggle = toggle;
        switch (payload) {
               case 0x03U:
                       sel--;
                       break;
               case 0x01U:
                        sel++;
                        break;
}

but code works for first click on button on remote control (changes by "1") and then **all freezes**. I need to restart power, again, click on remote control, sel is changed by "1" and again freezes.

So how I can test it when Your code freezes my device after one click??

bengtmartensson, 2019-04-12:
It was not a full sketch, only an outline, that you were supposed to complete. Your version lacks     `irrecv.resume();` which explains the "freeze". Try this:

````static unsigned int old_toggle = 2U; // Different from 0 and 1 to always accept first press

void loop() {
    if (irrecv.decode(&results)) {
        irrecv.resume();
        unsigned int toggle = (results.value & 0x800U) >> 11U; // 0 or 1
        unsigned int payload = results.value & (~0x800U);
        if (toggle == old_toggle) // key held down
            return; // Discard!

        old_toggle = toggle;
        switch (payload) {
               case 0x03U:
                       sel--;
                       break;
               case 0x01U:
                        sel++;
                        break;
              // No cases for 0x8**!
                    
          }
}````

gavron04, 2019-04-12:
**bengtmartensson**

thanks!!! works perfect :)

I have question about this simple code:

```
static unsigned int old_toggle = 2U; // Different from 0 and 1 to always accept first press

//IR
#include <IRremote.h>
#define irPin 11
IRrecv irrecv(irPin);
decode_results results;

#define OPT1led 9
#define OPT2led 10

int sel = 0;

void setup()
{
  irrecv.enableIRIn();

  pinMode(OPT1led, OUTPUT);
  pinMode(OPT2led, OUTPUT);
}

void loop() {
    if (irrecv.decode(&results)) {
        irrecv.resume();
        unsigned int toggle = (results.value & 0x800U) >> 11U; // 0 or 1
        unsigned int payload = results.value & (~0x800U);
        if (toggle == old_toggle) // key held down
            return; // Discard!

        old_toggle = toggle;
        switch (payload) {
               case 0x03U:
                       sel--;
                       break;
               case 0x01U:
                        sel++;
                        break;
              // No cases for 0x8**!
                    
          }
  if (sel < 1)
      sel = 1;

    if (sel > 6)
      sel = 6;
    }

  //OPT1//
  if (sel == 1)
  {
    digitalWrite(OPT1led, HIGH);
  }

  //OPT2//
  if (sel == 2)
  {
    digitalWrite(OPT2led, HIGH);
  }
}
```

when Im holding button on remote control OPT1 (if value of "sel" is 1) LED or OPT2 (if value of "sel" is 2) is flashing. Something like this:
- holding "1" key (value of "sel" is now "1"
- if "sel" is "1" >  digitalWrite(OPT1led, HIGH); and again (all the time Im holding "1" button) 

The code is executed over and over again. How to stop it? 
- holding "1" key (value of "sel" is now "1"
- if "sel" is "1" then digitalWrite(OPT1led, HIGH) and STOP! Wait for next command from remote control ? With this I think led will be in "ON" state, not flashing during holding key.

closed 2020-10-11 by ArminJo