Arduino-IRremote/Arduino-IRremote #621: Нужен простой скетч

toll101, 2019-04-17:
Дорогие друзья, помогите решить такую задачу: Есть необходимость по команде (внешняя кнопка - например) включить шаговый двигатель. После выполнения заданного числа оборотов (1000) остановить его и включить второй шаговый двигатель с аналогичной задачей (допустим 2000 оборотов). После получения второй команды программа отрабатывает все в обратном порядке. Сначала второй мотор 2000 оборотов в обратном направлении, затем первый. Оба двигателя должны выдавать по 50 Вт мощности. То есть необходимы достаточно мощные контроллеры. На данный момент есть плата Ардуино уно, и два двигателя 42 серии. 
Нужны рекомендации по контроллерам, схема соединения Ардуино, контроллеров, кнопок управления и двигателей. И, конечно скетч программы.

AnalysIR, 2019-04-17:
not related to the IRremote library, please close.

toll101, 2019-04-17:
Я правильно понял, вы не можете мне помочь?
Задача, мне кажется очень простой, для программиста...
Вопросы подбора элементной базы я уже решил.
Нужна только программа для ардуино. 
С уважением и наилучшими пожеланиями.

AnalysIR, 2019-04-17:
You are correct, this is the wrong place for this issue.

maybe try the Arduino forum (Russian Section) via
[Россия - Russia](https://forum.arduino.cc/index.php?board=120.0)

Please close the issue

closed 2020-07-04 by ArminJo