Arduino-IRremote/Arduino-IRremote #623: ESP32 wrong received raw data using IRrecvDumpv2 and IRrecvDump

GiampaMan, 2019-04-21:
Hi!
I'm a beginner. I hope you can help me.
I'm trying to read IR codes from ESP32. The thing is that I'm getting wrong raw data.

This is the wrong raw: 
[57] 500,1600, 450,600, 450,550, 450,600, 450,1600, 450,600, 450,600, 450,550, 450,1600, 500,1600, 500,550, 450,550, 500,550, 500,550, 450,600, 450,550, 500,550, 450,550, 500,600, 450,550, 450,550, 500,1600, 450,600, 450,1600, 550,500, 450,600, 450,600, 450,1600, 450
The correct one is: 
[59] 3096, 9888, 492, 1608, 464, 580, 460, 576, 456, 576, 460, 1640, 460, 560, 484, 552, 468, 564, 484, 1604, 468, 1612, 464, 556, 488, 556, 484, 564, 484, 556, 488, 556, 488, 548, 484, 548, 488, 548, 488, 548, 484, 548, 488, 548, 484, 1604, 468, 552, 488, 1592, 496, 544, 488, 548, 484, 548, 488, 1608, 468

This is the IR code of an air conditioner. 

I hope you can help me.
Thanks!

AnalysIR, 2019-04-21:
Not enough info supplied to assist.

GiampaMan, 2019-04-21:
Thanks for reply. 
Sorry. Here is the code:
```
#include <Arduino.h>

//------------------------------------------------------------------------------
// Include the IRremote library header
//
#include <IRremote.h>

//------------------------------------------------------------------------------
// Tell IRremote which Arduino pin is connected to the IR Receiver (TSOP4838)
//
int recvPin = 17;
IRrecv irrecv(recvPin);

//+=============================================================================
// Configure the Arduino
//
void  setup ( )
{
  Serial.begin(115200);   // Status message will be sent to PC at 9600 baud
  irrecv.enableIRIn();  // Start the receiver
}

//+=============================================================================
// Display IR code
//
void  ircode (decode_results *results)
{
  // Panasonic has an Address
  if (results->decode_type == PANASONIC) {
    Serial.print(results->address, HEX);
    Serial.print(":");
  }

  // Print Code
  Serial.print(results->value, HEX);
}

//+=============================================================================
// Display encoding type
//
void  encoding (decode_results *results)
{
  switch (results->decode_type) {
    default:
    case UNKNOWN:      Serial.print("UNKNOWN");       break ;
    case NEC:          Serial.print("NEC");           break ;
    case SONY:         Serial.print("SONY");          break ;
    case RC5:          Serial.print("RC5");           break ;
    case RC6:          Serial.print("RC6");           break ;
    case DISH:         Serial.print("DISH");          break ;
    case SHARP:        Serial.print("SHARP");         break ;
    case JVC:          Serial.print("JVC");           break ;
    case SANYO:        Serial.print("SANYO");         break ;
    case MITSUBISHI:   Serial.print("MITSUBISHI");    break ;
    case SAMSUNG:      Serial.print("SAMSUNG");       break ;
    case LG:           Serial.print("LG");            break ;
    case WHYNTER:      Serial.print("WHYNTER");       break ;
    case AIWA_RC_T501: Serial.print("AIWA_RC_T501");  break ;
    case PANASONIC:    Serial.print("PANASONIC");     break ;
    case DENON:        Serial.print("Denon");         break ;
  }
}

//+=============================================================================
// Dump out the decode_results structure.
//
void  dumpInfo (decode_results *results)
{
  // Check if the buffer overflowed
  if (results->overflow) {
    Serial.println("IR code too long. Edit IRremoteInt.h and increase RAWBUF");
    return;
  }

  // Show Encoding standard
  Serial.print("Encoding  : ");
  encoding(results);
  Serial.println("");

  // Show Code & length
  Serial.print("Code      : ");
  ircode(results);
  Serial.print(" (");
  Serial.print(results->bits, DEC);
  Serial.println(" bits)");
}

//+=============================================================================
// Dump out the decode_results structure.
//
void  dumpRaw (decode_results *results)
{
  // Print Raw data
  Serial.print("Timing[");
  Serial.print(results->rawlen-1, DEC);
  Serial.println("]: ");

  for (int i = 1;  i < results->rawlen;  i++) {
    unsigned long  x = results->rawbuf[i] * USECPERTICK;
    if (!(i & 1)) {  // even
      Serial.print("-");
      if (x < 1000)  Serial.print(" ") ;
      if (x < 100)   Serial.print(" ") ;
      Serial.print(x, DEC);
    } else {  // odd
      Serial.print("     ");
      Serial.print("+");
      if (x < 1000)  Serial.print(" ") ;
      if (x < 100)   Serial.print(" ") ;
      Serial.print(x, DEC);
      if (i < results->rawlen-1) Serial.print(", "); //',' not needed for last one
    }
    if (!(i % 8))  Serial.println("");
  }
  Serial.println("");                    // Newline
}

//+=============================================================================
// Dump out the decode_results structure.
//
void  dumpCode (decode_results *results)
{
  pinMode(21, LOW);
  // Start declaration
  Serial.print("unsigned int  ");          // variable type
  Serial.print("rawData[");                // array name
  Serial.print(results->rawlen - 1, DEC);  // array size
  Serial.print("] = {");                   // Start declaration

  // Dump data
  for (int i = 1;  i < results->rawlen;  i++) {
    Serial.print(results->rawbuf[i] * USECPERTICK, DEC);
    if ( i < results->rawlen-1 ) Serial.print(","); // ',' not needed on last one
    if (!(i & 1))  Serial.print(" ");
  }

  // End declaration
  Serial.print("};");  // 

  // Comment
  Serial.print("  // ");
  encoding(results);
  Serial.print(" ");
  ircode(results);

  // Newline
  Serial.println("");

  // Now dump "known" codes
  if (results->decode_type != UNKNOWN) {

    // Some protocols have an address
    if (results->decode_type == PANASONIC) {
      Serial.print("unsigned int  addr = 0x");
      Serial.print(results->address, HEX);
      Serial.println(";");
    }

    // All protocols have data
    Serial.print("unsigned int  data = 0x");
    Serial.print(results->value, HEX);
    Serial.println(";");
  }
}

//+=============================================================================
// The repeating section of the code
//
void  loop ( )
{
  decode_results  results;        // Somewhere to store the results

  if (irrecv.decode(&results)) {  // Grab an IR code
    dumpInfo(&results);           // Output the results
    dumpRaw(&results);            // Output the results in RAW format
    dumpCode(&results);           // Output the results as source code
    Serial.println("");           // Blank line between entries
    irrecv.resume();              // Prepare for the next value
  }
}

```

GiampaMan, 2019-04-21:
Do you need any other information?

AnalysIR, 2019-04-21:
Nothing immediately obvious with a **quick** review.

try the other dump example as well.

Also try to press the remote button twice quickly to see if you can get 2 captures

Also, where did you get the 'correct one' ?

GiampaMan, 2019-04-22:
> Nothing immediately obvious with a **quick** review.
> 
> try the other dump example as well.
> 
> Also try to press the remote button twice quickly to see if you can get 2 captures
> 
> Also, where did you get the 'correct one' ?

Hi @AnalysIR . Thanks for being interested in my issue.

The Dumpv2 result is this:
`unsigned int  rawData[57] = {550,1550, 450,600, 500,500, 500,550, 450,1650, 450,550, 450,600, 450,550, 500,1600, 450,1600, 500,550, 450,600, 450,550, 500,550, 500,500, 500,550, 500,550, 500,550, 450,550, 500,550, 450,550, 550,1550, 500,550, 450,1600, 450,600, 550,500, 450,550, 500,1600, 450}; `
The Dump result is this:
`Raw (58): 450 -1600 500 -600 450 -550 450 -600 450 -1600 500 -550 450 -600 450 -550 500 -1600 450 -1600 450 -600 450 -600 450 -600 450 -600 450 -550 500 -550 450 -600 450 -550 500 -550 450 -600 450 -600 450 -1550 500 -600 450 -1600 500 -550 500 -550 450 -550 450 -1650 450`
The correct result is provided by ESP8266 IR library, using Dumpv2 sketch. This is the result:
`uint16_t powerOff[59] = {3096, 9888, 492, 1608, 464, 580, 460, 576, 456, 576, 460, 1640, 460, 560, 484, 552, 468, 564, 484, 1604, 468, 1612, 464, 556, 488, 556, 484, 564, 484, 556, 488, 556, 488, 548, 484, 548, 488, 548, 488, 548, 484, 548, 488, 548, 484, 1604, 468, 552, 488, 1592, 496, 544, 488, 548, 484, 548, 488, 1608, 468};`

The ESP8266 result is the correct one. I tested it tons of times, sending again that IR code, and worked perfectly. I tried sending ESP32 results but nothing happens.

I pressed remote button two times quickly and I got two captures.

Thanks for all
Giampaolo

bengtmartensson, 2019-04-22:
The signal has a fat lead-in and then a rather long gap (9888 micros) gap. The IRremote sketch disposes this first pulse/pause pair. This is the problem.

I was able to reproduce by shooting "the correct one" using IrScrutinizer at an Arduino running IRrecvDumpV2. It turns out (`IRremote.cpp` line 171)  that IRremote disallows IR signals with an internal gap exceeding GAP_TICKS (= 5000 microseconds)!!! AFAIK nowhere documented. 

You can try to increase GAP_TICKS. Anothers solution would be to use my Infrared4Arduino library, available in the Arduino library manager with the name "Infrared".

GiampaMan, 2019-04-22:
Hi @bengtmartensson ! Thanks for your reply! 

Now LG IR off code working. Was gap as you said. Thanks! I modified that parameter on IR library. Also I upgrade buffer to 1024.

Now problem is another, I hope you can help me. Still wrong data but now scanning Panasonic IR code.

Here is the correct Panasonic air off rawData:
`uint16_t powerOff[439] = {3480, 1760, 400, 444, 424, 1308, 420, 444, 424, 444, 424, 440, 424, 440, 424, 444, 420, 444, 420, 444, 424, 444, 448, 444, 420, 444, 420, 448, 420, 1332, 400, 444, 420, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 444, 420, 1332, 400, 1332, 396, 1312, 420, 444, 452, 440, 424, 1308, 420, 448, 420, 444, 420, 444, 420, 448, 420, 444, 420, 444, 424, 440, 424, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 444, 448, 444, 420, 444, 424, 444, 420, 444, 420, 444, 420, 448, 420, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 444, 420, 444, 420, 448, 448, 440, 424, 444, 420, 444, 420, 1336, 396, 1332, 400, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 10008, 3484, 1736, 404, 440, 424, 1308, 448, 448, 420, 444, 420, 444, 424, 440, 424, 444, 420, 444, 424, 440, 424, 444, 420, 444, 424, 444, 420, 444, 420, 1332, 400, 444, 424, 440, 452, 444, 420, 444, 420, 444, 424, 440, 424, 444, 424, 1304, 424, 1332, 400, 1308, 424, 440, 424, 444, 420, 1332, 400, 444, 424, 440, 424, 440, 424, 444, 424, 440, 448, 444, 424, 440, 424, 444, 424, 440, 424, 444, 420, 444, 424, 440, 424, 444, 424, 440, 424, 440, 424, 444, 424, 1308, 420, 1308, 424, 1308, 424, 440, 452, 440, 424, 440, 428, 440, 424, 1308, 424, 1304, 424, 444, 424, 1304, 424, 444, 424, 440, 428, 440, 424, 440, 428, 436, 428, 436, 428, 440, 428, 436, 456, 436, 428, 1304, 424, 1308, 424, 440, 428, 436, 428, 440, 428, 1300, 428, 1304, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 452, 440, 424, 440, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 424, 440, 428, 440, 424, 1308, 424, 1300, 432, 1304, 452, 440, 424, 440, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 1328, 404, 1304, 428, 1304, 424, 440, 428, 436, 428, 440, 452, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 428, 436, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 1304, 452, 440, 428, 436, 428, 1304, 428, 436, 428, 440, 424, 440, 428, 1304, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 452, 440, 428, 436, 428, 436, 428, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 428, 1300, 428, 440, 424, 440, 428, 1300, 428, 440, 428, 436, 456, 1324, 408};`

And here is the wrong RAW:

`unsigned int  rawData[183] = {450,450, 450,400, 400,1350, 400,1300, 450,400, 450,450, 400,450, 450,1300, 400,1300, 450,400, 500,400, 450,400, 450,450, 400,450, 400,450, 450,450, 450,400, 450,450, 400,450, 450,400, 450,450, 450,400, 450,400, 450,450, 450,400, 400,450, 450,450, 450,400, 450,1250, 450,1300, 450,1300, 450,450, 450,400, 400,450, 450,400, 450,450, 450,400, 450,400, 450,450, 450,400, 450,1300, 450,1250, 450,1300, 450,400, 450,400, 500,400, 450,450, 450,400, 450,400, 450,450, 450,400, 450,400, 450,400, 500,400, 450,400, 450,400, 450,450, 450,400, 450,400, 450,1300, 450,450, 450,400, 500,1250, 450,400, 450,400, 450,450, 450,1250, 450,450, 450,400, 450,400, 450,400, 500,400, 450,400, 450,400, 450,400, 500,450, 450,400, 450,400, 450,400, 500,400, 450,400, 450,400, 450,450, 450,400, 450,1300, 450,1250, 450,1300, 450,400, 450,450, 450,400, 450,1300, 450};`

I hope you can help me. Thanks for all!
Giampaolo

bengtmartensson, 2019-04-23:
Nice that one problem was solved. However, it may be somewhat dangerous to "fumble" with undocumented internal parameters like that..

The new one I cannot reproduce. I shoot the "correct Panasonic air off rawData" at an Arduino running (unmodified) IRrecvDumpv2, and get, as expected, the 100 (or so) first numbers, up to measurement errors. (Possibly you fix has side effects?)

BTW, the "correct Pana..." looks kind of implausible to me. IrpTransmogrifier transmogrifies it to the IRP form
````
{432,msb}<1,-1|1,-3>(8,-3,A:64,1,-23,8,-3,B:152,1,-50m){A=0x4004072000000060,B=0x40040720001c34018c00007007000091000049}
````

which not *have* to be wrong, just looks very different from other AC signals.

Have you tried Infrared4Arduino?

GiampaMan, 2019-04-23:
Hello @bengtmartensson 
Thanks for your reply!

Now I'm little bit confused using this code. What do you recommend me to do?
I liked the idea of using your library (Infrared4Arduino), but I don't see compatibility with ESP32. Does it work on this platform?

Thanks for all!
Giampaolo

bengtmartensson, 2019-04-23:
> What do you recommend me to do?

Depends on what your ultimate goal is, which you have never told us... In particular, does your final solution require reading or "learning", or is that something that can be done once and for all, and not needed for deployment? 

> I liked the idea of using your library (Infrared4Arduino), but I don't see compatibility with ESP32. Does it work on this platform?

Presently no. I wanted to create a solid low-level framework for the hardware-related low-level stuff #516 that both this and my library can use, but that appears to have been a waste of time :-(.

GiampaMan, 2019-04-23:
Hi @bengtmartensson 

I'm really happy that you are interested on my issue and you are helping me! Thanks you so much! 

As you said, my final project needs to be capable of "learning", so is important to read these RAW codes without problems. I think Panasonic IR signal is complex, but maybe is something possible to do changing some part of code (I don't understand about this code).
I used IR remote library for ESP8266 and  scanning there worked great! As far as I know is modified and have some upgrades at difference of this one library. Problem is that I can't use ESP8266 library on ESP32 because is not optimized.
What do you think I can do?

I really appreciate your help.
Giampaolo.

bengtmartensson, 2019-04-24:
I am not sure why

> my final project needs to be capable of "learning"

would you enlighten me? Best thing would be if you get a sufficiently deep understanding of the IR protocol to be able to generate the desired IR signals on the fly, say for setting temperature (in F or C) and all the other parameters. There are tons of different articles on the internet, for example at AnalysIR site. Possibly some will fit to your device? Or you learn the sort of questions to ask. [post 36 and following](http://www.remotecentral.com/cgi-bin/mboard/rc-discrete/thread.cgi?4894,3) are interesting.  Then you would not need "learning".

There were some patches for the esp32 (partially by myself) for IRremote, one or two years ago.

GiampaMan, 2019-05-11:
Hi @bengtmartensson !

Sorry about delays, was bit busy, but now I'm here. Thanks for be interested in helping me!
Still I have issues with reading Panasonic IR code on ESP32.
When I read IR codes on my Arduino UNO I don't use a library, I simple use this sample code from @AnalysIR that really works cool:

```
#include <Arduino.h>
/*
Author: AnalysIR
Revision: 1.0

This code is provided to overcome an issue with Arduino IR libraries
It allows you to capture raw timings for signals longer than 255 marks & spaces.
Typical use case is for long Air conditioner signals.

You can use the output to plug back into IRremote, to resend the signal.

This Software was written by AnalysIR.

Usage: Free to use, subject to conditions posted on blog below.
Please credit AnalysIR and provide a link to our website/blog, where possible.

Copyright AnalysIR 2014

Please refer to the blog posting for conditions associated with use.
http://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

Connections:
IR Receiver      Arduino
V+          ->  +5v
GND          ->  GND
Signal Out   ->  Digital Pin 2
(If using a 3V Arduino, you may connect V+ to +3V)
*/

#define LEDPIN 13
//you may increase this value on Arduinos with greater than 2k SRAM
#define maxLen 800

volatile unsigned int irBuffer[maxLen]; //stores timings - volatile because changed by ISR
volatile unsigned int x = 0;            //Pointer thru irBuffer - volatile because changed by ISR

void setup()
{
  Serial.begin(115200);                               //change BAUD rate as required
  attachInterrupt(0, rxIR_Interrupt_Handler, CHANGE); //set up ISR for receiving IR signal
}

void loop()
{
  // put your main code here, to run repeatedly:

  Serial.println(F("Press the button on the remote now - once only"));
  delay(5000); // pause 5 secs
  if (x)
  {                             //if a signal is captured
    digitalWrite(LEDPIN, HIGH); //visual indicator that signal received
    Serial.println();
    Serial.print(F("Raw: (")); //dump raw header format - for library
    Serial.print((x - 1));
    Serial.print(F(") "));
    detachInterrupt(0); //stop interrupts & capture until finshed here
    for (int i = 1; i < x; i++)
    { //now dump the times
      if (!(i & 0x1))
        Serial.print(F("")); ////////
      Serial.print(irBuffer[i] - irBuffer[i - 1]);
      Serial.print(F(", "));
    }
    x = 0;
    Serial.println();
    Serial.println();
    digitalWrite(LEDPIN, LOW);                          //end of visual indicator, for this time
    attachInterrupt(0, rxIR_Interrupt_Handler, CHANGE); //re-enable ISR for receiving IR signal
  }
}

void rxIR_Interrupt_Handler()
{
  if (x > maxLen)
    return;                 //ignore if irBuffer is already full
  irBuffer[x++] = micros(); //just continually record the time-stamp of signal transitions
}

```
Problem is that I don't know how to translate it to ESP32. Maybe this one should work.

What do you think?
Thanks for all!

Regards, Giampaolo.

bengtmartensson, 2019-05-12:
> This Software was written by AnalysIR.

I think someone else is summoned here... ;-)

The present project contains experimental esp32, but not in the default master branch. Possibly it is in the branch "beta" (not to be confused with beta releases) or it hangs in a PR... There should be a separate file(s) esp32.h and/or esp32.cpp. (I was involved myself.)

There will be a day when Infrared4Arduino will support esp32...

AnalysIR, 2019-05-12:
@GiampaMan If you need assistance with our sketch, it might be better to post on our **[IRforum ](https://irforum.analysir.com/)** as it is not directly related to an issue with this IRremote library.

_I have just uploaded a new release of the sketch on our blog, which may address your particular issue (The sketch was initially written for original Arduino that only had External interrupts on D2 & D3. The latest release is more generic and supports most other Arduino platforms now.)_

GiampaMan, 2019-05-15:
> > This Software was written by AnalysIR.
> 
> I think someone else is summoned here... ;-)
> 
> The present project contains experimental esp32 support, but not in the default master branch. Possibly it is in the branch "beta" (not to be confused with beta releases) or it hangs in a PR... There should be a separate file(s) esp32.h and/or esp32.cpp. (I was involved myself.)
> 
> There will be a day when Infrared4Arduino will support esp32...

Hi @bengtmartensson !
Thanks for your reply!

Do you mean there is another code in beta phase in another branch, that experimentally supports ESP32? In that case, where can I find it?
What is a PR?

I'm happy to hear from you that your library will support ESP32! That's awesome!
When do you plan to release that version of your library? 

Thanks for all!

Regards,
Giampaolo

GiampaMan, 2019-05-15:
> @GiampaMan If you need assistance with our sketch, it might be better to post on our **[IRforum ](https://irforum.analysir.com/)** as it is not directly related to an issue with this IRremote [...]

Hi @AnalysIR ! 
Thanks for your reply!

I was checking your blog, but I don't know where do you uploaded this code. Could you please send the link here?

Regards,
Giampaolo

bengtmartensson, 2019-05-15:
> Do you mean there is another code in beta phase in another branch, that experimentally supports ESP32? In that case, where can I find it?

Exactly. Branch [beta](https://github.com/z3t0/Arduino-IRremote/tree/beta). Checkout with (for example)

````
git clone git@github.com:z3t0/Arduino-IRremote.git
git checkout beta
````
> What is a PR?

Pull Request, see [the Git book](https://git-scm.com/book/en/v2). The ones in the present project found [here](https://github.com/z3t0/Arduino-IRremote/pulls). You can search for "esp32" among them. See in particular #425 and #437.

> I'm happy to hear from you that your library will support ESP32! That's awesome!
> When do you plan to release that version of your library?

Nice to hear. It has been planned for "next month" for the last two years or so ;-).

GiampaMan, 2019-06-06:
Hi @bengtmartensson 

Thanks again for all your help!

I'm still trying to make my ESP32 work correctly. I downloaded beta library but still experiencing problems, code didn't compile yet. I'm so confused and a little bit sad that still I cannot use ESP32 on IR applications correctly. I really need to make ESP32 work. Do you know about any upgrade to the information or new codes?

Thanks!

Regards,
Giampaolo

bengtmartensson, 2019-06-07:
The only thing I know that I have not told you yet is that I have a [forked version of the library](https://github.com/bengtmartensson/Arduino-IRremote/). Possibly there is something. (Be sure to examine branches too.)

Otherwise, if you show the compilation problems, there is a slim chance that someone can help...

Having 3 unprocessed PRs, each now two years old, I have given up on trying to contribute to this project ;-).

ArminJo, 2020-07-04:
ESP is supported by the current Version.

closed 2020-07-04 by ArminJo