Arduino-IRremote/Arduino-IRremote #624: problem sending NEC codes to 3rd-party devices

pille, 2019-04-25:
**Board:** ARDUINO NANO
**Library Version:** 2.2.3
**Protocol:** NEC

i'm trying to use this library to send IR codes to 3rd-party devices (RGB-LED, radio, TV) and have used the `IRrecv` functions to get the codes from their original remotes before.
the problem is that none of them seem to work when i send them via `IRsend`

they work though for self-made devices, that use `IRrecv`. so i'm pretty sure the IR-LED is working and distance  and angle are ok.

i investigated the raw signals using the `IRrecvDump` example. here's a dump of the code 0xFF18E7 (32bit):

original remote:
```
Raw (68): 8900 -4500 650 -500 600 -550 600 -550 600 -550 550 -600 600 -500 650 -500 600 -550 600 -1650 600 -1650 650 -1650 600 -1650 600 -1650 650 -1650 600 -1650 600 -1650 650 -550 550 -600 600 -550 600 -1650 600 -1650 600 -550 600 -550 600 -550 600 -1650 600 -1650 650 -1600 650 -550 600 -500 600 -1700 600 -1650 600 -1650 600
```

`IRsend` clone:
```
Raw (68): 8850 -4500 500 -600 550 -600 500 -600 500 -600 550 -600 500 -600 550 -600 500 -600 500 -1750 450 -1750 500 -1750 500 -1750 500 -1700 550 -1700 550 -1700 500 -1750 500 -600 500 -600 550 -600 500 -1700 550 -1700 550 -600 500 -600 500 -600 550 -1700 500 -1750 550 -1700 500 -600 500 -600 500 -1750 500 -1750 500 -1750 500
```

i noticed the tiny variations in timing. could they be the culprit for my devices not recognizing the codes?
how much do they usually tolerate? i mean, the leading pulse should be 9000µs.
positive feedback on using `IRsend.sendNEC()` successfully with 3rd-party hardware is also welcome.

AnalysIR, 2019-04-25:
If your code is not working...might be a good idea to post it! 

IR receivers can vary the timing by up to +/- 200uSecs...but circa  +/- 100 or less is more common to see

pille, 2019-04-25:
i don't think the code is not working, as i've shown its result here. it's really just the `IRsend.sendNEC()`.
given your timing tolerance, the raw data looks ok, but the IRsend version differs slightly more.
did you ever successfully controlled a device with that function?

bengtmartensson, 2019-04-26:
I would like to make a few comments:

1. Most receiver-software, probably also in your devices, have a fairly large tolerance, and will accept a fair amount of error. ("You" select the tolerance parameter as high as possible, without accepting signals that are clearly something else.) It is unlikely that your device rejected the signal for that reason.

2. Receiving devices are often programmed to require a certain numbers of repeats, and are rejecting sends with too few repeats. Exactly what is your "3rd-party device"?

pille, 2019-04-26:
@bengtmartensson: i already tried repeating the signal. actually that's what my test code boiled down to:  send one code in `loop()`.

the devices i've tested are:
- RGB LED light bulb
- radio (20yo)
- TV (10yo)

cmaxmedia, 2019-06-12:
Hello, i have also troubles with a NEC Device. Receiving the code for Power On/Off works fine but resend the RAW Data again will not work. Input are the same on Arduino Mega 2560 and Artuino Nano:

**Dump from Arduino Mega 2560/Pin 11:**
Encoding  : NEC
Code      : 1FE39C6 (32 bits)
Timing[67]: 
     +9000, -4400     + 600, - 550     + 600, - 550     + 550, - 550
     + 600, - 550     + 550, - 550     + 550, - 550     + 600, - 550
     + 550, -1700     + 550, -1700     + 550, -1650     + 600, -1650
     + 600, -1650     + 550, -1650     + 600, -1700     + 550, -1700
     + 550, - 550     + 600, - 500     + 600, - 550     + 550, -1700
     + 550, -1700     + 550, -1650     + 600, - 550     + 550, - 550
     + 600, -1650     + 600, -1650     + 550, -1700     + 550, - 550
     + 600, - 550     + 550, - 550     + 600, -1650     + 550, -1700
     + 550, - 550     + 600
unsigned int  rawData[67] = {9000,4400, 600,550, 600,550, 550,550, 600,550, 550,550, 550,550, 600,550, 550,1700, 550,1700, 550,1650, 600,1650, 600,1650, 550,1650, 600,1700, 550,1700, 550,550, 600,500, 600,550, 550,1700, 550,1700, 550,1650, 600,550, 550,550, 600,1650, 600,1650, 550,1700, 550,550, 600,550, 550,550, 600,1650, 550,1700, 550,550, 600};  // NEC 1FE39C6
unsigned int  data = 0x1FE39C6;

**sending over IRsendDemo on PWM Pin 3:**
irsend.sendNEC(0x1FE39C6, 32);

**sending over IRsendRawDemo:**
  int khz = 38; // 38kHz carrier frequency for the NEC protocol
  unsigned int  irSignal[67] = {9000,4400, 600,550, 600,550, 550,550, 600,550, 550,550, 550,550, 600,550, 550,1700, 550,1700, 550,1650, 600,1650, 600,1650, 550,1650, 600,1700, 550,1700, 550,550, 600,500, 600,550, 550,1700, 550,1700, 550,1650, 600,550, 550,550, 600,1650, 600,1650, 550,1700, 550,550, 600,550, 550,550, 600,1650, 550,1700, 550,550, 600};  // NEC 1FE39C6
  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.

checking the IR Code on my Arduino Nano it shows and unknown device.
can someone help please?
thnx Markus

AnalysIR, 2019-06-12:
Please open a new issue - to avoid confusion and wasting time reading all the previous posts.

Might also be a good idea to delete your post here, as it is not directly related.

Once you do all that I should be able to point you in the right direction.

cmaxmedia, 2019-06-13:
Using a receiver for sending could probably go wrong - so everything is fine ;)
thank you! 
#facepalm

closed 2020-07-26 by ArminJo