Arduino-IRremote/Arduino-IRremote #625: Issue sending codes from ATTiny85

xyx0826, 2019-05-04:
**Board:** ATTiny85
**Library Version:** 2.3.3, 2.4.0
**Protocol:** Sony, NEC +

**Code Block:**
```cpp
#include <Arduino.h>
#include <IRremote.h>

IRsend _irSend;

void setup()
{
  OSCCAL = 0x8A;
}

void loop()
{
  for (int i = 0; i < 3; i++)
  {
    _irSend.sendSony(0xa90, 12);
    _irSend.custom_delay_usec(40000);
  }
  _irSend.custom_delay_usec(1000000);
}
```

**Expected**
An IR LED at pin PB1 will be driven to send IR signal bursts every second.

**Actual**
The IR LED sends random codes in unknown protocol.

My receiver is also an ATTiny85 designed to print any code received with a VS1838B sensor at 38kHz onto software serial. I can confirm the receiver works with actual remotes.

I can confirm the processor clock on the transmitter being correct by uploading a blinking sketch for the same pin `PB1` as the IR pin. Both `delay()` and `IRsend.custom_delay_usec()` work as intended.

With the code above, an IR LED at pin `PB1` will cause the receiver to print random code in -1 (unknown) protocol. when I replace the IR LED with a visible LED it appears to be constantly on at reduced power or pulsing at very high frequency. The 1000ms delay is not observed. After an unspecified amount of time it completely turns off. Frequently the transmitted signal is received as `632192736`.

xyx0826, 2019-05-05:
The following code is intended to send a signal at the end of each second, twice.
```cpp
#include <Arduino.h>
#include <IRremote.h>

IRsend _irSend;

void setup()
{
  OSCCAL = 0x8A;
  delay(1000);
  _irSend.sendSony(0xa90, 12);
  delay(1000);
  _irSend.sendSony(0xa90, 12);
}

void loop() { }
```
It looks like only the first `delay()` worked. The receiver prints a random value once, then nothing afterwards.

AnalysIR, 2019-05-05:
Have you considered if your IR emmitter circuit, might be an issue?

IR remote will always try to give a result...so the random signal is more likely just a bad signal.

other comments/possibilities:
- Those VS1828Bs are not great &  I would never consider using them (But as you say it works for you with actual remotes)
- How do you know theOSCCAL is correct?
- How do you know your receiver worked with actual remotes?
- How do you know that the AVR fuses are set correctly for the '85 (osc clock speed etc)

xyx0826, 2019-05-05:
My circuit is currently simply the uC powered at 3.3V with an IR LED chained with a 330Ω resistor and attached to pin PB1.

- I agree VS1838Bs can perform poorly but it seems so far I haven't had any issues with them.
- I acquired the OSCCAL values with [TinyTuner](https://github.com/rambo/arduino-tiny/blob/master/libraries/TinyTuner/examples/Interactive_to_Serial_with_Details/Interactive_to_Serial_with_Details.pde); I don't have an oscilloscope. I had another project involving precise PWM width measurement and could confirm the TinyTuner calibration values work for that.
- The remote is actually an Android app making use of my phone's IR transceiver. It works with actual appliances, and when I put it to Sony mode and use it on my ATTiny receiver I see consistent value readings in the correct protocol.
- The same values worked for a LED blinking sketch and the PWM project I mentioned above. They are `E:FF, H:D7, L:E2`.

The most unexplainable thing to me was the `delay()` that doesn't seem to work; both `delay()` and `custom_delay_usec()` seems to work **before** the first send call. After that, they doesn't work anymore. In the first code example it's supposed to wait for 1000ms before sending the next IR burst, but what I saw is a continuous stream of bad signal.

AnalysIR, 2019-05-05:
It could be that the tiny core you are using is conflicting its Arduino internal timer with IRremote.

IRremote uses Timer0 on the '85, so that may be confilicting with the timer being used on the core you are using???

Could you try to use another Arduino core for the ATtiny85?

Alternatively....
Instead of delay(), you could try the: (AVR built in functions for delay, search for details)

void _delay_ms | ( | double | __ms | )  function, which does not use a clock(instruction cycles instead)

you will also need (at the top)
#define F_CPU 8000000UL  // 1 MHz
#include <util/delay.h>

xyx0826, 2019-05-05:
Thanks, I think you're right on the timer conflict. `_delay_ms()` seems to work. However I don't think it has the right timing...

I've changed the transmitter code to use JVC:
```cpp
#define F_CPU 8000000UL

#include <IRremote.h>
#include <util/delay.h>

IRsend _irSend;

void setup()
{
  OSCCAL = 0x8A;
}

void loop()
{
  _irSend.sendJVC(0xFFFFFFUL, 32, false);
  _delay_ms(1000);
  _irSend.sendJVC(0x0000FFUL, 32, false);
  _delay_ms(2000);
}
```

And it became apparent that in a loop, the interval between two signals is clearly longer than 1000ms. The interval between loops is clearly longer than 2000ms. From the transmitter the signals sent are still invalid.
Interestingly if I don't put in any `IRsend` code (i.e. only using `digitalWrite()`), `_delay_ms()` works as intended.

AnalysIR, 2019-05-05:
try

 ```
 _irSend.sendJVC(0xFFFFFFUL, 32, false);
noInterrupts();//turn off interrupts for time delays
  _delay_ms(1000);
interrupts();
  _irSend.sendJVC(0x0000FFUL, 32, false);
noInterrupts();//turn off interrupts for time delays
  _delay_ms(2000);
interrupts();

```
The IRremote timer/interrupts might be messing with the _delay_ms (not sure?? as you dont have the receiver activated) or could even be the Arduino clock itself....worth a try

Another basic  check is to blink a LED every 10 or 60 secs and time it to see if it is close to the PC clock or other timer.

Forgot to mention earlier, your fuses seem OK, after a quick glance.

xyx0826, 2019-05-05:
That works! Thank you very much. The delays are now working as intended.
Though the receiver still spits out random values. I thought it might be helpful to post its code here:
```cpp
#include <IRremote.h>
#include <util/delay.h>
#include <SoftwareSerial.h>

SoftwareSerial _serial(-1, PB2);
char _message[32];

IRrecv _irRecv(PB1);
decode_results _results;

void setup()
{
  OSCCAL = 0x4F;
  _serial.begin(9600);
  _irRecv.enableIRIn();
}

void loop()
{
  if (_irRecv.decode(&_results))
  {
    _irRecv.resume();
    sprintf(_message, "Received %lu in %d", _results.value, _results.decode_type);
    _serial.println(_message);
  }
}
```
I've confirmed again that it gives consistent results with my Android remote.
I can rule out `SoftwareSerial` interference by comparing the received value with `0xAAFFFFUL` and blinking a LED on another pin upon receiving a larger value. The JVC code should cause the LED to blink on the first send, not both, which is what happened.
I tried changing the `sprintf()` to:
```cpp
sprintf(_message, "%lu: Received %lu in %d", millis(), _results.value, _results.decode_type);
```
To my surprise `millis()` always returns zero. Maybe that's part of the problem?

I have tried 60 second blinks on both transmitter and receiver. They work as intended.
I'm using [SpenceKonde's](https://github.com/SpenceKonde/ATTinyCore) core; that's the default for PlatformIO IDE. I can't use Arduino official IDE since this core gives a compiler segfault.

AnalysIR, 2019-05-05:
For what you are doing, it would be advisable to have a Scope or Logic Analyzer.
You can pick up a cheap, yet very good 8 channel LA for <$10 and an even better 16ch one for ~$30, via Aliexpress.

Useful tools to have when needed, like in this case.

_Did you consider using a Nano pro mini or micro....also very cheap via AE & can be made to work in low power mode if needed._

Best of luck, I dont think I have any further suggetions for you.

xyx0826, 2019-05-05:
Thank you very much for your help. I guess I'll try swapping hardware and see if it helps. I should definitely get an analyzer.
Closing this issue now unless I have further updates.

closed 2019-05-05 by xyx0826

xyx0826, 2019-05-06:
A new discovery: I replaced `custom_delay_usec()` implementation with:
```cpp
void IRsend::custom_delay_usec(unsigned long uSecs) {
  while (uSecs--)
  {
      _delay_us(1);
  }
}
```
And now the receiver gives consistent values! Still in protocol `-1`, but at least possible as a workaround.