Arduino-IRremote/Arduino-IRremote #626: How to work only with hone protocol?

gavron04, 2019-05-08:
Hello. I have question. How to change library or something in code to works only with one protocol? For example Im using Apple remote (NEC) and sometimes when I press some keys on SONY remote (RC6) my arduino changes state like it was signal from Apple Remote.

Regards

AnalysIR, 2019-05-08:
Sounds like you need to fix the issue, which is most likely in your own code.

Unfortunately, not enough info provided for more assistance than this.

 _(I presume you read the text you deleted when opening a new issue.)_

gavron04, 2019-05-08:
Ok. I will add today code.

gavron04, 2019-05-08:
Atmega16 (arduino IDE + export to hex with mightycore library)

`//EEPROM
#include <EEPROM.h>
int lastInput;

//prawo lewo zrodla
int sel = 0;

//IR
#include <IRremote.h>
#define irPin 11
IRrecv irrecv(irPin);
decode_results results;

const int OPT1sw = 18;
const int OPT2sw = 19;
const int OPT3sw = 20;
const int COAXsw = 21;
const int USBsw = 22;
const int BTsw = 23;

int statusOPT1sw = 0;
int statusOPT2sw = 0;
int statusOPT3sw = 0;
int statusCOAXsw = 0;
int statusUSBsw = 0;
int statusBTsw = 0;

//LEDy//
#define OPT1led 9
#define OPT2led 10
#define OPT3led 8
#define COAXled 31
#define USBled 14
#define BTled 30

//AK4118
#define IPS0 4
#define IPS1 3
#define ISOusb 0
#define MUXbt 1
#define MUXak 2

//RGB//
int redPin = 12;
int greenPin = 13;
int bluePin = 15;
//RGB//
int potPin_red = A7;
int potPin_green = A6;
int potPin_blue = A5;
//RGB//
int readValue_red;
int readValue_green;
int readValue_blue;
//RGB//
int writeValue_red;
int writeValue_green;
int writeValue_blue;

void setup()
{
  //odczyt ostatniego zrodla//
  getLastInput();

  irrecv.enableIRIn();

  //PRZYCISKI//
  pinMode(OPT1sw, INPUT_PULLUP);
  pinMode(OPT2sw, INPUT_PULLUP);
  pinMode(OPT3sw, INPUT_PULLUP);
  pinMode(COAXsw, INPUT_PULLUP);
  pinMode(USBsw, INPUT_PULLUP);
  pinMode(BTsw, INPUT_PULLUP);

  //LEDy//
  pinMode(OPT1led, OUTPUT);
  pinMode(OPT2led, OUTPUT);
  pinMode(OPT3led, OUTPUT);
  pinMode(COAXled, OUTPUT);
  pinMode(USBled, OUTPUT);
  pinMode(BTled, OUTPUT);

  //MUXy//
  pinMode(MUXak, OUTPUT);
  pinMode(MUXbt, OUTPUT);

  //IZOLATOR//
  pinMode(ISOusb, OUTPUT);

  //AK4118//
  pinMode(IPS0, OUTPUT);
  pinMode(IPS1, OUTPUT);

  //RGB//
  pinMode(potPin_red, INPUT);
  pinMode(potPin_green, INPUT);
  pinMode(potPin_blue, INPUT);
  //RGB//
  pinMode(redPin, OUTPUT);
  pinMode(bluePin, OUTPUT);
  pinMode(greenPin, OUTPUT);
}

void ledsoff()
{
  digitalWrite(OPT1led, LOW);
  digitalWrite(OPT2led, LOW);
  digitalWrite(OPT3led, LOW);
  analogWrite(COAXled, LOW);
  digitalWrite(USBled, LOW);
  analogWrite(BTled, LOW);
}

void opt1source()
{
  digitalWrite(IPS0, LOW);
  digitalWrite(IPS1, LOW);
  digitalWrite(ISOusb, HIGH);
  digitalWrite(MUXbt, LOW);
  digitalWrite(MUXak, HIGH);
}

void opt2source()
{
  digitalWrite(IPS0, HIGH);
  digitalWrite(IPS1, LOW);
  digitalWrite(ISOusb, HIGH);
  digitalWrite(MUXbt, LOW);
  digitalWrite(MUXak, HIGH);
}

void opt3source()
{
  digitalWrite(IPS0, LOW);
  digitalWrite(IPS1, HIGH);
  digitalWrite(ISOusb, HIGH);
  digitalWrite(MUXbt, LOW);
  digitalWrite(MUXak, HIGH);
}

void coaxsource()
{
  digitalWrite(IPS0, HIGH);
  digitalWrite(IPS1, HIGH);
  digitalWrite(ISOusb, HIGH);
  digitalWrite(MUXbt, LOW);
  digitalWrite(MUXak, HIGH);
}

void btsource()
{
  digitalWrite(IPS0, HIGH);
  digitalWrite(IPS1, HIGH);
  digitalWrite(ISOusb, HIGH);
  digitalWrite(MUXbt, HIGH);
  digitalWrite(MUXak, LOW);
}

void usbsource()
{
  digitalWrite(IPS0, HIGH);
  digitalWrite(IPS1, HIGH);
  digitalWrite(ISOusb, LOW);
  digitalWrite(MUXbt, LOW);
  digitalWrite(MUXak, LOW);
}

void loop()
{
  //ODCZYT PRZYCISKOW//
  statusOPT1sw = digitalRead(OPT1sw);
  statusOPT2sw = digitalRead(OPT2sw);
  statusOPT3sw = digitalRead(OPT3sw);
  statusCOAXsw = digitalRead(COAXsw);
  statusUSBsw = digitalRead(USBsw);
  statusBTsw = digitalRead(BTsw);

  //RGB//
  analogWrite(redPin,  analogRead(potPin_red) / 4);
  analogWrite(greenPin, analogRead(potPin_green) / 4);
  analogWrite(bluePin, analogRead(potPin_blue) / 4);

  //OPT1//
  if (statusOPT1sw == LOW)
  {
    sel = 1;
  }

  //OPT2//
  if (statusOPT2sw == LOW)
  {
    sel = 2;
  }

  //OPT3//
  if (statusOPT3sw == LOW)
  {
    sel = 3;
  }

  //COAX//
  if (statusCOAXsw == LOW)
  {
    sel = 4;
  }

  //USB//
  if (statusUSBsw == LOW)
  {
    sel = 5;
  }

  //BT//
  if (statusBTsw == LOW)
  {
    sel = 6;
  }

  if (irrecv.decode(&results))
  {
    switch (results.value)
    {
      
      case 0x77E16001:  //prawo
      case 0x77E1E0F3:  //prawo
        sel++;
        break;

      case 0x77E19001:  //lewo
      case 0x77E110F3:  //lewo
        sel--;
        break;

    }

    //zakres 1-6//
    if (sel < 1)
      sel = 1;

    if (sel > 6)
      sel = 6;

    irrecv.resume();
  }

  //OPT1//
  if (sel == 1)
  {
    ledsoff();
    opt1source();
    digitalWrite(OPT1led, HIGH);
    EEPROM.update(0, 1);
  }

  //OPT2//
  if (sel == 2)
  {
    ledsoff();
    opt2source();
    digitalWrite(OPT2led, HIGH);
    EEPROM.update(0, 2);
  }

  //OPT3//
  if (sel == 3)
  {
    ledsoff();
    opt3source();
    digitalWrite(OPT3led, HIGH);
    EEPROM.update(0, 3);
  }

  //COAX//
  if (sel == 4)
  {
    ledsoff();
    coaxsource();
    digitalWrite(COAXled, HIGH);
    EEPROM.update(0, 4);
  }

  //USB//
  if (sel == 5)
  {
    ledsoff();
    usbsource();
    digitalWrite(USBled, HIGH);
    EEPROM.update(0, 5);
  }

  //BT//
  if (sel == 6)
  {
    ledsoff();
    btsource();
    digitalWrite(BTled, HIGH);
    EEPROM.update(0, 6);
  }

}

void getLastInput()

{
  lastInput = EEPROM.read(0);

  if (lastInput == 1)
  {
    ledsoff();
    opt1source();
    digitalWrite(OPT1led, HIGH);
  }

  if (lastInput == 2)
  {
    ledsoff();
    opt2source();
    digitalWrite(OPT2led, HIGH);
  }

  if (lastInput == 3)
  {
    ledsoff();
    opt3source();
    digitalWrite(OPT3led, HIGH);
  }

  if (lastInput == 4)
  {
    ledsoff();
    coaxsource();
    digitalWrite(COAXled, HIGH);
  }

  if (lastInput == 5)
  {
    ledsoff();
    usbsource();
    digitalWrite(USBled, HIGH);
  }

  if (lastInput == 6)
  {
    ledsoff();
    btsource();
    digitalWrite(BTled, HIGH);
  }

}`

AnalysIR, 2019-05-08:
Add in a test to check the protocol type.
**_results->decode_type_**
Ignore any signal that is not NEC.

Refer to the provided examples, to see how this is all used.

[ Arduino-IRremote/examples/IRrecvDumpV2/IRrecvDumpV2.ino ](https://github.com/z3t0/Arduino-IRremote/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino)

**_Note: Please tidy up you last post by puuting all of the code inside code tags (see menu)_**

gavron04, 2019-05-08:
(Protocol type its RC6 (SONY TV) , but on driver I need only NEC. I will check hot to ignore any signal thats not NEC)

gavron04, 2019-05-08:
```
  if (results->decode_type == NEC)
  {
    switch (results.value)
    {
      
      case 0x77E16001:  //prawo
      case 0x77E1E0F3:  //prawo
        sel++;
        break;

      case 0x77E19001:  //lewo
      case 0x77E110F3:  //lewo
        sel--;
        break;

    }

    //zakres 1-6//
    if (sel < 1)
      sel = 1;

    if (sel > 6)
      sel = 6;

    irrecv.resume();
  }
```

Something like this?

AnalysIR, 2019-05-08:
Yes.

However, put the **irrecv.resume();** outside the if statement or rx will stop forever when you receive any non-NEC signal (e.g.  SONY or even noise)

gavron04, 2019-05-08:
Thanks :) Last question. WIth original code from my post when I receive IR signal, LED on my device (which shows actual input) its blinking. I have same situation on others devices with this IRremote library. When I have 7segment LCD or some LEDs on device and I will press some keys on IR 7segment LCD or LEDs is blinking. Its possible to solve this?

AnalysIR, 2019-05-08:
Unfortunately, this is not related to IRremote.
Try posting on the Arduino forum.

Alternatively, try to debug the issue yourself, by inserting Serial.prints in your code.

z3t0, 2019-05-09:
I am adding a "Documentation" label, with the hopes of writing tutorials based on issues marked with this label.

gavron04, 2019-05-27:
Hi all.

After tests:

```
void loop()
{
   if (results.decode_type == NEC)
  {

    enc_change = 1;
    sel_change = 1;

    switch (results.value)
    {

      // VOL+
      case 0x77E1E000: //NEC
      case 0x77E16001: //NEC
      case 0x77E1E0F3: //NEC
      case 0x810: // RC5
      case 0x10:  // RC5
        enc ++;
        digitalWrite(K7, LOW);
        break;

      // VOL-
      case 0x77E11000: //NEC
      case 0x77E19001: //NEC
      case 0x77E110F3: //NEC
      case 0x811: //RC5
      case 0x11: //RC5
        enc --;
        digitalWrite(K7, LOW);
        break;

      //sel--
      case 0x77E1B000: //NEC
      case 0x77E13001: //NEC
      case 0x77E1B0F3: //NEC
        sel --;
        break;

      //sel++
      case 0x77E1D000: //NEC
      case 0x77E15001: //NEC
      case 0x77E1D0F3: //NEC
        sel ++;
        break;

      //vol 25 //menu
      case 0x77E14000: //NEC
      case 0x77E1C001: //NEC
      case 0x77E140F3: //NEC
        enc = 25;
        break;

      //vol 00 //play-pause
      case 0x77E17A00: //NEC
      case 0x77E17AF3: //NEC
      case 0x77E1FA01: //NEC
        enc = 00;
        break;

    }

    if (enc < 0)
      enc = 0;
    if (enc > 63)
      enc = 63;

    if (sel < 0)
      sel = 0;
    if (sel > 3)
      sel = 3;

    irrecv.resume();

  }
```

and this:

```
void loop()
{
    if (results.decode_type == NEC)
  {

    enc_change = 1;
    sel_change = 1;

    switch (results.value)
    {

      // VOL+
      case 0x77E1E000: //NEC
      case 0x77E16001: //NEC
      case 0x77E1E0F3: //NEC
      case 0x810: // RC5
      case 0x10:  // RC5
        enc ++;
        digitalWrite(K7, LOW);
        break;

      // VOL-
      case 0x77E11000: //NEC
      case 0x77E19001: //NEC
      case 0x77E110F3: //NEC
      case 0x811: //RC5
      case 0x11: //RC5
        enc --;
        digitalWrite(K7, LOW);
        break;

      //sel--
      case 0x77E1B000: //NEC
      case 0x77E13001: //NEC
      case 0x77E1B0F3: //NEC
        sel --;
        break;

      //sel++
      case 0x77E1D000: //NEC
      case 0x77E15001: //NEC
      case 0x77E1D0F3: //NEC
        sel ++;
        break;

      //vol 25 //menu
      case 0x77E14000: //NEC
      case 0x77E1C001: //NEC
      case 0x77E140F3: //NEC
        enc = 25;
        break;

      //vol 00 //play-pause
      case 0x77E17A00: //NEC
      case 0x77E17AF3: //NEC
      case 0x77E1FA01: //NEC
        enc = 00;
        break;

    }

    if (enc < 0)
      enc = 0;
    if (enc > 63)
      enc = 63;

    if (sel < 0)
      sel = 0;
    if (sel > 3)
      sel = 3;

  }

    irrecv.resume();
}
```

cant work.

Work only:

```
void loop()
{

  if (irrecv.decode(&results))

  {

    enc_change = 1;
    sel_change = 1;

    switch (results.value)
    {

      // VOL+
      case 0x77E1E000: //NEC
      case 0x77E16001: //NEC
      case 0x77E1E0F3: //NEC
      case 0x810: // RC5
      case 0x10:  // RC5
        enc ++;
        digitalWrite(K7, LOW);
        break;

      // VOL-
      case 0x77E11000: //NEC
      case 0x77E19001: //NEC
      case 0x77E110F3: //NEC
      case 0x811: //RC5
      case 0x11: //RC5
        enc --;
        digitalWrite(K7, LOW);
        break;

      //sel--
      case 0x77E1B000: //NEC
      case 0x77E13001: //NEC
      case 0x77E1B0F3: //NEC
        sel --;
        break;

      //sel++
      case 0x77E1D000: //NEC
      case 0x77E15001: //NEC
      case 0x77E1D0F3: //NEC
        sel ++;
        break;

      //vol 25 //menu
      case 0x77E14000: //NEC
      case 0x77E1C001: //NEC
      case 0x77E140F3: //NEC
        enc = 25;
        break;

      //vol 00 //play-pause
      case 0x77E17A00: //NEC
      case 0x77E17AF3: //NEC
      case 0x77E1FA01: //NEC
        enc = 00;
        break;

    }

    if (enc < 0)
      enc = 0;
    if (enc > 63)
      enc = 63;

    if (sel < 0)
      sel = 0;
    if (sel > 3)
      sel = 3;

    irrecv.resume();

  }
```

Also when i put "irrecv.resume(); outside, then IR stop working.

I have changed "->" to " . " because arduino have problem "base operand of '->' has non-pointer type 'decode_results' arduino".

So "    if (results.decode_type == NEC " is not working. (for all protocols).

Some idea?

Thanks a lot!

AnalysIR, 2019-05-30:
For NEC style signals....the repeat code is reported by IRremote as FFFFFFFF. You can use that as the repeat signal, if you have previously received a valid NEC style signal. It is left up to the user to provide the code to handle this.

gavron04, 2019-05-30:
Sounds interesting. I will test.

But how about chaning codes after pressing "menu" + "ok" on apple remote (original and fake) for about 6sec and then i have different codes?

AnalysIR, 2019-05-30:
Your post is not clear to me.

Some buttons may only repeat if it makes sense. For example vol+, vol-, up down left right CH+, CH-. For other buttons repeat actions may not make sense and thus the manufacturer may not issue a repeat signal when kept pressed down.

NEC sends a repeat signal instead of the full signal when the button is kept pressed. This is to save battery usage by sending a shorter signals for the repeat. Most other devices repeat the full signal and some others do not repeat the signal at all when the button is kept preseed.

AnalysIR, 2019-05-30:
Sounds like it may a feature of the remotes and not an issue with the IRremote library.

what is the purpose of pressing menu+OK for 6 seconds?

Sphinix1, 2019-06-21:
I have the codes for my AV(pioneer vsx1021) but its in a HEX form 
eg : Power On is  = "0000   0068 0000 0022 0168 00B4 0016 0043 0016 0016 0016 0043 0016 0016 0016 0016   0016 0043 0016 0016 0016 0043 0016 0016 0016 0043 0016 0016 0016 0043 0016   0043 0016 0016 0016 0043 0016 0016 0016 0016 0016 0016 0016 0043 0016 0043   0016 0043 0016 0016 0016 0016 0016 0016 0016 0043 0016 0043 0016 0016 0016   0016 0016 0016 0016 0043 0016 0043 0016 0043 0016 03DD";

is there a way to convert it to NEC

also the Data sheet also has another Code format  **A51C** for same Power On function. which one can i use

AnalysIR, 2019-06-21:
@Sphinix1 You should really open a new issue as your query is not related to this issue and will waste a lot of time for other users by needlessly confusing things.

There is a simple solution to your query, but I will defer answering until you open your own/new issue.

Sphinix1, 2019-06-21:
> @Sphinix1 You should really open a new issue as your query is not related to this issue and will waste a lot of time for other users by needlessly confusing things.
> 
> There is a simple solution to your query, but I will defer answering until you open your own/new issue.

lol, my Bad

closed 2020-08-29 by ArminJo