Arduino-IRremote/Arduino-IRremote #627: RAWBUF is in IRremote.h

z3t0, 2019-05-09:
https://github.com/z3t0/Arduino-IRremote/blob/2dc97b29f2da74e342a308f6e99f153e132ce06a/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L69

The line above prints out a misleading error message.

It should be asking the user to modify RAWBUF in IRremote.h

Thank you to a user who informed me of the issue

AnalysIR, 2019-05-09:
FYI: That message originally referred to RAWLEN & along with RAWBUF were in IRremoteInt.h. Looks like it was missed when the files were re-arranged.  :)

z3t0, 2019-05-09:
Yep, that explains it ahah!

jtannenba, 2019-05-30:
Question:  Looking at the codes from shutter release remotes for Nikon and Pentax, they both come us with a 32 bit code, but the raw data is only 3 blips for Nikon with a code of 22AE7A29, and 7 blips for Pentax with a code of 8DCDAE60.  They come back "Unknown" as decode type.  How does the library come up with the 32 bit codes that it does?
[Raw_Printout.txt](https://github.com/z3t0/Arduino-IRremote/files/3234904/Raw_Printout.txt)

Thank you for any illumination...This has drove me nuts.
btw the Raw seen by just an IR transistor is basically same as the decoded output.

AnalysIR, 2019-05-30:
> I don't think this should have gone here, but didn't see anywhere else to comment. Sorry if in the wrong place.

You are correct..this is the wrong place.

You should open a **[new issue](https://github.com/z3t0/Arduino-IRremote/issues/new)**.

ArminJo, 2020-07-03:
Fixed in current version

closed 2020-07-03 by ArminJo