Arduino-IRremote/Arduino-IRremote #628: IR Receiver not receive right  raw code that is send by IR Emitter

shivansh9675, 2019-05-11:
###  I am use this code to send raw data
#include <IRremote.h>
IRsend irsend;
void setup()
{

}

void loop() {
  int khz = 38; // 38kHz carrier frequency for the NEC protocol
  unsigned int irSignal[] = {9000, 4500, 560, 560, 560, 560, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 1690, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 39416, 9000, 2210, 560}; //AnalysIR Batch Export (IRremote) - RAW
  
  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.

  delay(5000); //In this example, the signal will be repeated every 5 seconds, approximately.
}
### Use this code for receive that raw dat
#include <IRremote.h>
int RECV_PIN = 11;
IRrecv irrecv(RECV_PIN);
IRsend irsend;
decode_results results;
void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver

}

// Storage for the recorded code
int codeType = -1; // The type of code
unsigned long codeValue; // The code value if not raw
unsigned int rawCodes[RAWBUF]; // The durations if raw
int codeLen; // The length of the code
int toggle = 0; // The RC5/6 toggle state

// Stores the code for later playback
// Most of this code is just logging
void storeCode(decode_results *results) {
  codeType = results->decode_type;
  int count = results->rawlen;

  if (codeType == UNKNOWN) {
    Serial.println("Received unknown code, saving as raw");
    codeLen = results->rawlen - 1;
    // To store raw codes:
    // Drop first value (gap)
    // Convert from ticks to microseconds
    // Tweak marks shorter, and spaces longer to cancel out IR receiver distortion
    for (int i = 1; i <= codeLen; i++) {
      if (i % 2) {
        // Mark
        rawCodes[i - 1] = results->rawbuf[i]*USECPERTICK - MARK_EXCESS;
        Serial.print(" m");
      } 
      else {
        // Space
        rawCodes[i - 1] = results->rawbuf[i]*USECPERTICK + MARK_EXCESS;
        Serial.print(" s");
      }
      Serial.print(rawCodes[i - 1]);
    }
    Serial.println("");
  }

}

void loop() {
  // If button pressed, send the code.
  
 if (irrecv.decode(&results)) {
  
    storeCode(&results);
    delay(300);
    irrecv.resume(); // resume receiver
    
  }
  
}

IN output it will not receive right raw code that is send by emitter.sometime it will receive NEC code some time diff. code

AnalysIR, 2019-05-12:
> sometime it will receive NEC code some time diff. code

might be a good idea to post information about this.
For example, it could be that you are just receiving repeat NEC signals (which happens when you keep the button on the remote pressed).

Why not test with the examples provided first, until you are happy it is working. Only then..add in your own code.

harsha7addanki, 2020-05-28:
I'm Having The same Problem this is my Code:
`
#include <IRremote.h>
//#include <nRF24L01.h>
//#include <RF24.h>
//#include <SPI.h>

//RF24 radio(13,10);
//const uint64_t pipe = 0xE6E6E6E6E6E6;
IRrecv irrecv(6);
decode_results results;

void setup() {
  // put your setup code here, to run once:
  //radio.begin();
  //radio.openWritingPie(pipe);
  Serial.begin(9600);
  irrecv.enableIRIn();
}

void loop() {
  // put your main code here, to run repeatedly:
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
  }
}
`
Help!

AnalysIR, 2020-05-28:
Please open a new issue and delete the one above.

closed 2020-07-04 by ArminJo