Arduino-IRremote/Arduino-IRremote #629: Weird outputs of camera shutter remotes.

jtannenba, 2019-05-30:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Sony (if any)

**Code Block:**
```c

#include <IRremote.h> 

Question: Looking at the codes from shutter release remotes for Nikon and Pentax with IRrcvdumpV2. they both come us with a 32 bit code, but the raw data is only 3 blips for Nikon with a code of 22AE7A29, and 7 blips for Pentax with a code of 8DCDAE60. They come back "Unknown" as decode type. How does the library come up with the 32 bit codes that it does?
Dumps: 
[Raw_Printout.txt](https://github.com/z3t0/Arduino-IRremote/files/3237545/Raw_Printout.txt)

Thank you for any illumination...This has drove me nuts.
btw the Raw seen by just an IR transistor on an oscilloscope, is basically same as the decoded output above

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2019-06-02:
IRremote always tries to provide a HEX value, regardless. So for your situation just ignore this and use the RAW signal only.

Well done on using code tags...but you put the text inside the code tags which makes your post very difficult to read. Might be a good idea to tidy it up a bit.

jtannenba, 2019-06-02:
How does IRremote decide the HEX values?  They are consistent with the device. (not random).
This photo shows IR remote receiver output for Pentax that the library interprets as 8DCDAE60.  The Raw signal to the IR remote receiver corresponds to the output.
[image]

AnalysIR, 2019-06-02:
It first tries to decode every enabled IR protocol until it finds one. If it can't find one that decodes correctly, it then creates a HASH value, which should be consistent for most signals received. (There are some signals that may not provide a unique hash code). So when the protocol is reported as unknown, the HEX value provided is the calculated HASH.

For more details  check out the actual code.

jtannenba, 2019-06-02:
Thank you.  I'm 76 and my coding years are way back to the early nineties, so I'm more than a little rusty at following CPP code.  I was trained in C before object orienting came along.  I did try to look at some of the code but got myself lost.
Thank you for replying.

jtannenba, 2019-06-02:
Found this in IRrecv.cpp:
http://arcfn.com/2010/01/using-arbitrary-remotes-with-arduino.html

Ken explains it well.

Thanks again.

closed 2019-06-02 by jtannenba