Arduino-IRremote/Arduino-IRremote #630: Problem sending raw codes with attiny85

jtannenba, 2019-06-16:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Raw - This code works perfectly with arduino uno or nano, but trying to install in attiny85 to make a small remote.  The number of pulses look correct, but it's not riding on 38k frequency.  Can the attiny85 do this, and I'm not sure what I am doing wrong.  I did try to play with the fuses to no avail.
Thank you

**Code Block:**
```c

#define F_CPU 1600000UL
#include <IRremote.h>

IRsend irsend;

unsigned int  shutterOn[15] = {13150,2800, 1200,800, 1100,900, 1050,900, 1100,850, 1050,950, 1000,950, 1050};
void setup()
{
  
}

void loop() {
 
    irsend.sendRaw(shutterOn,15,38);
    delay(1000);
}
.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2019-06-16:
You have the Board flagged as UNO...but then refer to ATtiny85.

Make the delay 5 seconds when testing.

Yes it can do this. However, as you have not provided many additional details it is not possible to assist much further.

eg. Crystal or internal osc.....Fuse settings....pins used....device you are trying to control (model# of device & remote)...posting details about how you know it is not 38kHz might be useful

Looking at the timings after the first 2, they should probably all be 1000. (Timings are distorted by IR receivers & IRremote)

l740416, 2019-06-17:
I believe the current implementation for attiny85 in IRremote is using timer0 which is reserved by arduino kernel to implement functions such as delay(). Instead of using timer0, we need to switch to timer1.
I forked the project and modified attiny85 part to use timer1. It should work on both recv and send routines. https://github.com/l740416/Arduino-IRremote

jtannenba, 2019-06-17:
Should I delete my present copy of IRremote before loading yours?  (Still pretty new at the Arduino thing.)

l740416, 2019-06-17:
For windows, the libraries are installed under C:\Users\[USER]\Documents\Arduino\libraries
You can directly replace the library there

jtannenba, 2019-06-17:
Will try when I get some time...Thanks

AnalysIR, 2019-06-26:
Don't see anything obvious with the fuses (as long as you know what frequency you are operating at? You have PLL 64MHz & DIV/8 set). It is possible that there are other clock related parameters set in the code/core.

The brown out is set at 4.3V. So you should be running at 5V??? (in which case using a crystal @16MHz might make sense)

No obvious connection between 833Hz & 38KHz

I suggest you print out **F_CPU** to confirm what the core thinks the clock freq is.

Then toggle a LED on a pin every 1, 5 or 10 seconds to see that the clock is correct.

TIP: As its possible to get an Arduino clone delivered from China for Less than $2, There seems little value in using ATtinys. The larger ATMega328s can also be made to run at low power - as low as 1uA or less.

jtannenba, 2019-06-26:
Thank you.  How do you print out F_CPU?  I also decided a nano or bare Arduino is probably small enough and less frustration.  I did think about putting an external 16 MHz crystal on it, but it would use up two pins. bu probably would have enough pins.  Thanks for your input.

AnalysIR, 2019-06-26:
just like any other number Serial,print(F_CPU);

However if you don't have a softserial connection on the ATtiny, then you can just test against what you thing it is and blink a LED if it matches.

Something like a nano is the way to go.

l740416, 2019-06-26:
I believe Attiny85 has internal 16MHz PLL option.

jtannenba, 2019-06-26:
> I believe Attiny85 has internal 16MHz PLL option.  I tried the fuses to select that, but still didn't work correctly.

AnalsIR--Thanks for the info.

All...I believe I'll close this and go with a nano.  Thanks again for the info

closed 2019-06-26 by jtannenba