Arduino-IRremote/Arduino-IRremote #631: How to send RC5 signal with different address.

gavron04, 2019-06-17:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** RC5

Hello. I have question. How to send RC5 code with for example "30" address?

Have a good day!

Regards,
gavron04

AnalysIR, 2019-06-17:
`irsend.RC5(30,x) or irsend.RC5(0x30,x) `

the first is 30 decimal and the second is 30 in HEX

where x is the number of bits (in your case for 30 it should be at least 6 bits

Tip: NEC is probably an easier protocol to use, It is also @ 38kHz which is the most common receiver. (RC^ is at 36kHz)

gavron04, 2019-06-17:
Thanks.

About tip - right. I see always 38kHz IR receiver. So NEC protocol will be better.

irsend.sendNEC(0x30 , 0xEF30CF, 32);

AnalysIR, 2019-06-17:
> irsend.sendNEC(0x30 , 0xEF30CF, 32);

this is wrong, instead try 

`irsend.sendNEC(0x30 , 32);`

gavron04, 2019-06-18:
But:
0x30 - address
0xEF30CF - code for "1" (0xFF30CF: 1)
32 - bits

with 
`irsend.sendNEC(0x30 , 32);`
I have only address , and bits, right?

Or I dont need to put information about bits and its ok to send only address with value like this:
`irsend.sendNEC(0x30 , 0xFF30CF);`

AnalysIR, 2019-06-18:
if you want to send 0xFF30CF

use  `irsend.sendNEC(0xFF30CF,32);`

if you want to send 0x30

use  `irsend.sendNEC(0x30,32);`

gavron04, 2019-06-18:
I would like to send 0xFF30CF (its "1" button on remote control) but with different andress which has a lot of remote control from China. 

So I need send 0xFF30CF, but with different address (new)

AnalysIR, 2019-06-18:
Do you know hte device you are trying to control and/or the model number?

gavron04, 2019-06-18:
Im making my own device with own remote control.

AnalysIR, 2019-06-18:
Then you can use any number for any commant

like 0x1....for button 1, 0x2 for button 2 etc

You do not have to think about command or address....just as long as your sender & receiver agree on the numbers.

gavron04, 2019-06-18:
Yes, I know. I have tested yesterday command like "111" ; "222" ; "333" with NEC protocol. But anyway its possible to send standard commands NEC like this:

CH- | 0xFFA25D
-- | --
CH | 0xFF629D
CH+ | 0xFFE21D
PREV | 0xFF22DD
NEXT | 0xFF02FD
PLAY/PAUSE | 0xFFC23D
VOL- | 0xFFE01F
VOL+ | 0xFFA857
EQ | 0xFF906F
0 | 0xFF6897
100+ | 0xFF9867
200+ | 0xFFB04F
1 | 0xFF30CF
2 | 0xFF18E7
3 | 0xFF7A85
4 | 0xFF10EF
5 | 0xFF38C7
6 | 0xFF5AA5
7 | 0xFF42BD
8 | 0xFF4AB5
9 | 0xFF52AD

with different address? with IRremote?

AnalysIR, 2019-06-18:
As I said it only matters if the sender and receiver aggree on the numbers mapped to the commands.

If you really need it you could use 

0x123430CF for 1
0x123418E7 for 2
etc

PS: I am assuming this is for non-commercial/private use.

gavron04, 2019-06-18:
Ok, thanks.

DIY projects.

closed 2019-06-18 by gavron04