Arduino-IRremote/Arduino-IRremote #635: Pronto

Sphinix1, 2019-06-21:
I have the codes for my AV(pioneer vsx1021) but its in a HEX form
eg : Power On is = "0000 0068 0000 0022 0168 00B4 0016 0043 0016 0016 0016 0043 0016 0016 0016 0016 0016 0043 0016 0016 0016 0043 0016 0016 0016 0043 0016 0016 0016 0043 0016 0043 0016 0016 0016 0043 0016 0016 0016 0016 0016 0016 0016 0043 0016 0043 0016 0043 0016 0016 0016 0016 0016 0016 0016 0043 0016 0043 0016 0016 0016 0016 0016 0016 0016 0043 0016 0043 0016 0043 0016 03DD";

is there a way to convert it to NEC

also the Data sheet also has another Code format A51C for same Power On function. which one can i use

AnalysIR, 2019-06-21:
is there a way to convert it to NEC

Yes, you could write some code to do it.

However, There is a **sendPronto** function in IRremote that you could use.

The code you posted is an NEC signal and seems to have a carrier frequency of 40kHz instead of 38kHz.
(cahnge the first 0068 to 006D and it should be fine.

Your NEC signal decodes to Hex: A55A38C7

bengtmartensson, 2019-06-21:
You  can paste the string into IrScrutinizer, and it decodes to NEC (NEC2 really), but with modulation frequency 39.9 kHz. (This is sometimes called "Pioneer protocol"). Parameters D=165,F=28. This coincides pretty nicely with the "A51C"-statement, since A5 is hexadecimal form of 165, and 1c is hexadecimal for 28. To translate to "IRremote form", note that IRremote expresses all parameters most significant bit first, while "all the rest of the world" (including NEC the firm (now Renesas)) takes them least significant bit first. Thus, it translates to A55A38C7 as per above. So you should be able to easily send it with IRremote. Use 40 kHz.

Having said that, the `sendPronto` approach above is not good advice. The given string is a 359 characters long, taking 360 bytes to store (of RAM!), just for doing a computation that comes out  the same every time, and can just as well be done on a PC. I also have severe doubts over the quality of the coding of the function...

Sphinix1, 2019-06-22:
> > is there a way to convert it to NEC
> 
> Yes, you could write some code to do it.
> 
> However, There is a **sendPronto** function in IRremote that you could use.
> 
> The code you posted is an NEC signal and seems to have a carrier frequency of 40kHz instead of 38kHz.
> (change the first 0068 to 006D and it should be fine.
> 
> Your NEC signal decodes to Hex: A55A38C7

sendPronto is not really working well in my code.

**Main Code**
`// IR_WiFi_esp -- Remote control for TV, DVD, etc.
//

// Wiring:
// NodeMCU  GPIO    Pin to program  Connect to

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <stdio.h>
#include <string.h>
#include <FS.h>
#include <ArduinoOTA.h>
//#include "ircodes.h"
extern "C"
{
  #include "user_interface.h"
}

//******************************************************************************************
// Some definitions                                                                        *
//******************************************************************************************
// Debug buffer size
#define DEBUG_BUFFER_SIZE 100
// Version number
#define VERSION "20-jul-2017"
// Output pins
#define IR_PIN D5
#define RF_PIN D6

//******************************************************************************************
// Forward declaration of methods                                                          *
//******************************************************************************************
char* dbgprint( const char* format, ... ) ;

//******************************************************************************************
// Global data section.                                                                    *
//******************************************************************************************
bool            myDEBUG = true ;          // Enable or disable debug output
String          ssid ;                    // SSID of selected WiFi network
WiFiServer      server(80) ;              // A server on port 80
AsyncWebServer  cmdserver ( 80 ) ;        // Instance of embedded webserver
IRsend          irsend ( IR_PIN ) ;       // Object for IR sender
bool            cmdreq = false ;          // Command received by server
String          argument[5] ;             // Arguments in command received by server

uint16_t PIONEERS[]={};

//******************************************************************************************
// End of lobal data section.                                                              *
//******************************************************************************************

//******************************************************************************************
//                                  D B G P R I N T                                        *
[... 582 lines cut ...]
      var xhr = new XMLHttpRequest() ;
      xhr.onreadystatechange = function() {
        if ( xhr.readyState == XMLHttpRequest.DONE )
        {
 resultstr.value = xhr.responseText ;
        }
      }
      xhr.open ( "GET", theUrl, true ) ;
      xhr.send ( null ) ;
      document.getElementById ( 'tick' ).play() ;
    } 

    // Button (area) of Samsung remote control pressed
 
	    function pklik(a)
    {
      var hexcode = pcode[a] ;	// Hexcode to be send

     
        httpGet ( "?pioneer&" + hexcode + "&" + a ) ;
     
    }
</script>
</body>
</html>`

Sphinix1, 2019-06-22:
> Yes, you could write some code to do it.
>
> However, There is a **sendPronto** function in IRremote that you could use.
>
> The code you posted is an NEC signal and seems to have a carrier frequency of [...]

do you have any such code? , sorry this is my first IR project

Sphinix1, 2019-06-22:
> You can paste the string into IrScrutinizer, and it decodes to NEC (NEC2 really), but with modulation frequency 39.9 kHz. (This is sometimes called "Pioneer protocol"). Parameters D=165,F=28. This [...]

is there a code to convert directly to NEC eg A55A38C7

bengtmartensson, 2019-06-22:
> is there a code to convert directly to NEC eg A55A38C7

Parse error. Possibly you mean this:

[image]

Sphinix1, 2019-06-22:
> > is there a code to convert directly to NEC eg A55A38C7
> 
> Parse error. Possibly you mean this:
> 

i get this error, is the payload (A55A38C7) constant for all this ProntoHex code for my pioneer model

[
[image]
](url)

bengtmartensson, 2019-06-22:
> i get this error, 

The error message is not *that* bad... If you look closer at the screendump...

Sphinix1, 2019-06-22:
> > i get this error,
> 
> The error message is not _that_ bad... If you look closer at the screendump...

what do recommend i do. my code compiles and runs successfully but it does not seem to send the command . Can you kindly look at it ;if you have the time

bengtmartensson, 2019-06-22:
What code?

Sphinix1, 2019-06-22:
> > > is there a way to convert it to NEC
> > 
> > 
> > Yes, you could write some code to do it.
> > However, There is a **sendPronto** function in IRremote that you could use.
> > The code you posted is an NEC signal and seems to have a carrier frequency of 40kHz instead of 38kHz.
> > (change the first 0068 to 006D and it should be fine.
> > Your NEC signal decodes to Hex: A55A38C7
> 
> sendPronto is not really working well in my code.
> 
> **Main Code**
> `// IR_WiFi_esp -- Remote control for TV, DVD, etc.
> //
> 
> // Wiring:
> // NodeMCU GPIO Pin to program Connect to
> 
> #include <IRremoteESP8266.h>
> #include <IRsend.h>
> #include <ESP8266WiFi.h>
> #include <ESPAsyncTCP.h>
> #include <ESPAsyncWebServer.h>
> #include <stdio.h>
> #include <string.h>
> #include <FS.h>
> #include <ArduinoOTA.h>
> //#include "ircodes.h"
> extern "C"
> {
> #include "user_interface.h"
> }
> 
> //******************************************************************************************
> // Some definitions *
> //******************************************************************************************
> // Debug buffer size
> #define DEBUG_BUFFER_SIZE 100
> // Version number
> #define VERSION "20-jul-2017"
> // Output pins
> #define IR_PIN D5
> #define RF_PIN D6
> 
> //******************************************************************************************
> // Forward declaration of methods *
> //******************************************************************************************
> char* dbgprint( const char* format, ... ) ;
> 
> //******************************************************************************************
> // Global data section. *
> //******************************************************************************************
> bool myDEBUG = true ; // Enable or disable debug output
> String ssid ; // SSID of selected WiFi network
> WiFiServer server(80) ; // A server on port 80
> AsyncWebServer cmdserver ( 80 ) ; // Instance of embedded webserver
> IRsend irsend ( IR_PIN ) ; // Object for IR sender
> bool cmdreq = false ; // Command received by server
> String argument[5] ; // Arguments in command received by server
> 
> uint16_t PIONEERS[]={};
> 
> //******************************************************************************************
> // End of lobal data section. *
> //******************************************************************************************
> 
> //******************************************************************************************
> // D B G P R I N T *
> //******************************************************************************************
> // Send a line of info to serial output. Works like vsprintf(), but checks the BEDUg flag.*
[... 429 lines cut ...]
> {
> resultstr.value = xhr.responseText ;
> }
> }
> xhr.open ( "GET", theUrl, true ) ;
> xhr.send ( null ) ;
> document.getElementById ( 'tick' ).play() ;
> }
> 
> ```
> // Button (area) of Samsung remote control pressed
> 
>     function pklik(a)
> {
>   var hexcode = pcode[a] ;	// Hexcode to be send
> 
>  
>     httpGet ( "?pioneer&" + hexcode + "&" + a ) ;
>  
> }
> ```
> 
> </script> `

this

closed 2020-07-04 by ArminJo