Arduino-IRremote/Arduino-IRremote #639: Support for new Arduino Nano Every (ATMEGA4809) planned?

seisfeld, 2019-06-28:
Hi,

I tried to compile the `IRrecvDumpV2` example on the new [Arduino Nano Every](https://store.arduino.cc/nano-every) (ATMEGA4809) and get the following errors. Are you planning support for this new board?

Thanks
Stephan

### with register emulation for ATMEGA328

```
Arduino: 1.8.9 (Mac OS X), Board: "Arduino Nano Every, ATMEGA328"

Build-Optionen wurden verändert, alles wird neu kompiliert
In file included from /Users/stephan/Projects/arduino/libraries/IRremote/IRremoteInt.h:111:0,
                 from /Users/stephan/Projects/arduino/libraries/IRremote/IRremote.h:24,
                 from /Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:1:
/Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp: In member function 'void IRsend::mark(unsigned int)':
/Users/stephan/Projects/arduino/libraries/IRremote/boarddefs.h:208:30: error: 'TCCR2A' was not declared in this scope
 #define TIMER_ENABLE_PWM    (TCCR2A |= _BV(COM2B1))
                              ^
/Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:69:2: note: in expansion of macro 'TIMER_ENABLE_PWM'
  TIMER_ENABLE_PWM; // Enable pin 3 PWM output
  ^~~~~~~~~~~~~~~~
/Users/stephan/Projects/arduino/libraries/IRremote/boarddefs.h:208:30: note: suggested alternative: 'TCB2'
 #define TIMER_ENABLE_PWM    (TCCR2A |= _BV(COM2B1))
                              ^
/Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:69:2: note: in expansion of macro 'TIMER_ENABLE_PWM'
  TIMER_ENABLE_PWM; // Enable pin 3 PWM output
  ^~~~~~~~~~~~~~~~
In file included from /Users/stephan/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/io.h:99:0,
                 from /Users/stephan/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/pgmspace.h:90,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/String.h:30,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/Print.h:24,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/Stream.h:25,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/Client.h:22,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/ArduinoAPI.h:29,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/Arduino.h:23,
                 from /Users/stephan/Projects/arduino/libraries/IRremote/IRremoteInt.h:24,
                 from /Users/stephan/Projects/arduino/libraries/IRremote/IRremote.h:24,
                 from /Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:1:
/Users/stephan/Projects/arduino/libraries/IRremote/boarddefs.h:208:44: error: 'COM2B1' was not declared in this scope
 #define TIMER_ENABLE_PWM    (TCCR2A |= _BV(COM2B1))
                                            ^
/Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:69:2: note: in expansion of macro 'TIMER_ENABLE_PWM'
  TIMER_ENABLE_PWM; // Enable pin 3 PWM output
  ^~~~~~~~~~~~~~~~
In file included from /Users/stephan/Projects/arduino/libraries/IRremote/IRremoteInt.h:111:0,
                 from /Users/stephan/Projects/arduino/libraries/IRremote/IRremote.h:24,
                 from /Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:1:
/Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp: In member function 'void IRsend::space(unsigned int)':
/Users/stephan/Projects/arduino/libraries/IRremote/boarddefs.h:209:30: error: 'TCCR2A' was not declared in this scope
 #define TIMER_DISABLE_PWM   (TCCR2A &= ~(_BV(COM2B1)))
                              ^
/Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:81:2: note: in expansion of macro 'TIMER_DISABLE_PWM'
  TIMER_DISABLE_PWM; // Disable pin 3 PWM output
  ^~~~~~~~~~~~~~~~~
/Users/stephan/Projects/arduino/libraries/IRremote/boarddefs.h:209:30: note: suggested alternative: 'TCB2'
 #define TIMER_DISABLE_PWM   (TCCR2A &= ~(_BV(COM2B1)))
                              ^
/Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:81:2: note: in expansion of macro 'TIMER_DISABLE_PWM'
  TIMER_DISABLE_PWM; // Disable pin 3 PWM output
  ^~~~~~~~~~~~~~~~~
In file included from /Users/stephan/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/io.h:99:0,
                 from /Users/stephan/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/pgmspace.h:90,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/String.h:30,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/Print.h:24,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/Stream.h:25,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/Client.h:22,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/api/ArduinoAPI.h:29,
                 from /Users/stephan/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/Arduino.h:23,
[... 539 lines cut ...]
  TIMER_CONFIG_KHZ(khz);
  ^~~~~~~~~~~~~~~~
/Users/stephan/Projects/arduino/libraries/IRremote/boarddefs.h:217:42: error: 'CS20' was not declared in this scope
  TCCR2B               = _BV(WGM22) | _BV(CS20); \
                                          ^
/Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:119:2: note: in expansion of macro 'TIMER_CONFIG_KHZ'
  TIMER_CONFIG_KHZ(khz);
  ^~~~~~~~~~~~~~~~
In file included from /Users/stephan/Projects/arduino/libraries/IRremote/IRremoteInt.h:111:0,
                 from /Users/stephan/Projects/arduino/libraries/IRremote/IRremote.h:24,
                 from /Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:1:
/Users/stephan/Projects/arduino/libraries/IRremote/boarddefs.h:218:2: error: 'OCR2A' was not declared in this scope
  OCR2A                = pwmval; \
  ^
/Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:119:2: note: in expansion of macro 'TIMER_CONFIG_KHZ'
  TIMER_CONFIG_KHZ(khz);
  ^~~~~~~~~~~~~~~~
/Users/stephan/Projects/arduino/libraries/IRremote/boarddefs.h:219:2: error: 'OCR2B' was not declared in this scope
  OCR2B                = pwmval / 3; \
  ^
/Users/stephan/Projects/arduino/libraries/IRremote/irSend.cpp:119:2: note: in expansion of macro 'TIMER_CONFIG_KHZ'
  TIMER_CONFIG_KHZ(khz);
  ^~~~~~~~~~~~~~~~
exit status 1
```

AnalysIR, 2019-06-28:
Typically new boards/platforms are added when contributed by the community. As this is a very recently announced platform & the timer definitions are different from traditional AVRs, it may take some time. 

I expect it is just a matter of updating a few definitions in the library to cater for the new timer types  available.

Other than the additional Flash & Sram available...not a lot should change. _(Although the CCL support may introduce some interesting possibilities for IR remote Control beyond what IRremote offers)_

seisfeld, 2019-06-28:
Thanks for the insight.

ArminJo, 2020-07-04:
Atmega4809 is now supported.

closed 2020-07-04 by ArminJo