Arduino-IRremote/Arduino-IRremote #640: Error compiling on IR

innvoation, 2019-06-29:
**Board:** ARDUINO WEMOS D1 R2
**Library Version:I IRremote 2.1.0

**Code Block:**
#include "IRremote.h"
/*
 * IRremote: IRrecvDemo - demonstrates receiving IR codes with IRrecv
 * An IR detector/demodulator must be connected to the input RECV_PIN.
 * Version 0.1 July, 2009
 * Copyright 2009 Ken Shirriff
 * http://arcfn.com
 */

int RECV_PIN = 2;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  // In case the interrupt driver crashes on setup, give a clue
  // to the user what's going on.
  Serial.println("Enabling IRin");
  irrecv.enableIRIn(); // Start the receiver
  Serial.println("Enabled IRin");
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  delay(100);
}
-------------------------------------------------------

I cant't compile
I tried some but failed,
1.I try Update libraries > https://github.com/z3t0/Arduino-IRremote/releases/tag/2.1.0
2.Delete libraries "RobotIRremot"

 log as below
Arduino: 1.8.9 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (3M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 9600"

C:\Users\user\Documents\Arduino\libraries\IRremote\IRremote.cpp:21:27: fatal error: avr/interrupt.h: No such file or directory

Multiple libraries were found for "IRremote.h"
 #include <avr/interrupt.h>

 Used: C:\Users\user\Documents\Arduino\libraries\IRremote
                           ^

 Not used: C:\Users\user\Desktop\arduino-1.8.9\libraries\IRremote
compilation terminated.

exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

AnalysIR, 2019-06-29:
The answer is in the **readme**  !!

> _ESP8266 is supported in a fork based on an old codebase that isn't as recent, but it works reasonably well given that perfectly timed sub millisecond interrupts are different on that chip. See https://github.com/markszabo/IRremoteESP8266_
 
I believe the ESP8266 fork has improved a lot since the above text was written.

innvoation, 2019-06-30:
Reference your comment is solve this issue
Thanks your great support for issue.

closed 2019-06-30 by innvoation