Arduino-IRremote/Arduino-IRremote #641: Library will NOT compile on the UNO WiFi Rev 2

cpapir, 2019-07-02:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Sony (if any)

**Code Block:**
```c

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2019-07-02:
Yes, this is to be expected.

...because this board uses the ATMEGA4809, it is not currently supported.

Hopefully, over time someone (who has this bord) will contribute support for this newer platform. There are a few similar boards that have been released in the last few months. It should be relatively easy to add support.

The README for the library, lists the Boards/Platforms/MCUs that do have support.

cpapir, 2019-07-02:
I am trying to implement an IR remote control on a UNO wifi rev 2 can you suggest any one or anywhere I can get some help?

AnalysIR, 2019-07-02:
I am not aware of anyone working on those new platforms yet, for IR.

You could ask the manufacturer/Seller for support or post on the Arduino forum.

I would normally suggest our own IRForum, but we dont have any ATMEGA4809 platforms available yet.

It should only be a small amount of code for anyone with the platform.

ArminJo, 2020-07-04:
ATMEGA4809 is now supported.

closed 2020-07-04 by ArminJo