Arduino-IRremote/Arduino-IRremote #643: SoftPWM.h conflicts with IRremote.h

CarlouT, 2019-07-11:
I tried using SoftPWM.h and IRremote.h but it looks like these two libraries has conficts. Can anybody help resolve the issue? Here's the error log. Thanks.

Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

libraries\IRremote\IRremote.cpp.o (symbol from plugin): In function `MATCH(int, int)':

(.text+0x0): multiple definition of `__vector_13'

libraries\SoftPWM-master\SoftPWM.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files/windowsapps/arduinollc.arduinoide_1.8.21.0_x86__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

CarlouT, 2019-07-11:
Here's the code if needed. Not much on it for now. Its jsut declaring the two libraries and upload when I get the error.

`#include <SoftPWM.h>
#include <IRremote.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}`

AnalysIR, 2019-07-12:
It looks like both libraries are using Timer2 by default & the interrupt vectors are being declared twice, which results in the compile error.

Solution: in boarddefs.h of IRremote...configure another timer instead of Timer2.
You can see the list of available timers and associated output pins in the README. Your IR Tx pin will change with each timer.

Out of curiosity, why do you need both softpwm & IRremote in the same application?

CarlouT, 2019-07-13:
Hi,

This project is to control high powered LED's to lit my aquarium. Unlike standard aquarium lighting, when you turn them on, the bulbs will transition from 0 brightness to full brighness which stresses the fish. Conventional aquarium lights also has to be manually turned on and off. My goal is to build my custom aquarium lighting with:

1. A soft start functionality
2. be able to schedule when the LED's will turn on and off.
3. Control my lights with an IR remote control
4. Display the light parameters and set configuraiton using a 3.5" TFT LCD Screen Module 480x320 shield. 

The board I am using is an Arduino MEGA. This TFT shield has already used all the PWM outputs (pins 0 - 13) of the MEGA. So in order to drive the MOSFETS to power the LED's, I have to use the SOFTPWM library to create PMW outputs to some of the remaining pins of the MEGA. Unfortunatly, the IR remote library has some conflicts.

I don't have experience to configure boarddefs.h to use another timer. I would appreciate if you can provide some instructions on how to do this.

[image]

An additional feature I am looking into are the following:

1. Add a thermocouple to measure the water temperature. (To be added)
2. Add an ultrasonic sensor module to measure the hight of the aquarium water. (To be added)
3. Add a motion detector to overide the scheduled lighting and turn the LED's to the certain setpoint when a motion is detected. (This is already working)

I tried prototypng the temperature sensor using the DallasTemperature.h library but my program does not compile. I am getting this error.

0100_My_TFT_Screen.ino_V3:53:1: error: 'DallasTemperature' does not name a type

 DallasTemperature sensors(&oneWire);                                          // Pass our oneWire reference to Dallas Temperature.

It might be too much to ask but if you encountered this error before, kindly share the solution.

Many thanks
Carlou

AnalysIR, 2019-07-14:
I believe there are a few more PWM pins available on the Mega(beyond 0-13)...so use them instead and drop the softpwm in favour of analogWrite(). The conflict should be removed.

ref: [analogWrite()](https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/)

_General Comment
- using that many pins for the display should be avoided if possible
- Many of these module can also  be accessed in SPI mode , using much less pins. Check if yours does.
- If not buy displays with SPI support, in future.
- same applies for any peripheral (use I2C or SPI to reduce  pin count). Usually there are libraries available for most modules._

CarlouT, 2019-07-14:
Thanks heaps for the advice. I will defenitly try I2c or SPI driven displays if ever I will build another one again. This is my first Arduino project and I did not have any clue about this limitations when I bought my devices. A big mistake buying this display that uses too many pins but I have to live with it for now and find solutions to the conflicts.

Yes pins 44, 45, & 46 of the Mega are PWM's. But I need more. Four for the RGBW LED's and another one for the passive buzzer. A very big price I have to bear in buying the wrong display. The price of the new SPI display is not an issue. The issue I am foreseeing is to redo the code for the new display to re-create the menues, pass they dynamic parameters to the display and redoing the touch screen functions. Big work if my current code will not be compatible to the new SPI display. This is the TFT display I have 
[image]

Again thanks a lot for the help.

AnalysIR, 2019-07-14:
ok, change the fooling lines in boarders.h, around line 82

```
// Arduino Mega
#elif defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
	//#define IR_USE_TIMER1   // tx = pin 11
	#define IR_USE_TIMER2     // tx = pin 9
	//#define IR_USE_TIMER3   // tx = pin 5
	//#define IR_USE_TIMER4   // tx = pin 6
	//#define IR_USE_TIMER5   // tx = pin 46
```

just comment the timer2 line and uncomment one of the others. For for IR rx only it shouldn't matter which other one you use

after saving the changes, restart the arduino IDE

CarlouT, 2019-07-15:
I tried your suggestion. Putting the two libraries is a blank sketch has fixed the problem. Not unlike before that by putting the two libraties in a blank sketch ends up in an error when you compile. 

Unfortunatly, if I use the IRremote in my project, there are new conflics that appeared.

AnalysIR, 2019-07-15:
OK

Maybe try using the [TimerOne](https://www.pjrc.com/teensy/td_libs_TimerOne.html) library as follows:

- to generate an interrupt every 1 (or more)milliSec.
- Then within the interrupt routine(ISR), write the code/logic to do your own 'softPWM'
- Every time the interrupt routine is called, you have to decide when to toggle the output to generate your PWM & suty cycle and/or tone. ...for each output pin.
- This way you can use any pins for the PWM, for the extra interrupt overhead/cost
- This library uses Timer1 or Timer3
- Play around with this library first, **on its own**, to get used to it and test out the PWM feature (use the interrupt features only...not the actual PWM of this library)
- TIP: Using digitalWriteFast or diect port mapping, will reduce the overhead substantially, when toggling the softpwm pins. But start with the normla digitalWrite approach first.

ArminJo, 2020-07-01:
You may want to try [IRMP](https://github.com/ukw100/IRMP), which supports receiving protocols without using a timer. See Interrupt example.

closed 2020-07-01 by ArminJo