Arduino-IRremote/Arduino-IRremote #645: PWM LED problem with IRremote library (timers).

gavron04, 2019-07-19:
**Board:** ARDUINO Micro
**Library Version:** 2.1.0
**Protocol:** All

Hello. I have simple code which drive PWM LED.

Board - arduino micro 32u4. IR connected to pin 5 

https://www.arduino.cc/en/Hacking/PinMapping32u4

Problem:
```
//RGB//
int redPin = 9;
int greenPin = 6;
int bluePin = 4;
//RGB//
int potPin_red = A5;
int potPin_green = A4;
int potPin_blue = A3;
//RGB//
int readValue_red;
int readValue_green;
int readValue_blue;
//RGB//
int writeValue_red;
int writeValue_green;
int writeValue_blue;

void loop()
{
  //RGB//
  analogWrite(redPin,  analogRead(potPin_red) / 4);
  analogWrite(greenPin, analogRead(potPin_green) / 4);
  analogWrite(bluePin, analogRead(potPin_blue) / 4);
}
```

On 6pin which is green color there is 4 timer for IR remote library. Without IRremote library - code works ok. But after added IRremote - on 6 pin I can only work like 0-1 not PWM. 

There is a fix to change timer4 from 6pin to different where IR will works properly, and 6 pin will work ok with PWM??

Regards

gavron04, 2019-07-19:
Same situation https://forum.arduino.cc/index.php?topic=333861.0

AnalysIR, 2019-07-19:
First thing: on the micro pin 4(blue) doesnt seem to support PWM. Please check/verify.

Otherwise, hard to comment as I dont see any use of IRremote in your code.

Using pins 6,9,10 for color (try some others for IR Rx) & Timer3 for IRremote may be your best shot with this particular platform,

The micro suffers from low pin availablity...resulting in your conflicts (assuming it not just a coding bug)

If not, consider using a nano or UNO, instead.

gavron04, 2019-07-19:
First thing: on the micro pin 4(blue) doesnt seem to support PWM. Please check/verify.
>> right. My fault. I will try softPWM

Nano or UNO - I cant. I have changed from them to micro , because micro (32u4) have more SRAM Bytes (2,560 vs 2,048 , which I need).

How to change Timer for IR quickly?

AnalysIR, 2019-07-21:
The chages to the timer used are made in file boarddfs.h.
Look for USE_TMIERx (x =timer number) for the platform you are using.

Note the 32u can be found under Sparkfun micro, Teensy, ATMega32U4...which adds to the confusiom. So make sure the one you select matches your choice in the IDE.

> 
Nano or UNO - I cant. I have changed from them to micro , because micro (32u4) have more SRAM Bytes (2,560 vs 2,048 , which I need).

Have you considered using the available EEPROM or flash/PROGMEM instead of SRAM?

gavron04, 2019-07-28:
I have 32u4 Robotdyn ; in IDE - Genuino Micro

After changed from Timer4HS to Timer3 I get:
error "Please add OC3A pin number here\n ; in boarddefs.h

```
//-----------------
#if defined(CORE_OC3A_PIN)
#	define TIMER_PWM_PIN  CORE_OC3A_PIN  // Teensy
#elif defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
#	define TIMER_PWM_PIN  5              // Arduino Mega
#elif defined(__AVR_ATmega1284__) || defined(__AVR_ATmega1284P__)
#	define TIMER_PWM_PIN  6              // MightyCore
#else
#	error "Please add OC3A pin number here\n"
#endif
```

Have you considered using the available EEPROM or flash/PROGMEM instead of SRAM?
>> yes, i have use PROGMEM and other. But Adafruit GFX library for OLED need more space.

closed 2020-07-04 by ArminJo