Arduino-IRremote/Arduino-IRremote #647: Issue with sending raw signal to Acson air conditioner

hamzasharuf, 2019-07-25:
Hello,
I'm trying to reverse engineering my own air conditioner remote control (**ACSON** brand) https://www.acson.com.my/controller .

So First I made a simple circuit using **Arduino Uno** and **IR receiver** then I uploaded **IRrecvDumpV2** example And here is what I got

> Encoding  : UNKNOWN
> Code      : D90ABF7C (32 bits)
> Timing[131]: 
>      +4550, -2450     + 400, - 350     + 400, - 900     + 400, - 900
>      + 400, - 350     + 350, - 950     + 350, - 350     + 400, - 350
>      + 350, - 400     + 350, - 350     + 350, - 950     + 350, - 400
>      + 350, - 350     + 400, - 350     + 350, - 950     + 350, - 350
>      + 400, - 350     + 350, - 350     + 350, - 950     + 400, - 900
>      + 400, - 350     + 350, - 950     + 350, - 350     + 350, - 400
>      + 350, - 400     + 350, - 950     + 300, - 400     + 350, - 400
>      + 350, - 350     + 400, - 900     + 350, - 400     + 350, - 350
>      + 400, - 350     + 350, - 350     + 400, - 350     + 350, - 400
>      + 350, - 350     + 350, - 950     + 350, - 400     + 300, - 400
>      + 350, - 400     + 350, - 350     + 400, - 350     + 350, - 350
>      + 400, - 350     + 350, - 950     + 350, - 350     + 350, - 400
>      + 350, - 400     + 350, - 350     + 350, - 950     + 350, - 950
>      + 350, - 400     + 300, -1000     + 350, - 350     + 350, - 400
>      + 350, - 350     + 400, - 350     + 350, - 350     + 350, - 400
>      + 350, - 950     + 350, - 950     + 350, - 350     + 400, - 900
>      + 350, - 400     + 350
>      
> unsigned int  rawData[131] = {4550,2450, 400,350, 400,900, 400,900, 400,350, 350,950, 350,350, 400,350, 350,400, 350,350, 350,950, 350,400, 350,350, 400,350, 350,950, 350,350, 400,350, 350,350, 350,950, 400,900, 400,350, 350,950, 350,350, 350,400, 350,400, 350,950, 300,400, 350,400, 350,350, 400,900, 350,400, 350,350, 400,350, 350,350, 400,350, 350,400, 350,350, 350,950, 350,400, 300,400, 350,400, 350,350, 400,350, 350,350, 400,350, 350,950, 350,350, 350,400, 350,400, 350,350, 350,950, 350,950, 350,400, 300,1000, 350,350, 350,400, 350,350, 400,350, 350,350, 350,400, 350,950, 350,950, 350,350, 400,900, 350,400, 350};  // UNKNOWN D90ABF7C

I used my Xiaomi remote control app to control AC and it gave me the same code.
but when I used another android app called **SURE** and tried to control the AC it worked also ,however when i tried to read the raw code again using arduino it gave a different code.

> Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
Timing[27]: 
     + 200, - 450     + 250, -1150     + 200, - 450     + 200, -1150
     + 250, - 450     + 200, - 450     + 200, -1150     + 250, - 450
     + 200, - 450     + 200, -1150     + 250, -1150     + 200, - 450
     + 250, - 400     + 300
unsigned int  rawData[27] = {200,450, 250,1150, 200,450, 200,1150, 250,450, 200,450, 200,1150, 250,450, 200,450, 200,1150, 250,1150, 200,450, 250,400, 300};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;

I tried to resend both codes using my own app **but the AC gave no response**, so I decided to test my **Samsung brand TV** following same steps and it worked 

So How did the AC accept two different Raw codes with different protocols?

AnalysIR, 2019-07-25:
The first signal you posted decodes as a valid DAIKIN 64 bit AC signal on our AnalysIR application.

The second signal just looks like a partial capture of the signal.

My **guess** with the information provided is that there may be some issue with the way you are recording the signals.

You can ignore the encoding & Code values from IRremote as the library doesn't recognize the signal.

Check out this post from our blog which may help (_Its more accurate_): [Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

hamzasharuf, 2019-07-25:
You are completely right
After following the [example ](https://www.analysir.com/blog/wp-content/uploads/2014/03/Arduino_Record_Long_AirConditioner_Infrared_Signals_10.txt)mentioned in this topic [Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/) I found a different buffer size.
Now they are **133** instead of **131** in the first time.

> Raw: (133) 4620, -2464, 400, -364, 372, -940, 372, -940, 372, -364, 348, -964, 372, -364, 372, -364, 372, -360, 348, -388, 372, -940, 372, -364, 344, -392, 368, -364, 372, -940, 372, -364, 372, -364, 372, -940, 372, -364, 372, -360, 376, -360, 372, -940, 372, -940, 376, -360, 372, -364, 372, -364, 372, -936, 376, -936, 376, -360, 376, -936, 376, -360, 372, -364, 368, -368, 372, -360, 376, -360, 372, -364, 372, -364, 368, -940, 376, -360, 372, -364, 376, -360, 372, -364, 368, -364, 372, -364, 372, -364, 372, -936, 376, -360, 376, -360, 376, -360, 372, -364, 368, -940, 376, -936, 376, -364, 368, -940, 372, -368, 368, -364, 348, -388, 372, -940, 372, -364, 372, -360, 348, -964, 372, -364, 372, -364, 372, -364, 372, -936, 376, -20256, 4620

So the difference I found is that the first example [IRrecvDumpV2 ](https://github.com/z3t0/Arduino-IRremote/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino)for some reason has ignored the last two bits **-20256, 4620**.

But now it's working as expected
Thank you so much @AnalysIR

hamzasharuf, 2019-07-25:
But now I'm still wondering
I tried to test the SURE remote again it gave me a different Raw buffer

> Raw: (285) 4368, -2540, 328, -336, 324, -344, 324, -340, 324, -344, 324, -340, 324, -344, 324, -340, 324, -340, 324, -344, 324, -340, 324, -344, 324, -340, 324, -344, 324, -340, 324, -340, 324, -344, 324, -340, 324, -344, 324, -340, 324, -340, 324, -344, 324, -340, 324, -344, 324, -340, 324, -344, 320, -344, 324, -340, 328, -340, 324, -340, 324, -344, 324, -340, 324, -344, 324, -340, 324, -344, 324, -340, 324, -344, 324, -340, 324, -340, 328, -340, 324, -340, 328, -336, 328, -340, 324, -340, 328, -340, 324, -340, 328, -336, 328, -340, 328, -336, 328, -340, 328, -336, 328, -340, 324, -340, 328, -340, 324, -340, 324, -340, 328, -340, 324, -340, 328, -340, 324, -340, 328, -340, 324, -340, 328, -336, 328, -340, 324, -340, 328, -340, 324, -340, 324, -344, 320, -344, 324, -340, 328, -340, 324, -340, 324, -344, 324, -340, 324, -344, 324, -340, 324, -340, 328, -340, 324, -340, 324, -344, 324, -340, 324, -344, 324, -340, 324, -340, 328, -340, 324, -340, 324, -344, 324, -340, 324, -340, 328, -340, 324, -340, 328, -340, 324, -340, 324, -344, 324, -340, 324, -340, 328, -340, 324, -340, 328, -340, 324, -340, 324, -344, 324, -340, 324, -340, 328, -340, 324, -340, 328, -340, 324, -340, 324, -344, 324, -340, 324, -340, 324, -344, 324, -340, 328, -340, 324, -340, 324, -344, 324, -340, 324, -340, 328, -1064, 328, -344, 320, -344, 324, -1068, 324, -1068, 328, -340, 324, -340, 328, -340, 324, -340, 324, -1068, 328, -340, 324, -344, 324, -1064, 332, -340, 324, -1064, 332, -336, 328, -340, 324, -1068, 328, -340, 324, -340, 324, -1068, 328, -1064, 332, -336, 328, -340, 324, -20540, 1368

And when I tested again It also worked, I guess because of the buffer size (285) it wasn't give me the correct Raw in the first time, But how is it possible for the AC to accept two different Raw signals ?

AnalysIR, 2019-07-25:
The AC just sees then as signals.

I have seen systems where Remotes will send 2 different signals for every button press. presumably because hte manufactureer wants to use 1 remote for 2 different ACs.

There is no reason to suspect that manufacturers of the AC control boards wouldnt do somethng similar.

hamzasharuf, 2019-08-07:
I just reopened the issue to add some info
After analyzing the signal transmitted by [SURE](https://play.google.com/store/apps/details?id=com.tekoia.sure.activities&hl=en) remote app I found out that my AC signal is just a part of this signal and the second one is maybe another complete signal as @AnalysIR said because Acson AC accepts two different protocols so maybe this is the second one and they want to send both signal at the same time I'm not sure.
Any way .. Thank you so much for your help @AnalysIR.

closed 2019-08-07 by hamzasharuf