Arduino-IRremote/Arduino-IRremote #651: redefinition of 'IRrecv irrecv'

pronoym99, 2019-09-05:
Following is the error I received after executing this code. I changed the variable name 'RECV_PIN' to something else if at all there was any conflicts. however that didn't help.

**exit status 1**
**redefinition of 'IRrecv irrecv'**

#include <IRremote.h> 

// Define sensor pin
const int RECV_PIN = 4;

// Define IR Receiver and Results Objects
IRrecv irrecv(RECV_PIN);
decode_results results;

void setup()
{
  // Serial Monitor @ 9600 baud
  Serial.begin(9600);
  // Enable the IR Receiver
  irrecv.enableIRIn();
}

void loop()
{
  if (irrecv.decode(&results))
  {
    // Print Code in HEX
    Serial.println(results.value, HEX);
    irrecv.resume();
  }
}

AnalysIR, 2019-09-05:
if you look at the IRrecv demo, you will notice that RECV_PIN is not defined as **const**...

...maybe that is the issue???

pronoym99, 2019-09-05:
I am sorry to say @AnalysIR but that doesn't seem to help. I changed the definition to a purely integer type value but the exit status and the error code remains the same. *It would be worthwhile to note that a month back I had already executed and tested this code with my TV remotes and it worked fine then.* I haven't modified the code since those initial tests.

lpuser13, 2019-09-05:
your code above compiles fine on my system for a UNO using the Arduino IDE.

So based on the info you provided, it seems unlikey to be an issue with the library.

All I can think of now is to update all your libraries & boards to the latest versions & restart the IDE.

If that does not work try uninstalling the IDE fully and re-install the latest IDE

AnalysIR, 2019-09-05:
your code above compiles fine on my system for a UNO using the Arduino IDE.

So based on the info you provided, it seems unlikey to be an issue with the library.

All I can think of now is to update all your libraries & boards to the latest versions & restart the IDE.

If that does not work try uninstalling the IDE fully and re-install the latest IDE

pronoym99, 2019-09-05:
I started with an up to date installation of the IDE, the IRrecv demo example file works fine even with *int* being replaced with *const int*. However at the same time the test file which I wrote continues throwing up the same error although its exactly the same as the demo file with receive pin being 4 instead of 11.

s-light, 2019-09-26:
what do you mean by
>copy pasting the example file in my project

1. so if you copy the example to another folder it works?!
1. and if you copy the content into your project id does not work?
1. do you use multiple files for you project?
  1. if yes - could you try to build up again from a fresh folder - so you know when (adding one part of your porject after each other) the error ocurs

the IDE / compiler can get very confusing on bigger projects - partly because the ide tries to do smart things for us - but eventually these smart things can get in the way of  complex usage...

MicheleCioccarelli, 2020-02-27:
i have a similar problem across multiple files with the platformIO IDE.

pronoym99, 2020-02-28:
@MicheleCioccarelli It would be relatively difficult to comment on the platformIO IDE because as far as I know it works on a relatively complex file system. However, since its quite universal I would be moving to the platformIO system very soon.

AnalysIR, 2020-02-28:
@MicheleCioccarelli envorinment
Likely that more than one IRremote library (or fork) is configured in your system. Unfortunatly, I cannot add any insight into how Platform.io works.

ArminJo, 2020-07-04:
It looks like the make process compiles two sources with the same IRrecv irrecv definitions.

Look carefully at the build log. The error should be explained there like `../src/IRrecvDemo.cpp:16:8: note: 'IRrecv irrecv' previously declared here`

closed 2020-07-04 by ArminJo