Arduino-IRremote/Arduino-IRremote #652: RC5 codes table

mrfinrod, 2019-09-08:
In examples I found this codes for RC5 protocol:
testRC5(0xfff, 12);
testRC5(0x000, 12);
testRC5(0xaaa, 12);
testRC5(0x555, 12);
testRC5(0x123, 12);

What does this codes means? And where can I find all RC5 codes and descriptions for each code?

AnalysIR, 2019-09-08:
Those values are just test values without any particular meaning.

For more info on the RC5 data formats see: =>[Philips RC-5 Protocol](https://www.sbprojects.net/knowledge/ir/rc5.php)

The site also covers a range of other IR protocols.

mrfinrod, 2019-09-08:
Thank you!
Can you help me, I doen't understand the values.
For example:
$0F - 15 | Brightness +

What code must I write to the function testRC5(0xfff, 12) ?

AnalysIR, 2019-09-08:
I suggest you get an RC5 remote and compare the values provided by IRremote to the description in the site above. That is the best way to learn.

If you don't have an RC5 remote use another common TV one and do the same, assuming its also described on that site.

Also, I am not aware of any function in IRremote called testRC5  (sendRC5 is available)..so maybe a typo?

closed 2019-09-08 by mrfinrod